hamuchiwa/AutoRCCar #2: Some questions about AutoCar

myxingchao, 2016-02-28:
Read your AutoRCCar, I was interested, but would like to ask you some questions about your host computer what operating system? I'm under my windows host computer to compile and run the wrong side of the python, you are using what python editor? I am using pycharm. Looking forward to your reply

hamuchiwa, 2016-03-02:
Hi, I use Mac, Python 2.7 and PyCharm.

myxingchao, 2016-03-10:
Hello, I used pycharm, established in ubuntu project, what is the whole file inclusion into? Then run rc_driver.py it? Test run, last seen a bunch of errors
Date: Tue, 1 Mar 2016 20:06:20 -0800

hamuchiwa, 2016-03-17:
I need more info, what errors are you seeing? Are you using python 2?
Does the IP address and port on raspberry pi match the ones on the server?
Does the serial port match the one you assigned to the Arduino?

yashim, 2016-03-20:
Thank you for sharing the code! 
How much training data is needed?

myxingchao, 2016-04-08:
Hello, I recently to learn python opencv Raspberry pi programming, but that is not the initialization of the camera, can you help me write a little opencv camera collection of video and then displayed on the page samples ? / Look forward to your reply
Date: Thu, 17 Mar 2016 16:05:33 -0700

hamuchiwa, 2016-04-10:
Hi, 
yashid: depends on your track, in my case it's about 300 images
myxingchao: test files just added

MihaHoo, 2016-06-06:
Hi,
I have similar setup like yours, trying to get car moving with rc_control_test.py, Flashed .ino to Arduino Uno which is connected to laptop (running Ubuntu) via USB. Replaced /dev/tty/usb411something with my correct device /dev/tty/ACM0. Python is 2.7 and pygame is also installed as well all requirements.

However, when running rc_control_test.py via command line (with sudo or w/o) nothing happens.

pi@E20:~/git/AutoRCCar/test$ sudo python rc_control_test.py 
[sudo] password for pi:  (here I typed pwd, enter and then arrows). Following appears on terminal:
^[[A^[[B^[[C^[[D^[[A^[[B

And this after hitting ctrl-c:
^CTraceback (most recent call last):
  File "rc_control_test.py", line 69, in <module>
    RCTest()
  File "rc_control_test.py", line 13, in __init__
    self.steer()
  File "rc_control_test.py", line 18, in steer
    for event in pygame.event.get():
KeyboardInterrupt

Wires are correctly connected, I wrote simple .ino to test and it works OK.
int test = 6;

void setup()
{
  pinMode(test, OUTPUT);    
}

void loop()
{
  digitalWrite(test, HIGH);   
  delay(1000);                  
  digitalWrite(test, LOW);   
  delay(5);                 
}

Any ideas what I am missing? Thank you.

hamuchiwa, 2016-06-07:
Mihahoo: It works for me. I'm using Mac but I don't think there should be any difference. When you hit enter, the command line window should be grayed out, then the program starts to listen for key press. Just don't click the window.

MihaHoo, 2016-06-07:
Hi,
Switched to El Capitan.  pygame does not work with python 2.7.11 (no module
found) With python3.5.1 pygame ja serial both are ok nyt rc_control_tst.py
fails: (if i click terminal)

Traceback (most recent call last):

  File "rc_control_test.py", line 69, in <module>

    RCTest()

  File "rc_control_test.py", line 14, in __init__

    self.steer()

  File "rc_control_test.py", line 43, in steer

    self.ser.write(chr(1))

  File "/usr/local/lib/python3.5/site-packages/serial/serialposix.py", line
494, in write

    d = to_bytes(data)

  File "/usr/local/lib/python3.5/site-packages/serial/serialutil.py", line
63, in to_bytes

    raise TypeError('unicode strings are not supported, please encode to
bytes: {!r}'.format(seq))

TypeError: unicode strings are not supported, please encode to bytes: '\x01'

And if I DON*T click terminal windows, instead of that for example safari,
nothing happens. car does not move. Clicking python-icon on doc
kills script with error above.

What was your version of OSX and python? From where you found working
intructions to get pyjama working with python2? Thanks.

MihaHoo, 2016-06-07:
Hi again,

Managed to install pygame with this
https://github.com/Homebrew/homebrew-python and now rc_control_test.py
works just fine :)

MihaHoo, 2016-06-08:
Well, this is getting interesting.. for some reason collect_training_data.py behaves like rc_control_test.py did "^[[A^[[B^[[C^[[D^[[A^[[B" on screen. Weird thing is that rc_control_test.py works still. Tested on very same terminal window @OSX. And as side note, rc_control_test.py works on Ubuntu if GUI (X windows) is not running.. Environmental handling on scripts needs adjustment or something. I am now very pro with python

hamuchiwa, 2016-06-09:
MihaHoo: have you tried using a python IDE, for example, PyCharm?

MihaHoo, 2016-06-10:
No I haven't tried, yet. Managed to tackle problems and proceeding. Should I take images for calibration with resolution 320x240 like you have done? I

hamuchiwa, 2016-06-14:
MihaHoo: I used a Raspberry Pi model B+ at the time, I sort of had to compromise the resolution in order to achieve low latency video streaming. If you are using a newer model, probably you can go higher.

rohanalwaysrocks, 2016-06-14:
i am using windows 7 will i be able to do the project using windows 7 operating system?

hamuchiwa, 2016-06-14:
rohanalwaysrocks: Yes. Please refer to read me for what software you need in this project. I have closed your other thread.

rohanalwaysrocks, 2016-06-15:
thanks. i have install python pyserial pygame numpy all in 32bit and my system is 64 bit i have also shouldered uo pins to the rc board now how do i send wireless data from rasp to my laptop?

rohanalwaysrocks, 2016-06-15:
do i need to install visual studio?

rohanalwaysrocks, 2016-06-15:
i have burned the rc_keyboard_control.ino to the audrino board by simply clicking upload on audrino software and i am running rc_control_test.py using pyCharm on windows. i have installed pygame and pyserial on windows and i have connected it to com13 but as soon as i run the program on pyCharm nothing happens when i press the keys

MihaHoo, 2016-06-15:
rohanalwaysrocks , have you made wiring between arduino and rc controller? Can you see led on arduino flashing according to your keypresses? My arduino flashes on board led marked "RX"

MihaHoo, 2016-06-15:
Oh, now i have same "^[[A^[[B^[[C^[[D^[[A^[[B" problem with collect_training_data.py and old trick, adding line "pygame.display.set_mode((100, 100))" does not help (maybe putting it into wrong place inside script..)

rohanalwaysrocks, 2016-06-15:
yes i have made the wiring between arduino and rc controller. as per the program 6 7 10 11 pin of arduino i have shouldered to the remote control. and no the recieve led on arduino does not flash when i press arrow keys

MihaHoo, 2016-06-15:
Have you connected also GND? Minus on remote to GND on Arduino

rohanalwaysrocks, 2016-06-15:
yeah i have connected the ground. the rx led doesnt glow when i press the arrow button on keyboard

rohanalwaysrocks, 2016-06-15:
when i run the rc_control_test.py and press arrow buttons the cursor moves anywhere. it doesnt get locked in pycharm. how to solve that

MihaHoo, 2016-06-17:
Hi,
ultrasonic_client.py throws error to computer where rc_driver.py runs:

Connection closed on thread 2
----------------------------------------
Exception happened during processing of request from ('192.168.0.100', 60976)
Traceback (most recent call last):
  File "/usr/lib/python2.7/SocketServer.py", line 290, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 318, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 331, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python2.7/SocketServer.py", line 652, in __init__
    self.handle()
  File "rc_driver.py", line 138, in handle
    sensor_data = round(float(self.data), 1)
ValueError: could not convert string to float: 
----------------------------------------
Right
Right
Right

rickbeacham, 2016-06-17:
Your data may not be what you expect -- it seems you're expecting, but not
getting, floats some where.  A simple solution to figuring out where this
occurs would be to add a try/except or Isdigit() to a for-loop:  This way
you can debug.  self.request.recv(1024) is not returning strings that can
be converted to Floats.
Please echo out the data in self.request.recv(1024)

MihaHoo, 2016-06-17:
Competence exceeded. Not very familiar w/ python. ultrasonic_client.py is awful. First of all, i need to run it w/ python3 to bypass this:
SyntaxError: Non-ASCII character '\xe2' in file ultrasonic_client.py.orig on line 4, but no encoding declared; see http://python.org/dev/peps/pep-0263/ for details

And lot of these "IndentationError: unindent does not match any outer indentation level" and "IndentationError: unexpected indent" which are fixed manually.

This also fixed: SyntaxError: Missing parentheses in call to 'print' by modifying line
 print "Distance : %.1f cm" % distance
to
 print("Distance : %.1f cm") % distance

Quite much modified script looks like this:
from socket import *
import time
import RPi.GPIO as GPIO

# referring to the pins by GPIO numbers
GPIO.setmode(GPIO.BCM)

# define pi GPIO
GPIO_TRIGGER = 23
GPIO_ECHO    = 24
GPIO.setup(GPIO_TRIGGER,GPIO.OUT)
GPIO.setup(GPIO_ECHO,GPIO.IN)

GPIO.setwarnings(False)

# create a socket and bind socket to the host
client_socket = socket(AF_INET, SOCK_STREAM)
client_socket.connect(('192.168.0.101', 8002))

def measure():
    """
    measure distance
    """
GPIO.output(GPIO_TRIGGER, True)
time.sleep(0.00001)
GPIO.output(GPIO_TRIGGER, False)
start = time.time()

while GPIO.input(GPIO_ECHO)==0:
    start = time.time()

while GPIO.input(GPIO_ECHO)==1:
    stop = time.time()

elapsed = stop-start
distance = (elapsed * 34300)/2
print ("Distance:",distance,"cm")
#return distance
#quit()
# referring to the pins by GPIO numbers
#GPIO.setmode(GPIO.BCM)
#
## define pi GPIO
#GPIO_TRIGGER = 23
#GPIO_ECHO    = 24

# output pin: Trigger
GPIO.setup(GPIO_TRIGGER,GPIO.OUT)
# input pin: Echo
GPIO.setup(GPIO_ECHO,GPIO.IN)
# initialize trigger pin to low
GPIO.output(GPIO_TRIGGER, False)

try:
    while True:
        distance = measure()
        #print("Distance : %.1f cm") % distance
        # send data to the host every 0.5 sec
        client_socket.send(str(distance))
        time.sleep(0.5)
finally:
    client_socket.close()
    GPIO.cleanup()

And gives error:
Distance: 106.38035535812378 cm
Traceback (most recent call last):
  File "ultrasonic_client.py", line 59, in <module>
    client_socket.send(str(distance))
TypeError: 'str' does not support the buffer interface

On computer (rc_control.py) same error as mentioned previously:
Forward
Forward
Forward
Connection closed on thread 2
----------------------------------------
Exception happened during processing of request from ('192.168.0.100', 34758)
Traceback (most recent call last):
  File "/usr/lib/python2.7/SocketServer.py", line 290, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 318, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 331, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python2.7/SocketServer.py", line 652, in __init__
    self.handle()
  File "rc_driver.py", line 138, in handle
    sensor_data = round(float(self.data), 1)
ValueError: could not convert string to float: 
----------------------------------------
Forward
Forward
Forward
Forward
Forward

Issues:;
- Loop on ultrasonic_client.py is broken
- How to do this "add a try/except or Isdigit() to a for-loop" 

TypeError happens most likely because script is executed with python3 instead of python2. It is quite hard to understand how ultra-script works for others..

MihaHoo, 2016-06-17:
Modified script acts now:
pi@rc:~/git/AutoRCCar/raspberryPi $ python ultratest.py <pre>
ultratest.py:11: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(GPIO_TRIGGER,GPIO.OUT)
177.980709076</pre>
pi@rc:~/git/AutoRCCar/raspberryPi $ 

And on computer (rc_control.py):
<pre>
Right
Right
Right
Connection closed on thread 2
----------------------------------------
Exception happened during processing of request from ('192.168.0.100', 35084)
Traceback (most recent call last):
  File "/usr/lib/python2.7/SocketServer.py", line 290, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 318, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 331, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python2.7/SocketServer.py", line 652, in __init__
    self.handle()
  File "rc_driver.py", line 138, in handle
    sensor_data = round(float(self.data), 1)
ValueError: could not convert string to float: 
----------------------------------------
Connection closed on thread 2
----------------------------------------
`Right
Right
Right
Connection closed on thread 2
----------------------------------------
Exception happened during processing of request from ('192.168.0.100', 35084)
Traceback (most recent call last):
  File "/usr/lib/python2.7/SocketServer.py", line 290, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 318, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 331, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python2.7/SocketServer.py", line 652, in __init__
    self.handle()
  File "rc_driver.py", line 138, in handle
    sensor_data = round(float(self.data), 1)
ValueError: could not convert string to float: 
----------------------------------------
Connection closed on thread 2
---------------------------------------</pre>

My modified script:
<pre>
from socket import *
import time
import RPi.GPIO as GPIO

# referring to the pins by GPIO numbers
GPIO.setmode(GPIO.BCM)

# define pi GPIO
GPIO_TRIGGER = 23
GPIO_ECHO    = 24
GPIO.setup(GPIO_TRIGGER,GPIO.OUT)
GPIO.setup(GPIO_ECHO,GPIO.IN)

GPIO.setwarnings(False)

# create a socket and bind socket to the host
client_socket = socket(AF_INET, SOCK_STREAM)
client_socket.connect(('192.168.0.101', 8002))

def measure():
    """
    measure distance
    """
GPIO.output(GPIO_TRIGGER, True)
time.sleep(0.00001)
GPIO.output(GPIO_TRIGGER, False)
start = time.time()

while GPIO.input(GPIO_ECHO)==0:
    start = time.time()

while GPIO.input(GPIO_ECHO)==1:
    stop = time.time()

elapsed = stop-start
distance = (elapsed * 34300)/2
#print ("Distance:",distance,"cm")
print (distance)
#return distance
quit()
# referring to the pins by GPIO numbers
#GPIO.setmode(GPIO.BCM)
#
## define pi GPIO
#GPIO_TRIGGER = 23
#GPIO_ECHO    = 24

# output pin: Trigger
GPIO.setup(GPIO_TRIGGER,GPIO.OUT)
# input pin: Echo
GPIO.setup(GPIO_ECHO,GPIO.IN)
# initialize trigger pin to low
GPIO.output(GPIO_TRIGGER, False)

try:
    while True:
        distance = measure()
        #print("Distance : %.1f cm") % distance
        # send data to the host every 0.5 sec
        client_socket.send(str(distance))
        time.sleep(0.5)
finally:
    client_socket.close()
    GPIO.cleanup()
</pre>

I'm stuck with this, help would be nice to get. Thank you

hamuchiwa, 2016-06-18:
MiHaHoo: Try to do what rickbeacham suggested, print it out and check the type. Make sure you are sending the number as a string, receiving as string and then convert it to float.

MihaHoo, 2016-06-20:
"Please echo out the data in self.request.recv(1024)" Yeah, I know. But I don't know how.

MihaHoo, 2016-06-20:
Modified script (just to test)
<pre>
--
GPIO.output(GPIO_TRIGGER, False)

try:
    while True:
        distance = measure()
        print("distance")
        #print("Distance : %.1f cm") % distance
        # send data to the host every 0.5 sec
        client_socket.send(str(101))
        #client_socket.send(str(distance))
        time.sleep(0.5)
finally:
    client_socket.close()
--
</pre>

Prints once distance 41.666903496 and then text 'distance' on loop until Ctrl-c. 

On server-side (rc_control.py) following:
<pre>
 sensor_data
 sensor_data
Forward
 sensor_data
Forward
Forward
Forward
 sensor_data
 sensor_data
</pre>

Ideas? Hints?

MihaHoo, 2016-06-21:
Hi,
Still playing with ultrasonic_client.py. If I start python manually
<pre>
pi@pi:~/git/AutoRCCar/raspberryPi $ python
Python 2.7.9 (default, Mar  8 2015, 00:52:26) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from socket import *
>>> import time
>>> client_socket = socket(AF_INET, SOCK_STREAM)
>>> client_socket.connect(('192.168.0.101', 8002))
>>> client_socket.send(str(12))
2
</pre>

Following is visible on rc_control.py:
<pre>
--clip--
Stop sign ahead
Stop time: 4.85s
Stop sign ahead
Stop time: 5.00s
Waited for 5 seconds
Forward
Forward
Forward
Forward
Forward
Forward
Forward
Forward
Forward
Forward
 sensor_data
Stop, obstacle in front
Stop, obstacle in front
Stop, obstacle in front
--clip--
</pre>

For me this looks that PC-end (where rc_control.py runs) everything is more or less OK but ultrasonic_client.py needs to be fixed. Script send decimal point which is not accepted on server side.

MihaHoo, 2016-06-21:
Made own script which sends distance-data to the server. Works.

rohanalwaysrocks, 2016-06-21:
can you send the code

sarojtm, 2016-06-27:
The other day I was able to run collect_training_data.py and generate a .npz file but when I tried it today it is showing this error " temp_array = roi.reshape(1, 38400).astype(np.float32)
ValueError: total size of new array must be unchanged". Can anyone help me?

varun333, 2016-09-26:
hi, being new to raspi and arduino, i just want to know that it is necessary to have mac or os like windows 8 wil also work just fine so as to move and perform all the function carried in this project.
It will be great help as i have done the study related to it and interested  to make it as final year project.

varun333, 2016-09-26:
Working on my first module that is how to move car i had arduino uno board and windows 8 os, but its not working. If there is problem in my arduino IDE or i have buy to upgraded version.

sarojtm, 2016-09-26:
varun33 it worked for me please check the connection thoroughly. Then change "HIGH" to "LOW" and "LOW" to "HIGH" in the Arduino code

varun333, 2016-09-26:
well thanx, saroj but first i just wanna know that later there wont come any problem as i am using windows 8, well it will nice of you if provide me detail related arduino board that u have used.

sarojtm, 2016-09-27:
I have used arduino UNO but any other board might work as well.

varun333, 2016-09-27:
well saroj have u work on this single module or worked on the complete project, just asking in case i dwell myself in problem so can contact u for help.
Can have ur any source of contact or email for further assistance.

sarojtm, 2016-09-27:
Well I didn't exactly replicate it but this project was a great help.
Instead of using Picam I used Android phone's camera and interfaced with
the computer via python script. And then removed distance measurement part
of Raspberry Pie. Well you can reply to this thread. Cheers.

varun333, 2016-10-02:
hello Saroj i ran the ino and rc control test py file but there is coming command screen with underscore flashing, what does that mean? Am i going wrong or should I have to wait or I should start making changes to Arduino code

sarojtm, 2016-10-03:
1.does it show any error?
2.did you connect arduino to pc
3.did you configure the port number on the python code?
4.did you install pygame?
5.please add "pygame.display.set_mode((100,100))" under rc_test.py

I'll send you a code that worked for me

__author__ = 'zhengwang'

import serial
import pygame
from pygame.locals import *

class RCTest(object):

    def __init__(self):
        pygame.init()
        pygame.display.set_mode((100,100))
        self.ser = serial.Serial('COM4', 115200, timeout=1)
        self.send_inst = True
        self.steer()

    def steer(self):

        while self.send_inst:
            for event in pygame.event.get():
                if event.type == KEYDOWN:
                    key_input = pygame.key.get_pressed()

                    # complex orders
                    if key_input[pygame.K_UP] and key_input[pygame.K_RIGHT]:
                        print("Forward Right")
                        self.ser.write(chr(6))
                        #self.ser.write(chr(8))

                    elif key_input[pygame.K_UP] and key_input[pygame.K_LEFT]:
                        print("Forward Left")
                        self.ser.write(chr(7))
                        #self.ser.write(chr(8))

                    elif key_input[pygame.K_DOWN] and key_input[pygame.K_RIGHT]:
                        print("Reverse Right")
                        self.ser.write(chr(8))
                        #self.ser.write(chr(9))

                    elif key_input[pygame.K_DOWN] and key_input[pygame.K_LEFT]:
                        print("Reverse Left")
                        self.ser.write(chr(9))
                        #self.ser.write(chr(9))

                    # simple orders
                    elif key_input[pygame.K_UP]:
                        print("Forward")
                        self.ser.write(chr(1))

                    elif key_input[pygame.K_DOWN]:
                        print("Reverse")
                        self.ser.write(chr(2))

                    elif key_input[pygame.K_RIGHT]:
                        print("Right")
                        self.ser.write(chr(3))

                    elif key_input[pygame.K_LEFT]:
                        print("Left")
                        self.ser.write(chr(4))

                    # exit
                    elif key_input[pygame.K_x] or key_input[pygame.K_q]:
                        print 'Exit'
                        self.send_inst = False
                        self.ser.write(chr(0))
                        self.ser.close()
                        break

                elif event.type == pygame.KEYUP:
                    self.ser.write(chr(0))

if __name__ == '__main__':
    RCTest()

sarojtm, 2016-10-03:
Please go thoroughly the code comparing with one another.

varun333, 2016-10-03:
Big Thanx Saroj, it just worked fine.

sarojtm, 2016-10-03:
Cheers !!

KiranKanchi, 2016-10-31:
Has any one got this project Working ??????

isrugeek, 2017-02-11:
@ghost You solved my Problem! Thanks Man. about PyGame

varun333, 2017-03-04:
sarojtm Hey there again need your assistance? Anyhow I made the connectivity using vnc but not able to deploy the server_client.py code, so I am not able to stream the video on my laptop. Apart from this, is there any requirement to configure pycharm.
Reply soon or any other alternatives.
Any kind of help will be useful.Please.

isrugeek, 2017-03-05:
Check the IP address on both client and server files, then first run the server side script then client side

piguy12, 2017-05-03:
Hi guys I understand the camera feed maps frames  when the person drives around and maps to the output driving commands forward black left right out. I am unfamiliar with opecv MLP library, how does the neural network drive the car after its trained. 
Questions 
are we using the neural net to drive the car frame by frame? 

if you could just go from the point thread server and briefly describe how his classes work. 
Thanks!

farazhussain2015, 2017-08-08:
i am directly connecting my motors with arduino via motor driver L928N. everything is working but motors are not moving. any solution? PLZ

isrugeek, 2017-08-11:
@farazhussain2015 make sure you dedicate the right port in L928N and Arduino!

kxrf97, 2019-02-20:
i have connected all connected all the wires, when i run rc_control_test only left arrow key are working, other movement is no working but the LED and arduino RX are flashing. Any solution for this?