hamuchiwa/AutoRCCar #4: why not to use Raspberry Pi only?

ucy74, 2016-07-20:
Why You divide software to three platforms?
Raspberry can drive RC panel like an arduino.
Raspberry can run AI with native video/sonar calc too - no ext streaming.
Is this question of general Raspberry processor speed?

hamuchiwa, 2016-07-21:
Back then I used a Raspberry Pi model B+, I had to shrink the resolution to 320x240 to achieve "low" latency streaming. With the newer model, it may be possible to do all pieces on Raspberry pi. Another reason is majorly because it's so much easier and more efficient to write/test code on laptop. 
If you have a Raspberry Pi only version, please make a pull request. I'm sure everyone would love to see it. Thanks~

closed 2018-03-19 by hamuchiwa

jithendran1309, 2019-03-18:
why i cannot use raspberry pi for control also