hamuchiwa/AutoRCCar #5: Some question about the speed of car

r290085, 2016-08-02:
請問作者 你有調整遙控車的速度嗎?
我目前在做training 
可是車子速度太快 一直駛出白紙跑道 無法training完成
我想把車子速度調慢 有什麼方法嗎?

hamuchiwa, 2016-08-02:
你好， 一开始我有给车子加电位器（potentiometer），这样速度就可以调了。后来发现直接在轮子上缠一个皮筋反而更有效果。如果你用的电动车本身就有档位的话，那这个速度问题就解决了。

aha124226675, 2017-02-14:
請問一下 training 有什麼技巧嗎?
我training完跑rc_driver.py還是一直跑出界~
謝謝

hamuchiwa, 2017-02-15:
@aha124226675 如果自己开遥控车位置不是很理想的话，可以把车放到放到你认为合理的位置，然后收集一张图片。数据收的不错的话默认参数就可以，一层hidden layer足矣， 当然你也可以试着增加node的个数。

aha124226675, 2017-02-22:
了解 謝謝作者~
另外想請問一下，遙控車大概使用什麼大小的會比較適合? 
我目前使用1:14遙控車~
覺得略顯太大容易出界

hamuchiwa, 2017-02-22:
@aha124226675  一个是速度降下来 还有一个是摄像头对准路面 让收集的图片里只有路面的信息
虽然说只截取下半部分也可以 不过取整张照片的话准确率要好很多

aha124226675, 2017-02-23:
请问一下 potentiometer 要选几欧姆的会比较好?
另外请问一下作者关于橡皮筋减速要如何执行呢?

hamuchiwa, 2017-02-23:
@aha124226675 小一点儿的为好 用这个缺点是电流也跟着减小了 
橡皮筋的话 我是缠在轮轴上 当时手头上也没别的东西 结果发现橡皮筋还挺好用

aha124226675, 2017-03-02:
谢谢作者〜已使用橡皮筋降速
另外想请问为何红绿灯一直无法识别呢？
我已在网路跟作者买同一款灯型！
谢谢麻烦了

hamuchiwa, 2017-03-02:
@aha124226675 你真厉害 能买着同款的。。。
灯不亮的时候应该能给出一个矩形识别区域，能不能具体描述一下是怎么无法识别，是不识别物体呢还是不识别等的红绿

aha124226675, 2017-05-09:
不好意思这么慢才回复,我于掏宝上购买同款红绿灯。
也许是红绿灯亮度不足导致无法识别。
再来就是我将相机镜头调下以便照完整路面,但是对于红绿灯及STOP号志距离改变就无法辨识。

以下是我的youtube完成视频
https://www.youtube.com/watch?v=KrvvmLEQk8o

另外我想请问一下类神经网路的Hidden layer有几层啊?
及想训练新号志 xml档该如何建立。

谢谢作者 感激不尽。

hamuchiwa, 2017-05-10:
@aha124226675 视频很棒！
镜头角度的问题真的是没有办法，因为只有一个镜头所以多少要妥协一下。试着截取一半图片吧，上半部给红绿灯下半部给路面。训练只用下半张图片，看看准确率最高能达到多少。

我当是只用了一层，当然了你可以再加一层或者增加node的个数。训练完了自动保存xml。如果你指的是信号灯的cascade classifier， 看一下主页的readme， 在文档的最后我放了一个链接，我就是照着那个做的红绿灯和stop标志的识别。

aha124226675, 2017-05-17:
谢谢作者,目前持续实作中,请问一下教学平台是 linux吗? 我用Windows 7能否学习?

hamuchiwa, 2017-05-17:
@aha124226675 可以啊， Python是多平台的。我用的是Mac。

aha124226675, 2017-05-17:
恩恩我说的是XML那个网站的教学，是用LINUX吗？

hamuchiwa, 2017-05-19:
@aha124226675 你指的是那个教怎么训练haar-cascade的网站？他应该用的是Mac，Linux也一样。

aha124226675, 2017-06-13:
谢谢作者,我已成功完成分类器的xml识别。
另外想请问一下作者ROI 和高斯模糊应用能否额外做个教程呢?
想多多了解!
谢谢。

aha124226675, 2017-06-14:
另外想请问一下作者在DistanceToCamera中α角度需要更改吗？如不更改会有什么影响吗？

hamuchiwa, 2017-06-14:
@aha124226675 其实ROI谈不上什么教程了，红绿灯的框架就是ROI，然后高斯模糊去杂质，可用可不用，亮灯的时候对比对很高了，最亮的点很容易找到。
关于角度，那个是我自己拿量角器量的，因车而异吧，估计你也得重新测量一次。

aha124226675, 2017-06-15:
谢谢作者!另外想请问一下下列程式中
class DistanceToCamera(object):

        # camera params
        self.alpha = 8.0 * math.pi / 180
        self.v0 = 119.865631204
        self.ay = 332.262498472

8.0是如何量测出来的?
 self.v0 = 119.865631204
 self.ay = 332.262498472
上两行又是如何测得的?

hamuchiwa, 2017-06-15:
那个是摄像头校准后得来的，我参考了一篇文献做的单目测距。如果你有条件的话，直接上双目测距吧，一个摄像头真的是不准。

aha124226675, 2017-06-16:
謝謝作者但是目前還是考慮使用單目，另外想請問相機校正用的picam_calibration.py 裡的棋盤圖像需要自己重新拍攝嗎?
還是可以使用作者提供的?

aha124226675, 2017-06-16:
此外想请教在图像储存成mumpy数组方面， 一定要从 120*320转成1* 38400这样有什么好处吗?
用于类神经网路输入层吗?

hamuchiwa, 2017-06-16:
@aha124226675 我是用 pi camera module 拍的那些格子，虽然我觉得直接用也行，连校准也省了，不过还是拍一些试试吧，也不敢保证每个摄像头都完全一样。
对于一般的神经网络，一行就是一个样本，每一列就是一个特征也就是一个像素， 这样也就形成一个矩阵了， 为以后的运算做准备。

aha124226675, 2017-06-19:
了解 但是一定要轉池 1*38400 才能進行處理嗎?

hamuchiwa, 2017-06-19:
@aha124226675 对的，一般/标准的， 或者说fully connected神经网络是这个样子，其实用其他机器学习的算法也是如此，比如说SVM， KNN等等。图像（grayscale）是二维的，所以实际上做的是从图像到特征向量的转换，一张图片转成一行（flatten）。
卷及神经网络不太一样，可以想象为直接在二维图片上滑动一个二维小窗口，不过最后要连一层或多层fully connected神经网络，所以还要做flatten。

aha124226675, 2017-06-20:
谢谢作者!
另外想请问一下作者opencv_haartraining和opencv_traincascade 在HAAR CLASSIFIER训练中是选用哪个呢?

aha124226675, 2017-06-20:
还想在请问一下作者

  ＃ 红绿灯
             其他：
                 roi = gray_image [y_pos + 10：y_pos + height-10，x_pos + 10：x_pos + width-10]
                 mask = cv2.GaussianBlur（roi，（25,25），0）
                 （minVal，maxVal，minLoc，maxLoc）= cv2.minMaxLoc（mask）
                
                 ＃检查指示灯是否亮起
                 如果maxVal - minVal> threshold：
                     cv2.circle（roi，maxLoc，5，（255，0，0），2）

如果想去除高斯模糊的话该怎么改程式呢？
抱歉我的OpenCV没学好

aha124226675, 2017-06-21:
请问一下作者的角度是 alpha = 8.0 * math.pi / 180
是怎么算出来的?怎么量的?能提供一下解法吗?

hamuchiwa, 2017-06-21:
@aha124226675 
1. 我走的是这个 opencv_traincascade，另外一个我不太清楚，会不会是版本不同？
2. 直接把mask换成roi就可以了
3. 我是参考[这篇文章](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=4&cad=rja&uact=8&ved=0ahUKEwiJn8XbiM_UAhULdD4KHX1dB58QFgg5MAM&url=http%3A%2F%2Fwww.paper.edu.cn%2Fdownload%2FdownPaper%2F200909-748&usg=AFQjCNHdov0nPmLtQHK0hiJb7q80U5twjA)做的，要拿量角器测一下。

aha124226675, 2017-06-22:
RE
1.了解我是用了opencv_haartraining，感觉产生出来的XML似乎没很容易侦测。
2.我尝试换成ROI，结果差异不大。
3.恩谢谢作者，我会详读的，请问作者当初量的角度是几度呢?

Q:想请问一下作者，有类神经网路的教学吗，训练这部分没有很懂。能细说明吗?
感激不尽。

hamuchiwa, 2017-06-22:
@aha124226675 那个stop sign图片我是网上搜的，最后效果还不错。量的是8度，python里面做tan运算的要先转成弧度。
教学资料其实很多，书我就不给你推荐了。列几个网上的资源：
[这个](http://www.holehouse.org/mlclass/08_Neural_Networks_Representation.html)是讲义，就是coursera那门课
[第二个](http://neuralnetworksanddeeplearning.com/index.html)呢挺全面，把一些很实用的都讲了
有位大神写了个[博客](http://colah.github.io)，图文并茂，有兴趣你也可以看看
最后, 现在比较流行的是卷积神经网络，TensorFlow的[文档](https://www.tensorflow.org/get_started/get_started)就不错

aha124226675, 2017-06-26:
好的!謝謝作者
可以請作者解釋一下這段是怎麼應用的嗎?
layer_sizes = np.int32([38400, 32, 4])
model = cv2.ANN_MLP()
model.create(layer_sizes)
criteria = (cv2.TERM_CRITERIA_COUNT | cv2.TERM_CRITERIA_EPS, 500, 0.0001)
criteria2 = (cv2.TERM_CRITERIA_COUNT, 100, 0.001)
params = dict(term_crit = criteria,
               train_method = cv2.ANN_MLP_TRAIN_PARAMS_BACKPROP,
               bp_dw_scale = 0.001,
               bp_moment_scale = 0.0 )

aha124226675, 2017-06-26:
能只解释作者使用的类神经方法吗?
大概了解是监督式学习，
在各层中的运算能大概解释一下吗?

hamuchiwa, 2017-06-26:
@aha124226675 
第一行定义结构，一层hidden layer， 32个neuron。 38400是输入的个数，4是输出（前后左右各一个）
二三行是创建神经网络
第四行和第五行选一即可，我当时都给保留了，下一行params是具体的参数了
OpenCV文档关于机器学习写的很泛，不如看一下[sklean](http://scikit-learn.org/stable/modules/neural_networks_supervised.html)的文档，解释的好一点，而且例子也给了

使用的话基本上先明确有多少输入和输出，然后自己选择有几层hidden layer和每层有多少neuron
具体到各层运算的话一言两语不太好解释清楚， 还是建议看一下上次跟你说的第二个链接，训练的目的是为了得到合适的参数（神经网络里面是weights）然后能最小化cost function（实际值跟运算得出的值相差小）
第一个链接Coursera那门课程，从linear regression，logistic regression然后到neural network，很出名的一门课，循序渐进，后者的理解都是建立在前者之上，有时间你也可以看一下

aha124226675, 2017-06-27:
谢谢作者!我会详读的感激不尽

aha124226675, 2017-06-27:
请问测试样本的预测准确率为85%，而训练样本返回的准确率为96% 是如何测量得的?

hamuchiwa, 2017-06-27:
@aha124226675 其实就是计算 准确率 =（正确 /（正确 + 错误））
因为要验证model好不好，最简单的分法是把数据分成train和test，用train训练算一个准确率，然后把训练完的model用在test上再算出一个准确率
还有其他的分法， 比如分成train，validation，test， 另外还有k-folds cross validation

aha124226675, 2017-06-28:
恩请问有提供测试程式吗?
在样本方面，如何从图片中判定属train还是test?
如何检视有label?

hamuchiwa, 2017-06-28:
@aha124226675 有的，我最近把训练和测试放到一个文件里了 mlp_training.py
收集完数据之后，每一张图片都对应一个label，然后随机的分成train和test
对的，测试要用新图片（训练里没用到的），准确率就是拿算出来的label和已有的label比对

aha124226675, 2017-06-28:
Loading training data...
Image array shape:  (40L, 38400L)
Label array shape:  (40L, 4L)
Loading image duration: 3.66419359961
Training MLP ...
Training duration: 45.6526259974
Train accuracy:  82.14%
Test accuracy:  58.33%

这是我的数值，这样的模型算好吗?我的车可以很顺利地不出界。
其实我还是有点模糊，test用的是新图片测试吧，那怎么会有label呢?
有label的图片不已经都被拿去训练了?

aha124226675, 2017-06-28:
另外我看collect_training_data.py 最近也有更新，
我是使用旧的npz档来跑mlp_training.py
这样合适吗?
还是需要重新训练?

hamuchiwa, 2017-06-28:
@aha124226675 是这个样子，数据只收集一次（图片+label）然后分成train和test，所谓新图片只要是训练里没有用过的都算。最后predict的时候是没有label只有图片

如果跑道不变，你只看train就可以了，如果是新跑道的话我估计会跑出界吧，test准确率不高呀（随便才一次是25%）。我看你只用了40张图片，一般来讲数据越多越好，你可以试试多用些图片会怎么样

之前数据可以用，我只是改了一下文件保存路径相关的内容，每收集一次数据，用当前timestamp作为文件名，不用像以前那样一次一次的改了。。。

aha124226675, 2017-06-29:
Test accuracy: 58.33% 
这么低会不会是我 image资料夹里 有过多没有label的图片，
老实说在下label前，我的图片总是凌乱的，有些几乎没有跑道的画面。

aha124226675, 2017-06-29:
请问一下验证方法是交叉验证(cross-validation)吗?

hamuchiwa, 2017-06-29:
@aha124226675 
我记得看过你那个视频，效果真的很好。有可能是数据少所以结果底，其实跑道不变用不用test都无所谓。
我只分了train和test，最简单的分发，交叉验证的话要有validation. 
理解没有问题啦，OpenCV里面是标准的神经网络 （feed-forward）
图像最终都转成灰阶，只是显示的时候彩色。其实我还是觉得整张图片效果会好一些

hamuchiwa, 2017-06-29:
@aha124226675 其实训练时候都是用灰阶图片，识别也只是识别红绿灯物体的轮廓，并不识别是红灯还是绿灯。我其实是偷了个懒走了个捷径，先识别物体然后再根据相对位置，哪儿亮就是哪个灯

aha124226675, 2017-06-30:
1.了解!但在STOP标志部分是采用黑白的正样本训练，若但实际rc.driver侦测是全彩的这样不会有影响吗?

2.另外如果想改成侦测红绿灯颜色，而不用亮点侦测，使用ROI框住外框后再侦测R能否?

3.如果使用了上半部全彩输入侦测红绿灯颜色的方法，则下半部输入类神经model的图片还是得转成灰阶，ROI能一半输入全彩处理，下半部转灰阶再输入MLP model吗?

aha124226675, 2017-06-30:
[http://images2015.cnblogs.com/blog/606573/201704/606573-20170420110336806-921930691.png](url)

hamuchiwa, 2017-06-30:
@aha124226675 
1. 样本都是黑白的吧，而且图片越大训练时间越长，我记得stop sign用的是25x25，红绿灯是25x45
2. 我当时没想那么多，你考虑的比我周到，不如试试看吧说不定效果更好
3. 我觉得可以呀，第一步还是要先找到红绿灯，然后再看看什么颜色，光照条件可能会影响结果吧
4. 不影响，还是灰阶识别红绿灯轮廓，找到框框以后再从回到彩色图片找是红灯还是绿灯，神经网络不变

对的，可能我理解有误要么就是各家解释略有不同，我是觉得只分train和test应该叫train/test split，分train，validation， test叫做交叉验证，还有k-folds cross validation比较常用
比例的话没有一个固定值吧，6:4, 3:1, 7:3, 8:2 都有用的

aha124226675, 2017-06-30:
但在您的[介绍]中，红绿灯的正样本是为彩色的(https://zhengludwig.wordpress.com/projects/self-driving-rc-car/)

謝謝

hamuchiwa, 2017-06-30:
@aha124226675 
恩。。。上传的时候传了彩色的。。。我记得训练的时候先转成灰阶的，haar features是基于灰阶图像，最后生成的vec文件也是灰阶的
我大体看了一下opencv的文档，没看到明确说样本必须是灰阶，说不定直接读取彩色也可以，opencv在训练前先自动转成灰阶的

我是7：3分，如果你想更进一步的话，建议直接使用k-folds cross validation

aha124226675, 2017-07-01:
了解!感谢作者。
1.敢问当初作者正样本大小都是一样的吗?我看教学中裁减大小似乎都不一。输出才统一为25*25吧，
2.另外想请问一下作者使用点亮的红灯&绿灯当样本有什么特别用意吗?

pkletsko, 2017-07-02:
Hi , 
I have a speed problem , and have no idea how to overcome it , I have translated that you are talking about implement (add car potentiometer) , Can you explain what should I do.

hamuchiwa, 2017-07-03:
@aha124226675 输出的时候会resize一下，我当时用的是一样的图片，我是觉得既然都要自己拍图片了就不如一开始就保持大小一致，你说呢
红绿灯当时真没多想，其实我连黄灯也拍了。你想的比我周到，说不定真的不需要考虑灯的因素

hamuchiwa, 2017-07-03:
@pkletsko You can add a potentiometer in between the power and motor on your rc car, just like the way of adding a resistor.

pkletsko, 2017-07-03:
@hamuchiwa
I am week in electric engineering field , it would be nice to see a picture how did you implement it , in case it is possible of course.

pkletsko, 2017-07-04:
I works not well with potentiometer B50K - this is the one I have so far. Can you check/recall which type of potentiometer do you use for the project? Thanks a lot.

hamuchiwa, 2017-07-05:
@pkletsko Sorry, I don't have that anymore. My car had been disassembled a long time ago. 50K may be too large, a pot basically is a adjustable resistor, if you have resistors on your disposal, you could try out different combinations see which one works best.

pkletsko, 2017-07-05:
@hamuchiwa got it , I also think that it is to large, so I ordered smaller  , but have to wait for delivery.  Did you create a new car project after that one ?

aha124226675, 2017-07-07:
作者您好:
这几天我尝试改了隐藏层的节点个数，
比较了 64 32 16
发现还是32个准确率较高。

另外我改变过跑道来实际测试我的model发现效果还不错，虽然在Test accuracy: 58.33% 很低，但是效果不错呢!

另外想请问一下，能否计算出从图片传送到PC处理到发送指令给arduino之间的延迟大概是几秒呢?

[10 more replies]