hamuchiwa/AutoRCCar #7: hi i have a question about this! :)

kimseonho, 2016-11-01:
Hi! I wanna really make this RC car by myself so now I'm doing my best for self satisfaction, everything is fine including setting hardware but I have a problem with installing Opencv. now my laptop is Mac and I installed Opencv using Pycharm project interpreter. but  the problem is that when I'm trying to running 'picam_calibration.py', there is error like this 'OpenCV Error: Unspecified error (The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Carbon support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script) in cvShowImage, file /Users/travis/build/skvark/opencv-'python/opencv/modules/highgui/src/window.cpp, line 545" so, I wanna know how do you install opencv in mac. I tried to search google and try it, but I couldn't . 
sincerely  thank you and have a nice day!

hamuchiwa, 2016-11-01:
I followed this post when I installed OpenCV on mac, hope this works for you. 
https://jjyap.wordpress.com/2014/05/24/installing-opencv-2-4-9-on-mac-osx-with-python-support/

kimseonho, 2016-11-02:
oh,,, I finally did it!!! very thank you for your reply!!
have a good day!!

kimseonho, 2016-11-03:
Hi, I meet another problem but I can't solve this by self anymore:(  
in step7(last step), I mean that run 'rc_driver.py'  in computer and run 'stream_client.py' and 'ultrasonic_client.py' in raspberry pi. the RCcar is moving well and the video scene show in my monitor.  so connect works well between computer and raspberry pi. but when ultrasonic sensor sense below 30, the car doesn't stop and even 'Stop, obstacle in front' message doesn't show up. just distance message show up continuously. I guess that it doesn't run in class VideoStreamHandler in rc_driver.py. and I think that I maybe do run these three code wrong. when I run these three codes('rc_driver.py','stream_client.py',and 'ultrasonic_client.py', first, I run 'rc_driver.py' in computer in Pycharm. and then I run 'stream_client.py' on spe in raspberry pi. and I stop 'stream_client.py' code and then I run  'ultrasonic_client.py' code on spe in raspberry pi.  I think these two codes in raspberry pi should run simultaneously but  I don't know how to do that. anyway, could you teach me what's my problem and what I have to do?  
thank you very much and have a nice day :)

Arruda23, 2017-08-04:
@kimseonho what did you do? I'm with the same error and I don't see anything wrong with my set up. What did you do to make you project work?

closed 2018-06-29 by hamuchiwa