hamuchiwa/AutoRCCar #8: Hi, I meet another problem but I can't solve this by self anymore:(

kimseonho, 2016-11-04:
Hi, I meet another problem but I can't solve this by self anymore:(

in step7(last step), I mean that run 'rc_driver.py' in computer and run 'stream_client.py' and 'ultrasonic_client.py' in raspberry pi. the RCcar is moving well and the video scene show in my monitor. so connect works well between computer and raspberry pi. but when ultrasonic sensor sense below 30, the car doesn't stop and even 'Stop, obstacle in front' message doesn't show up. just distance message show up continuously. I guess that it doesn't run in class VideoStreamHandler in rc_driver.py. and I think that I maybe do run these three code wrong. when I run these three codes('rc_driver.py','stream_client.py',and 'ultrasonic_client.py', first, I run 'rc_driver.py' in computer in Pycharm. and then I run 'stream_client.py' on spe in raspberry pi. and I stop 'stream_client.py' code and then I run 'ultrasonic_client.py' code on spe in raspberry pi. I think these two codes in raspberry pi should run simultaneously but I don't know how to do that. anyway, could you teach me what's my problem and what I have to do?

thank you very much and have a nice day :)

hamuchiwa, 2016-11-04:
You can run "stream_client.py" in spe and execute "ultrasonic_client.py" in command line.

varun333, 2016-11-06:
hi, it's like i have just completed step 1 and that also using Windows 8 OS. Now i wanna ask whether rasp pi is compatible or not with windows and another thing I wanna ask according to u which rasp pi model should i go with for better result and also what are the power sources to make rasp pi function as in term to generate sufficient voltage

hamuchiwa, 2016-11-07:
varun333: windows works fine. I do believe the newest model of Raspberry Pi could make your life easier. Back then I used a model B+, I had to shrink the resolution to 320x240 to achieve "low-latency" streaming. For the power supply, any USB power bank should work.

varun333, 2016-11-09:
Thanks Hamuchiwa 

One more thing I want to ask, whether  the code and step above mentioned will work same or will require some changes. And also which pi camera should i go with 1.3 or 5MP. It will be so great of you if u can provide some of your your time and assistance through this period of time.

hamuchiwa, 2016-11-09:
Hi varun333, the code is for Python 2.7, Python 3 may not work. I think you can go with the newest camera module v2, it's supposed to work.

varun333, 2016-11-19:
Hi HAMUCHIWA, can i have the DATA FLOW DIAGRAM of this project.

hamuchiwa, 2016-11-21:
Hi varun333, I didn't make a data flow diagram for this project. 
Simply, it's just: grayscale image - numpy array - neural network - predictions

varun333, 2017-01-16:
HNY HAMUCHIWA, just purchased the raspi 3 kit n v2 module camera but few questions to ask:
1. had this 16 GB sd card that has already installed noobs,so how I install window OS on it,any site or software.
2. had no PC so trying to simulate this project using laptop installed with windows 10, will it work fine and carry the processing of both arduino and rasp pi.

hamuchiwa, 2017-01-17:
varun333: for your questions, 1 - If my memory is correct, the noobs os is already installed. If not, you can always download the os from raspberry pi official site. 2 - This is something that I haven't done yet, feel free to try it out. A lot of people have expressed their interests in this :)

closed 2018-03-19 by hamuchiwa