hamuchiwa/AutoRCCar #9: Few questions about server and white lines and camera

OwhPUS, 2016-11-09:
Hello,

I kinda want to make the same project, so I have a few questions about it:
About the wifi-connection you use between the raspberry pi and your PC: Do you make from your raspberry pi an accespoint and then connect to it via your computer?

And is it possible to do it without a server? So just a raspberry pi for camera and sensor and an arduino to control the motors. Or don't you recommend this?

I can't find the code where you analyse the stream and say to your car to stay between the white lines.  

Do you need to calibrate your camera everytime you start up your picamera? 

Thanks in advance! 

Regards,
OwhPUS

hamuchiwa, 2016-11-09:
Hi, please see the following:
1. It's a client to server communication within local network
2. I would say it's possible, a few people said they are going to implement this
3. It's neural network, train and predict. The prediction steers the car.
4. No, only once. If you switch to a different camera, you need to do calibration again.

OwhPUS, 2016-11-09:
Thanks for the fast reply!
2. Is it more difficult without server? Because I actually don't know anything about setting up a server on computer etc.. So it would save me a lot of time. But maybe it will lose me more time later in the project. What do you think?

3.  I don't really understand it. Isn't OpenCV just recognising the white lines? And don't you need to program your code so your program can send the correct commands (like left/right) to your arduino?
This is the first step I want to deceive in my project. But i'm a little bit stuck here. 

Regards,
OwhPUS

hamuchiwa, 2016-11-09:
Hi OwhPUS,

Networking programming in python is not as complex as you think. If you google it you will see a lot of socket programming examples. Personally I think it's easier and faster to do the coding on computer, raspberry pi is slow... but that doesn't mean you have to use a computer. 

It's a neural network model. First you train the neural network with images that you labeled. For example, you steer the rc car on the track as best as you can, and pair each image with a label you entered (moving direction, keypress). Once the neural network is trained, it makes predictions, which are the labels you defined in the first place, based on incoming streaming images.

OwhPUS, 2016-11-09:
Hi again,

Raspberry pi is indeed pretty slow. But don't you lose your time by sending the data over the wifi. Then calculate and sending the data back over the wifi? Or is it still faster? 

And oke I think I understand now. 
So I need to take a lot of pictures from the circuit when my car is on it. Then I label every image with a command. 
And so the raspberry pi/PC will compare the incoming streaming images with the images that I took. And the image that compares best, he will use that images label (command). Or am I still wrong?

Regards,
OwhPUS

hamuchiwa, 2016-11-09:
Hi OwhPUS,

I haven't tried to run everything on raspberry pi, so it's hard to say. But theoretically if raspberry pi is capable to run neural network, the overhead (wifi transmission) can be eliminated.   

Your interpretation is "correct", neural network is a supervised machine learning algorithm. You can consider it as a classification problem, given a input, output a prediction. To understand the process better, I would suggest you refer to some general machine learning tutorials, also I think the OpenCV python tutorial is helpful too.

OwhPUS, 2016-11-10:
Hi again,

Oke thank you. Do you have some links to the tutorials please? 

Regards,
OwhPUS

hamuchiwa, 2016-11-10:
Hi, there are many of them, you can take a look at these two. Also scikit-learn documentation is a nice one too. 

Python machine learning book:
https://github.com/rasbt/python-machine-learning-book

Machine learning open course on Coursera : 
https://www.coursera.org/learn/machine-learning

Dalius1994, 2016-12-27:
Hello maybe can share on windows server source code, i can't  write correct code which can work fine.