hamuchiwa/AutoRCCar #10: Hi, I have a question about how to drive this car in“collect_training_data.py” part:

kimseonho, 2016-11-14:
Hi, I always appreciate for you answer! 
Now, I don't have problem in coding and server. everything works well.
but when I run 'rc_driver.py', the car runs only left and straight even if I set up the car in "collect_tarining_data.py" and "stream_client.py" along the path that I set up totally.
so I suddenly wonder if maybe I train this car wrong method. 
now, I just press key in path in 'rc_driver.py' expecting that this car will act same in the same location in 'rc_driver.py' 
How can I solve this problem?

hamuchiwa, 2016-11-14:
What are the accuracies on training set and testing set? If you label each image correct, you could try to adjust parameters of the neural network to see if that can improve your accuracy. 

** It's possible to "cheat" a bit while collecting the train images: placing the rc car on the track where you believe it's reasonable, and then press key to label that image.

kimseonho, 2016-11-15:
Thank you, you mean that I have to make accuracy of image in training set and testing set higher?
and I wonder if the shaking problem when the car capture image effect my problem.

hamuchiwa, 2016-11-16:
Hi, the accuracy is one way to evaluate the performance of the neural network. The number you mentioned is not accuracy, it's the size of input layer. 

You can comment out the line that saves the image, this doesn't affect the final numpy array.

To reduce the speed, a quick way is to add a potentiometer, or you could buy a rc car has speed adjustments :)

kimseonho, 2016-11-17:
Thank you, but I'm not sure what the accuracy means.  
now, the car runs well roughly but I know that the more the car is trained, the more accuracy It has.
but I tried this in various way and I think that the more npz file I make, the better the car runs.
but actually, It doesn't. so For better working, should I train this car many times in one collecting_traininig_data?  thank you:)

hamuchiwa, 2016-11-17:
Hi kimseonho,

accuracy = # of correct labels/(# of correct labels + # of incorrect labels)
It's just one type of measurement, there are lots of other matrix there too. 
The quality of training data is very important. Make sure the image frame is filled with track, and each image is clear. Also as I mentioned before, you could also adjust the number of neurons in the hidden layer of the neural network to see if it improves the result. In general, I would say one hidden layer for this project is sufficient enough.

kimseonho, 2016-11-17:
Thank you very much! :))
I can understand absolutely! 
and maybe do you know how to use potentiometer in RC car? now I'm using rubber band combing the wheel of RC car to remove speed so that It's effective but unstable. so I'm searching potential meter in RC Car but I can't be willing to use it because of hardware.

hamuchiwa, 2016-11-18:
It's the same as adding a resistor to the motor . The downside of using potentiometer is that it cuts the voltage which reduce the torque. I totally agree with you rubber band works very effectively, it does the trick :)

Dalius1994, 2016-12-31:
Hello, i try to start code collect_training_data.py but then i do always show few error and always one of problem are  CollectTrainingData()

Dalius1994, 2017-01-08:
what need do? When i start collect_training_data.py and start stream_client, always show broken pipe error

isrugeek, 2017-01-26:
the connection between your Computer and Raspi is failed 
Solution: Check local IP, Router Status, and be sure about the stream_client.pi as well as server sequence .

it my help

closed 2018-03-19 by hamuchiwa