hamuchiwa/AutoRCCar #12: Hi, I want to know how I can use traffic signal!

kimseonho, 2016-11-29:
Hi, I bought the traffic signal on internet. 
this consists of basic LED light (red, yellow, green)
but the camera can't recognize these lights. 
this sometimes recognize red light. 
I saw your videos many times and I guess that the problem is light spread but not exactly.
I want to buy same model you used but It's difficult to find it and get it.
maybe It's my last question.
How can I solve this problem?
[image]  this is my traffic light !
Thank you :)
Merry christmas :))

hamuchiwa, 2016-11-30:
The classifier included in the folder is for the traffic light I used. You need to retrain the classifier for the object you want to detect. For the stop sign, on the other hand, I trained that classifier with actual stop sign images, that should work in a generic way.

kimseonho, 2016-12-02:
Thank you for your helpful answer!
and I want to try to make classifier file and I've read the reference site you wrote.
and I wonder if how can I note the name of traffic light like red, green in the classfier?
the reference site only teaches me how I can train the picture and make a file.
Danke!

tall49er, 2016-12-05:
You could threshold images using RGB or HSV model. 
May it help~

hamuchiwa, 2016-12-07:
First you need to locate the traffic light region. I took a short cut, light is based on the position. If you do thresholding, then keep a copy of color images when receiving the stream.

isrugeek, 2017-01-26:
First Train your classifier to know your Trafic light ( the Hardware ) then  use thresholding when the Hardware Detects 
Thanks

closed 2018-03-19 by hamuchiwa