hamuchiwa/AutoRCCar #14: Problem in "stream_server_test.py"

geek1111, 2016-12-17:
After running "stream_server_test.py" in python 2.7.9 , it shows following errors 
is there problem in my python packages ? or which python version should i install to run this code?
`"/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)`

this errors shows when i try to run 
`/AutoCar/test$ python stream_server_test.py
Traceback (most recent call last):
  File "stream_server_test.py", line 44, in <module>
    VideoStreamingTest()
  File "stream_server_test.py", line 12, in __init__
    self.server_socket.bind(('192.168.1.100', 8000))
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 99] Cannot assign requested address
`

hamuchiwa, 2016-12-19:
What's your local ip address? Is there another application listening on that port?

geek1111, 2016-12-21:
Its working after changing local IP address in a code, now its working fine , Thank you very much 
Merry Christmas & Happy New Year in Advance

hamuchiwa, 2016-12-21:
The same to you, Merry Christmas and Happy New Year :)

varun333, 2017-03-04:
hamuchiwa
Hey there I just wanna ask that I am having problem running server_client.py  code, so for this reason I am not able to stream any kind of image on my PC. And just also want to know about the configurtion related Pycharm IDE.
Reply soon it will be of great help.

KhyatiMehta3, 2018-03-02:
Hi, I'm getting this error: 

On computer:

Start collecting images...
Traceback (most recent call last):
  File "collect_training_data.py", line 174, in <module>
    CollectTrainingData()
  File "collect_training_data.py", line 36, in __init__
    self.collect_image()
  File "collect_training_data.py", line 55, in collect_image
    stream_bytes += self.connection.read(1024)
TypeError: must be str, not bytes

On RaspberryPi 3:

Traceback (most srecent call last):
File "stream_client.py", line50, in <module>
connection.close()
File "/usr/lib/python2.7/socket.py", line 283, in close
self.flush()
File "/usr/lib/python2.7/socket.py",  line 307, in flush
self._sock.sendall(view[write_offset:write_offset+buffer_size])
socket.error: [Errno 32] Broken pipe

Can you help me solve it?

IConquer96, 2018-03-19:
Hi, I'm getting this error:

On computer:

Start collecting images...
Traceback (most recent call last):
File "collect_training_data.py", line 174, in 
CollectTrainingData()
File "collect_training_data.py", line 36, in init
self.collect_image()
File "collect_training_data.py", line 55, in collect_image
stream_bytes += self.connection.read(1024)
TypeError: must be str, not bytes

On RaspberryPi 3:

Traceback (most srecent call last):
File "stream_client.py", line50, in 
connection.close()
File "/usr/lib/python2.7/socket.py", line 283, in close
self.flush()
File "/usr/lib/python2.7/socket.py", line 307, in flush
self._sock.sendall(view[write_offset:write_offset+buffer_size])
socket.error: [Errno 32] Broken pipe

Can you help me solve it?

hamuchiwa, 2018-03-19:
@IConquer96 I would think it's something to do with difference between Python2 and Python3, what's the Python version on your computer?

Moha-alfridi, 2018-04-05:
Hi @hamuchiwa.
So, after modifying the IP and the port on "collect_training_data.py" to my computer local IP and port, and running it on my computer terminal using `python` command, it ran without errors, and was waiting for the data to be sent from the RPi, so I used SSH and ran "stream_client.py" using `python` command, after changing it's IP and port to my computer IP & port.

My computer's Terminal show's

`Start collecting images...` 
for couple of seconds .. 

then it print these errors 
`Traceback (most recent call last):
  File "collect_training_data.py", line 172, in <module>
    CollectTrainingData()
  File "collect_training_data.py", line 35, in __init__
    self.collect_image()
  File "collect_training_data.py", line 53, in collect_image
    stream_bytes += self.connection.read(1024)
TypeError: must be str, not bytes`

Line 72 on collect_training_data.py
`    CollectTrainingData()`

Line 35 on collect_training_data.py
`        self.collect_image()`

Line 53 on collect_training_data.py
`                stream_bytes += self.connection.read(1024)`

and on the ssh terminal "RPi" it shows 
````
Message from syslogd@raspberrypi at Apr  5 22:22:58 ...
 kernel:[   85.732372] Internal error: Oops: 17 [#1] SMP ARM

Message from syslogd@raspberrypi at Apr  5 22:22:58 ...
 kernel:[   85.737360] Process python (pid: 922, stack limit = 0xab2fe210)

Message from syslogd@raspberrypi at Apr  5 22:22:58 ...
 kernel:[   85.737366] Stack: (0xab2ffd98 to 0xab300000)

Message from syslogd@raspberrypi at Apr  5 22:22:58 ...
 kernel:[   85.737373] fd80:                                                       000005a8 806d1930

Message from syslogd@raspberrypi at Apr  5 22:22:58 ...
 kernel:[   85.737384] fda0: 000076c8 ab2ffec0 ab060640 000076c8 ab2ffe54 ab2ffdc0 806d1950 806ce34c

Message from syslogd@raspberrypi at Apr  5 22:22:58 ...
 kernel:[   85.737395] fdc0: 000005a8 ab2ffdd0 804aeab8 b4dbbc00 80c069c4 80d0a08c 80c049c0 ab2fe000

Message from syslogd@raspberrypi at Apr  5 22:22:58 ...
 kernel:[   85.737405] fde0: 00000000 00000000 00000000 00000000 00000001 00000000 ab2ffe24 00000200

Message from syslogd@raspberrypi at Apr  5 22:22:58 ...
 kernel:[   85.737416] fe00: ab2ffec8 ab060790 000005a8 00000000 00000121 000005a8 7fffffff 3c334492

Message from syslogd@raspberrypi at Apr  5 22:22:58 ...
 kernel:[   85.737427] fe20: 801235c4 801e0000 ab060640 ab060640 0000b53e ab2ffec0 00000121 80108224

Message from syslogd@raspberrypi at Apr  5 22:22:58 ...
 kernel:[   85.737438] fe40: ab2fe000 00000000 ab2ffe74 ab2ffe58 806d247c 806d1654 806d2444 ab060640

Message from syslogd@raspberrypi at Apr  5 22:22:58 ...
 kernel:[   85.737448] fe60: 00000000 a4988fc0 ab2ffe94 ab2ffe78 806fe570 806d2450 802aa2ac 802aa200

Message from syslogd@raspberrypi at Apr  5 22:22:58 ...
 kernel:[   85.737459] fe80: 00000000 00000000 ab2ffea4 ab2ffe98 80657c84 806fe534 ab2fff8c ab2ffea8

Message from syslogd@raspberrypi at Apr  5 22:22:58 ...
 kernel:[   85.737469] fea0: 80658dd4 80657c6c ab2ffec8 80601cd4 fffffff7 00000001 01dd6d70 0000b53e

Message from syslogd@raspberrypi at Apr  5 22:22:58 ...
 kernel:[   85.737480] fec0: 00000000 00000000 00000001 000076c8 00003e76 ab2ffeb8 00000001 801ee204

Message from syslogd@raspberrypi at Apr  5 22:22:58 ...
 kernel:[   85.737490] fee0: 00000000 00000000 00000000 ab2ffef8 80101850 80123188 ab2fff1c ab2fff08

Message from syslogd@raspberrypi at Apr  5 22:22:58 ...
 kernel:[   85.737501] ff00: 00000001 00400000 80c0469c 8017a1b0 ab2fff34 ab2fff20 8017a1b0 80179dec

Message from syslogd@raspberrypi at Apr  5 22:22:58 ...
 kernel:[   85.737511] ff20: 8017a08c 80c0469c ab2fff44 ab2fff38 80174f10 804ae66c ab2fff5c ab2fff48

Message from syslogd@raspberrypi at Apr  5 22:22:58 ...
 kernel:[   85.737522] ff40: 804ae66c 804ae590 804ae630 8012377c ab2fff6c ab2fff60 8012377c 801845dc

Message from syslogd@raspberrypi at Apr  5 22:22:58 ...
 kernel:[   85.737533] ff60: ab2fff94 ab2fff70 80175554 801236e0 ab2fffb0 7ea748b8 76a57780 7ea748b8

Message from syslogd@raspberrypi at Apr  5 22:22:58 ...
 kernel:[   85.737544] ff80: ab2fffa4 ab2fff90 80658e24 80658d18 00000000 00000000 00000000 ab2fffa8

Message from syslogd@raspberrypi at Apr  5 22:22:58 ...
 kernel:[   85.737555] ffa0: 80108080 80658e08 7ea748b8 76a57780 00000004 01dd6d70 0000b53e 00000000

Message from syslogd@raspberrypi at Apr  5 22:22:58 ...
 kernel:[   85.737565] ffc0: 7ea748b8 76a57780 7ea748b8 00000121 00000000 00000000 00000001 01c91ae8

Message from syslogd@raspberrypi at Apr  5 22:22:58 ...
 kernel:[   85.737576] ffe0: 00000000 7ea74850 00000000 76eb8984 80000010 00000004 aaaaaaaa aaaaaaaa

Message from syslogd@raspberrypi at Apr  5 22:22:58 ...
 kernel:[   85.737714] Code: e155000c 03a05000 e2117902 1a000039 (e5d50028)```

I appreciate your help, God bless u

closed 2018-06-29 by hamuchiwa