hamuchiwa/AutoRCCar #16: Questins about implementing this into a single board

csuporzoltan, 2017-01-18:
Hello,
So at my school, we want to build a simular robot. I wanted to make up the whole code from 0 in Java, but luckily i found your project.
So i've got a few questions about the title, but let me introduce my surrodings.

I've got the latest (3 B) Raspberry PI, and the v2 Pi Camera (8MP, if i remember right), with all the neccessary components (ultrasonic, etc). It came to my mind, that you're using Arduino, to control the RC Transmitter. Lucily, we havent got a transmitter, and we had to solder jumper wires to the actual IC in order control the car. So in that setup, we had the pi connected to ardunio via USB, running all the 3 programs at once to make up everything. So it came to my mind that we could avoid arduino because the Pi had an actual GPIO as the same as Arduino does. 

So i modified your code just a littlebit, adding the GPIO library to the rc_drive, and modify the serial send to smth like that: GPIO.output(GPIO_FORWARD, True).

Unfortunately, i couldnt test it till next Wednesday, but besides that, what do you think about this setup? Do you think that this could handle all the 3 programs at once? (+ a VNC server to see the camera pic)

hamuchiwa, 2017-01-19:
Actually a lot of people asked the same question, whether Raspberry Pi 3 is capable for handling all the tasks. Please let us know how the test goes. We are looking forward to knowing the result :)

geek1111, 2017-01-25:
hello @hamuchiwa , 
I have done some experiment on raspberry pi 3 B  
running *"rc_server.py"* & then *"stream_client.py"* on only one raspberry pi taking same as server(host) & client even connected arduino to same raspberry pi and arduino outputs( pins which stimulates rc controller left ,right ,forward , backward) connected to motor driver LM-L298N(Its connected to front motor and rear motor ) to eliminate wireless rc transmitter and receiver and tested, using *"rc_control_test.py"* and after taking remote access of Raspberry pi and tried to control car Whoa! its working fine, but there was an problem camera stream was too late upto 9sec. thats why car taking late actions (not real time) I'm using mobile hotspot to establish connection between server(host) & client, i think because of that its slow... im gonna try with router as well ,still I think raspberry pi 3 b can all tasks at same time So i was thinking to build stand alone model can you help me building code  for stand alone model of this car using only camera, raspberry pi 3 & motor driver & car ?

// please note that i have used only one raspberry pi as a host & client at that time (CPU usage 23 to 30 %)only
[block _diagram_test.docx](https://github.com/hamuchiwa/AutoRCCar/files/730641/block._diagram_test.docx)

hamuchiwa, 2017-01-25:
@geek1111 
I'm just thinking, what about connecting pi to the car directly this time.

isrugeek, 2017-01-26:
i think this can be Possible: but here i don't recommend giving all task for Raspi rather try to integrate Arduino with it so Arduino will control your Motor Driver , Arduino Must be Plugged into RasPI using Serial. So Machine learning and Neural Network will Run on your Raspi then Electro-Mechanical Decision will be handled By Arduino. Here is Sample Code for Arduino and little Modification In Raspi Code `void moveStop(int time) {
  analogWrite(motorPin1, 0);
    analogWrite(motorPin2, 0);
   // analogWrite(motorPin3, 0);
    //analogWrite(motorPin4, 0);
    delay(time);
  } 
  
void moveForward(int time) {

    analogWrite(motorPin1, 180);
    analogWrite(motorPin2, 0);
   // analogWrite(motorPin3, 180);
   // analogWrite(motorPin4, 0);  
   delay(time);
  
}

void moveReverse(int time) {
    
    analogWrite(motorPin1, 0);
    analogWrite(motorPin2, 180);
   // analogWrite(motorPin3, 0);
    //analogWrite(motorPin4, 180);   
  delay(time);
}  

void turnRight(int time) {
       
  frontservo.write(50);
  delay(time);
} 
 
void turnLeft(int time) {
  frontservo.write(170);
  delay(time);
}  ` 

this is Sample code for Arduino so wire your Motor Drive with Arduino .  But i am not Sure Machine Learning and Neural Network will be effective in Raspi ( how much time it may take to train) .

csuporzoltan, 2017-01-26:
Basically what i did to control the motors without arduino is smth like this:

`import RPi.GPIO as GPIO

GPIO_ELORE = 10
GPIO_HATRA = 13
GPIO_BALRA = 11
GPIO_JOBBRA = 12
GPIO_LED = 13

class RCControl(object):

    def __init__(self):
        GPIO.setwarnings(False)
        GPIO.setmode(GPIO.BCM)
        GPIO.setup(GPIO_ELORE,GPIO.OUT)
        GPIO.setup(GPIO_HATRA,GPIO.OUT)
        GPIO.setup(GPIO_BALRA,GPIO.OUT)
        GPIO.setup(GPIO_JOBBRA,GPIO.OUT)
        GPIO.setup(GPIO_LED,GPIO.OUT)
        GPIO.output(GPIO_ELORE, False)
        GPIO.output(GPIO_BALRA, False)
        GPIO.output(GPIO_JOBBRA, False)
        GPIO.output(GPIO_LED, False)
        GPIO.output(GPIO_HATRA, False)
    def steer(self, prediction):
        if prediction == 2:
            self.stop()
            GPIO.output(GPIO_ELORE, True)
            print("Elore")
        elif prediction == 0:
            self.stop()
            GPIO.output(GPIO_JOBBRA, True)
            print("Jobbra")
        elif prediction == 1:
            self.stop()
            GPIO.output(GPIO_BALRA, True)
            print("Balra")
        else:
            self.stop()

    def stop(self):
        GPIO.output(GPIO_ELORE, False)
        GPIO.output(GPIO_HATRA, False)
        GPIO.output(GPIO_BALRA, False)
        GPIO.output(GPIO_JOBBRA, False)
`

I honestly do not know if it works, because i havent got any time to test it, but it SHOULD works

geek1111, 2017-01-30:
hello everyone i am interested in implementing this same project on single board(only raspberry pi) or using raspberry pi & arduino and eliminating "client-server camera stream" part and replacing it with direct camera access to opencv to process frames, anybody can help me in this case what to do?
there is similar repository [here](https://github.com/multunus/autonomous-rc-car)
thank you!

isrugeek, 2017-02-12:
@geek1111 i saw the repository but they make everything complicated! a bunch of codes ! i think you can implement my code to your in a single raspberry pi!

geek1111, 2017-02-12:
Hi @isrugeek can you send me link of your work ?

isrugeek, 2017-02-12:
@geek1111 Sure i am adding some following features on it so i didn't upload it here but i will send you the test part code ! but when you came to my Project i used one DC motor and Servo as steer. so you need to check the servo mode versus your appropriate prototype . 

````
import sys
import time
import RPi.GPIO as GPIO
import pygame
from pygame.locals import *

pygame.display.set_mode((100, 100))

GPIO.setmode(GPIO.BOARD)

GPIO.setwarnings(False)

StepPinForward=7
StepPinBackward=11

GPIO.setup(12,GPIO.OUT)
GPIO.setup(StepPinForward, GPIO.OUT)
GPIO.setup(StepPinBackward, GPIO.OUT)

p= GPIO.PWM(12,50)

p.start(5.5)

command = 'Stop'

# command is used for OpenCv 

def interactive_control():
    """Runs the interactive control"""
    #setup_interactive_control()
    #clock = pygame.time.Clock()
    pygame.init()
    command = 'idle'
    while True:
        for event in pygame.event.get():
            if event.type == KEYDOWN:
                key_input = pygame.key.get_pressed()

                # complex orders
                if key_input[pygame.K_UP] and key_input[pygame.K_RIGHT]:
                    GPIO.output(StepPinForward, True)
                    p.ChangeDutyCycle(8.5)

                    print("coming and  Right")

                    # for OpenCV
                    command = 'forward right'

                elif key_input[pygame.K_UP] and key_input[pygame.K_LEFT]:
                    GPIO.output(StepPinForward, True)
                    p.ChangeDutyCycle(3)

                    print("Forward Left")

                elif key_input[pygame.K_DOWN] and key_input[pygame.K_RIGHT]:

                    GPIO.output(StepPinBackward, True)
                    p.ChangeDutyCycle(8.5)

                    print("going and turn right")

                elif key_input[pygame.K_DOWN] and key_input[pygame.K_LEFT]:
                    GPIO.output(StepPinBackward, True)
                    p.ChangeDutyCycle(3)

                    print("back and Left")

                # simple orders
                elif key_input[pygame.K_UP]:
                    GPIO.output(StepPinForward, True)
                    p.ChangeDutyCycle(6.5)
                    print "forwarding  "
                    

                elif key_input[pygame.K_DOWN]:
                    GPIO.output(StepPinBackward, True)
                    p.ChangeDutyCycle(6.5)
                    print "backwarding running motor"

                elif key_input[pygame.K_RIGHT]:
                    p.ChangeDutyCycle(8.5)
                    print "Right Turn"

                elif key_input[pygame.K_LEFT]:
                    p.ChangeDutyCycle(3)
                    print("Left")

                # exit
                elif key_input[pygame.K_x] or key_input[pygame.K_q]:

                    print 'Exit'
                    sys.exit()
                    GPIO.cleanup()

                    break

            elif event.type == pygame.KEYUP:
                GPIO.output(StepPinForward, False)
                GPIO.output(StepPinBackward,False)
                p.ChangeDutyCycle(6.5)
                

                print "Corrected"

def main():
    """Main function"""

    interactive_control()

if __name__ == '__main__':
    main()

def forward():
    GPIO.output(StepPinForward, GPIO.HIGH)
    p.ChangeDutyCycle(5.5)
    print "forwarding running  motor "
    GPIO.output(StepPinForward, GPIO.LOW)

def reverse():
    GPIO.output(StepPinBackward, GPIO.HIGH)
    p.ChangeDutyCycle(5.5)
    print "backwarding running motor"
    GPIO.output(StepPinBackward, GPIO.LOW)

def right():
    p.ChangeDutyCycle(7.5)
    print "Right Turn"

def left():
    p.ChangeDutyCycle(2.5)
    print "LEft Turn"
`
```

isrugeek, 2017-02-12:
[image]
[image]

deepaksrini01, 2017-02-28:
can anyone please explain the process of running open CV NN and image processing.

aleonnet, 2017-05-21:
I just did it! A stand alone version using a Raspberry Pi 3, node-red, opencv neural network just like @hamuchiwa.
Also I'm not using Arduino. Instead, I'm controlling the engines and servos using node-red with both manual input and/or selfdriving inputs from python via websockets.
I was able to get a relative low delay from camera to steering by using raw mode (without jpeg encoding/decoding) and of course removing the socket connections between the notebook and Pi.

csuporzoltan, 2017-05-21:
Can you share the code? I'm corious to see what have you done

aleonnet <notifications@github.com> ezt írta (időpont: 2017. máj. 21., V
5:18):

> I just did it! A stand alone version using a Raspberry Pi 3, node-red,
> opencv neural network just like @hamuchiwa <https://github.com/hamuchiwa>.
> Also I'm not using Arduino. Instead, I'm controlling [...]
-- 

Üdvözlettel/Regards,

Csupor Zoltán

hamuchiwa, 2017-05-21:
@aleonnet Looks great, already upvoted. Thanks for sharing.

aleonnet, 2017-05-21:
Here is the code:

__author__ = 'zhengwang mod by alessandrobarbosa'

import threading
import SocketServer
import serial
import cv2
import numpy as np
import math

import websocket
import thread
import time
import json

import io
import picamera
from picamera.array import PiRGBArray

# distance data measured by ultrasonic sensor
sensor_data = " "

class NeuralNetwork(object):

    def __init__(self):
        self.model = cv2.ANN_MLP()

    def create(self):
        layer_size = np.int32([38400, 32, 4])
        self.model.create(layer_size)
        self.model.load('mlp_xml/mlp.xml')

    def predict(self, samples):
        ret, resp = self.model.predict(samples)
        return resp.argmax(-1)

class RCControl(object):

    def __init__(self):
        print 'Steer ready...'
        # connect to a websocket
        websocket.enableTrace(True)
        self.ws = websocket.WebSocketApp("ws://192.168.2.49:1880/ws/vr",
                                on_message = self.on_message,
                                on_error = self.on_error,
                                on_close = self.on_close)
        wst = threading.Thread(target=self.ws.run_forever)
        wst.daemon = True
        wst.start()

    def on_message(self, ws, message):
        payload = json.loads(message)
        if payload['topic']=="manual":
            print payload['movement'] + payload ['direction']

    def on_error(self, ws, error):
        print error

    def on_close(self, ws):
        print "### closed ###"
    
    def on_open(self, ws):
        print 'socket opened...'
    
    def sendit(self, movement, direction):
        if movement=="Forward":
            dy1 = "-40"
        if movement=="Reverse":
            dy1 = "40"
[... 244 lines cut ...]
                        break

            cv2.destroyAllWindows()

        finally:
            print "Video stream ended"

#class ThreadServer(object):

#    def server_thread(host, port):
#        server = SocketServer.TCPServer((host, port), VideoStreamHandler)
#        server.serve_forever()

#    def server_thread2(host, port):
#        server = SocketServer.TCPServer((host, port), SensorDataHandler)
#        server.serve_forever()

#    distance_thread = threading.Thread(target=server_thread2, args=('192.168.2.153', 8002))
#    distance_thread.start()
#    video_thread = threading.Thread(target=server_thread('192.168.2.153', 8000))
#    video_thread.start()

if __name__ == '__main__':
    VideoStreamHandler()
#    ThreadServer()

aleonnet, 2017-05-21:
You can check the code here:
https://github.com/aleonnet/AutoRCCar-1/blob/aleonnet-stand-alone/raspberryPi/rc_driver_ws_sa2.py

lakshand8, 2018-01-15:
@geek1111 How did you manage to run "rc_server.py" & "stream_client.py" on the same pi acting as both the client and the host?

aleonnet, 2018-01-16:
It’s in my previous post.
 I was able to get a relative low delay from camera to steering by using raw
mode (without jpeg encoding/decoding) and of course removing the
socket connections
between the notebook and Pi.

lakshand8, 2018-01-16:
Thanks for the quick reply @aleonnet .  I'm using a raspberry pi motorHAT from adafruit to drive my motors so I'm not using any arduino. So this brings up the issue in 'collect_training_data.py file as it sends a command to the arduino serial whenever you press the control keys. What I'm trying to do is stream the training data to the pc but control the car from the Pi itself. I'm not familiar with node-red so I'm not sure how you implemented this?  Appreciate your help!

Here's my code for a better understanding:
__author__ = 'zhengwang'

from Adafruit_MotorHAT import Adafruit_MotorHAT, Adafruit_DCMotor
import RPi.GPIO as GPIO
import numpy as np
import cv2
import serial
import pygame
from pygame.locals import *
import socket
import time
import os
import atexit

mh = Adafruit_MotorHAT(addr=0x60)

def turnOffMotors():
	mh.getMotor(1).run(Adafruit_MotorHAT.RELEASE)
	mh.getMotor(2).run(Adafruit_MotorHAT.RELEASE)
	mh.getMotor(3).run(Adafruit_MotorHAT.RELEASE)
	mh.getMotor(4).run(Adafruit_MotorHAT.RELEASE)

atexit.register(turnOffMotors)

rearMotor = mh.getMotor(1)
frontMotor = mh.getMotor(2)

rearMotor.setSpeed(250)
frontMotor.setSpeed(100)

class CollectTrainingData(object):
    
    def __init__(self):
        pygame.init()
        pygame.display.set_mode((100, 100), HWSURFACE | DOUBLEBUF | RESIZABLE)
        self.server_socket = socket.socket()
        self.server_socket.bind(('192.168.0.227', 8090))
        self.server_socket.listen(0)

        # accept a single connection
        self.connection = self.server_socket.accept()[0].makefile('rb')

        # connect to a seral port
        #self.ser = serial.Serial('/dev/tty.usbmodem1421', 115200, timeout=1)
        self.send_inst = True

        # create labels
        self.k = np.zeros((4, 4), 'float')
        for i in range(4):
            self.k[i, i] = 1
        self.temp_label = np.zeros((1, 4), 'float')

        self.collect_image()

    def collect_image(self):

        saved_frame = 0
        total_frame = 0

        # collect images for training
        print 'Start collecting images...'
        e1 = cv2.getTickCount()
        image_array = np.zeros((1, 38400))
        label_array = np.zeros((1, 4), 'float')

        # stream video frames one by one
        try:
[... 102 lines cut ...]
            directory = "training_data"
            if not os.path.exists(directory):
                os.makedirs(directory)
            try:    
                np.savez(directory + '/' + file_name + '.npz', train=train, train_labels=train_labels)
            except IOError as e:
                print(e)

            e2 = cv2.getTickCount()
            # calculate streaming duration
            time0 = (e2 - e1) / cv2.getTickFrequency()
            print 'Streaming duration:', time0

            print(train.shape)
            print(train_labels.shape)
            print 'Total frame:', total_frame
            print 'Saved frame:', saved_frame
            print 'Dropped frame', total_frame - saved_frame

        finally:
            self.connection.close()
            self.server_socket.close()

if __name__ == '__main__':
    CollectTrainingData()

aleonnet, 2018-01-17:
This is an old project, but in short I've used node-red as a backend to
control PCA9685 servo board via websockets. So, this way I can send JSON
formated commands (steering, stop, run) and use whichever controller I
choose, including a front end via browser or a PS4 controller via bluetooth.
And the most important, I've evolved the code to use a tensorflow model
(see https://github.com/SullyChen/Autopilot-TensorFlow)
My code follows:

____________________________________________
collect_training_data_ws_sc_ps4
____________________________________________

__author__ = 'barbosa.alessandro'

import numpy as np
import cv2
import serial
import pygame
from pygame.locals import *
import socket

import websocket
import thread
import time
import json

# steering angle
dx1 = 0

def on_message(ws, message):
    payload = json.loads(message)
    if payload['topic']=="manual":
        print payload['movement'] + payload['direction']

def on_error(ws, error):
    print error

def on_close(ws):
    print "### closed ###"

def on_open(ws):
    CollectTrainingData()

def sendit(dx1, dy1):
    topic = "computer"
    msg = '{"topic":"' + topic + '","dx1":"' +  str(dx1) + '","dy1":"' +
str(dy1) + '"}'
    ws.send(msg)

class CollectTrainingData(object):

    controller = None
    axis_data = None
    button_data = None
    hat_data = None

    def __init__(self):

        self.server_socket = socket.socket()
        self.server_socket.bind(('192.168.2.153', 8000))
        self.server_socket.listen(0)

        # accept a single connection
        self.connection = self.server_socket.accept()[0].makefile('rb')

        self.send_inst = True

        """Initialize the joystick components"""

        pygame.init()
[... 483 lines cut ...]
            cv2.destroyAllWindows()

        finally:
            print "Video stream ended"

#class ThreadServer(object):

#    def server_thread(host, port):
#        server = SocketServer.TCPServer((host, port), VideoStreamHandler)
#        server.serve_forever()

#    def server_thread2(host, port):
#        server = SocketServer.TCPServer((host, port), SensorDataHandler)
#        server.serve_forever()

#    distance_thread = threading.Thread(target=server_thread2,
args=('192.168.2.153', 8002))
#    distance_thread.start()
#    video_thread = threading.Thread(target=server_thread('192.168.2.153',
8000))
#    video_thread.start()

if __name__ == '__main__':
    VideoStreamHandler()
#    ThreadServer()

lakshand8, 2018-01-19:
@hamuchiwa @aleonnet thank you for your help so far. I ran into another error when I ran the rc_drive.py file. I updated the code using @aleonnet's code (https://github.com/aleonnet/AutoRCCar-1/blob/aleonnet-stand-alone/raspberryPi/rc_driver_ws_sa2.py) and I have a trained mlp.xml file to test out the code with, but when I run the program i get the following error:

`[ INFO:0] Initialize OpenCL runtime...
OpenCV Error: Unspecified error (The network has not been trained or loaded) in predict, file /home/pi/opencv/modules/ml/src/ann_mlp.cpp, line 405
Video Stream Ended
Traceback (most recent call last):
  File "drive.py", line 361, in <module>
    VideoStreamHandler()
  File "drive.py", line 298, in __init__
    prediction = self.model.predict(image_array)
  File "drive.py", line 55, in predict
    ret, resp = self.model.predict(samples)
cv2.error: /home/pi/opencv/modules/ml/src/ann_mlp.cpp:405: error: (-2) The network has not been trained or loaded in function predict
`

any idea why that is?

aleonnet, 2018-01-19:
It’s probably related to your open cv installation. It’s strange that
you’re using opencv modules from /home/pi/...
I’d try to reinstall and test it before returning to the code.

lakshand8, 2018-01-19:
@aleonnet I had a lot of problem installing OpenCV 3 on this raspberry Pi for some reason. Tried about 5 different methods to finally get it working and yes I installed it on the home directory. Is there a chance it could be the mlp file itself? reinstalling opencv will be my last resort as it takes so much effort. Any chance I can test out your mlp file please?

aleonnet, 2018-01-19:
Have you checked if opencv is really working?
Is you model file on the same path of my code?

lakshand8, 2018-01-19:
yes it is working, I can run a face tracker program and it works perfectly fine. and yes it is. That's why I dont understand this error

aleonnet, 2018-01-19:
If your using my code, your training model is SullyChen's Tensorflow. You
have to train your model using train.py.
What model are you using? OpenCV Neural Network or Tensorflow?

From https://github.com/SullyChen/Autopilot-TensorFlow :

Download the dataset
<https://drive.google.com/file/d/0B-KJCaaF7elleG1RbzVPZWV4Tlk/view?usp=sharing>
and extract into the repository folder

Use python train.py to train the model

lakshand8, 2018-01-20:
I'm not using SullyChen's Tensorflow model. I used @hamuchiwa's OpenCV model (mlp_training.py) to train it. The training worked fine, with an accuracy of around 70%. I'll try your dataset and see what happens. Thank you.

lakshand8, 2018-01-24:
@aleonnet @hamuchiwa I'm still running into the 'The network has not been trained or loaded in function predict' error when I run the rc_drive.py script. I've tried everything possible so far. I noticed that in the mlp_training.py code has 
`image_array = np.zeros((1, 38400))
label_array = np.zeros((1, 4), 'float')`

but when I run the script, it outputs 
`Image array shape:  (975, 38400)
Label array shape:  (975, 4)`

I'm not sure where the 975 comes from? Should it not be 1, 38400?

lakshand8, 2018-01-24:
@aleonnet @hamuchiwa so I managed to get the above error sorted, all I had to do was change `self.model.load('mlp_xml/mlp.xml')` to `self.model = cv2.ml.ANN_MLP_load ('mlp_xml/mlp.xml')`

But now when I run the program, the camera turns on for a second and the program terminates with the following error.
[ INFO:0] Initialize OpenCL runtime...
Video Stream Ended
Traceback (most recent call last):
  File "drive.py", line 364, in <module>
    VideoStreamHandler()
  File "drive.py", line 345, in __init__
    self.rc_car.steer(prediction)
TypeError: steer() takes exactly 1 argument (2 given)

The code for steer() function is as follows:

> def steer(prediction):
        if prediction == 2:
            rearMotor.run(Adafruit_MotorHAT.FORWARD)
            print("Forward")
        elif prediction == 0:
            frontMotor.run(Adafruit_MotorHAT.BACKWARD)
            print("Left")
        elif prediction == 1:
            frontMotor.run(Adafruit_MotorHAT.FORWARD)
            print("Right")
        else:
            rearMotor.run(Adafruit_MotorHAT.RELEASE)

aleonnet, 2018-01-24:
You are passing more than 1 argument or there’s something wrong with the
class declaration...

TypeError: steer() takes exactly 1 argument (2 given)

lakshand8, 2018-01-25:
left out the self in 'def steer(prediction)' ... simple fix that took alot of time.. @aleonnet  thank you.

code fix: `def steer(self, prediction):`

closed 2018-06-29 by hamuchiwa