hamuchiwa/AutoRCCar #18: Question about cascade xml for stop sign

Ficaa, 2017-01-26:
Hi, i have question about stop_sign.xml. I work on some project where i have sign on road and my robot must recognise them and does some operation, for example stop for 5 second, or 3 second etc... I tried to parse your stop_sign.xml in python but  i can't do that.  Where i can find clasificator for sign or how python can load them? Thanks in advence

hamuchiwa, 2017-01-30:
I trained that stop sign classifier in OpenCV. OpenCV comes with some pre-trained classifiers in the data folder. Also you can train your own haar-cascade classifier in OpenCV as well.

isrugeek, 2017-02-03:
Actually Train your own Classifier is such weird thing . rather Use SURF or SIFT Object detection for custom objects detection.

chandrakant-github, 2017-06-02:
@hamuchiwa can you please explain me process how you trained stop sign and traffic lights  

im new to OpenCv pls explain

hamuchiwa, 2017-06-05:
@chandrakant-github I followed this great [tutorial](http://coding-robin.de/2013/07/22/train-your-own-opencv-haar-classifier.html).

chandrakant-github, 2017-06-05:
@hamuchiwa thank you it helped a lott

closed 2018-03-19 by hamuchiwa