hamuchiwa/AutoRCCar #20: mlp_training.py not working

skywong1230, 2017-02-15:
Dear Author,

your code in mlp_training.py：

> model = cv2.ANN_MLP()
> model.create(layer_sizes)
> criteria = (cv2.TERM_CRITERIA_COUNT | cv2.TERM_CRITERIA_EPS, 500, 0.0001)
> criteria2 = (cv2.TERM_CRITERIA_COUNT, 100, 0.001)
> params = dict(term_crit = criteria,
>                train_method = cv2.ANN_MLP_TRAIN_PARAMS_BACKPROP,
>                bp_dw_scale = 0.001,
>                bp_moment_scale = 0.0 )
> 
> print 'Training MLP ...'
> num_iter = model.train(train, train_labels, None, params = params)

But it cannot work.

Then, I have modified as following:

> model = cv2.ml.ANN_MLP_create()
> model.setLayerSizes(layer_sizes)
> model.setTrainMethod(cv2.ml.ANN_MLP_BACKPROP)
> model.setBackpropMomentumScale(0.0)
> model.setBackpropWeightScale(0.001)
> model.setTermCriteria((cv2.TERM_CRITERIA_COUNT, 20, 0.01))
> model.setActivationFunction(cv2.ml.ANN_MLP_SIGMOID_SYM, 2, 1)
> 
> print 'Training MLP ...'
> num_iter = model.train(train, cv2.ml.COL_SAMPLE, train_labels)

It can create the model, but there is an error about the last line of the above code.
> num_iter = model.train(train, cv2.ml.COL_SAMPLE, train_labels)

do yo have any solutions???

Thank you!

skywong1230, 2017-02-15:
The error msg:

> OpenCV Error: Assertion failed (samples.type() == CV_32F || samples.type() == CV_32S) in setData, file /Users/jenkins/miniconda/1/x64/conda-bld/work/opencv-3.1.0/modules/ml/src/data.cpp, line 251
> Traceback (most recent call last):
>   File "mlp_training.py", line 62, in <module>
>     num_iter = model.train(train, cv2.ml.COL_SAMPLE, train_labels)
> cv2.error: /Users/jenkins/miniconda/1/x64/conda-bld/work/opencv-3.1.0/modules/ml/src/data.cpp:251: error: (-215) samples.type() == CV_32F || samples.type() == CV_32S in function setData

hamuchiwa, 2017-02-15:
@skywong1230 Try to change the response to float. Take a look at this [sample](https://github.com/opencv/opencv/blob/master/samples/python/letter_recog.py#L125).
self.model.train(samples, cv2.ml.ROW_SAMPLE, np.float32(new_responses))

skywong1230, 2017-02-15:
I have modified the code as you mention:

> num_iter = model.train(train, cv2.ml.ROW_SAMPLE, np.float32(train_labels))

But it stills not working and the error msg still the same:

> OpenCV Error: Assertion failed (samples.type() == CV_32F || samples.type() == CV_32S) in setData, file /Users/jenkins/miniconda/1/x64/conda-bld/work/opencv-3.1.0/modules/ml/src/data.cpp, line 251
> Traceback (most recent call last):
>   File "mlp_training.py", line 62, in <module>
>     num_iter = model.train(train, cv2.ml.ROW_SAMPLE, np.float32(train_labels))
> cv2.error: /Users/jenkins/miniconda/1/x64/conda-bld/work/opencv-3.1.0/modules/ml/src/data.cpp:251: error: (-215) samples.type() == CV_32F || samples.type() == CV_32S in function setData

I have no idea about how to solve the problem. Maybe the structure of the "train" or "train_labels" parameters need to be reshape?

hamuchiwa, 2017-02-15:
@skywong1230 I'm puzzled too, it's a very confusing error. Does the dimension of input data or label match the layer size?

skywong1230, 2017-02-15:
Here is my whole log:

> Loading training data...
> ['train_labels', 'train']
> (1, 307200)
> (1, 4)
> (1, 307200)
> (1, 4)
> Loading image duration: 0.02977331
> Training MLP ...
> OpenCV Error: Assertion failed (samples.type() == CV_32F || samples.type() == CV_32S) in setData, file /Users/jenkins/miniconda/1/x64/conda-bld/work/opencv-3.1.0/modules/ml/src/data.cpp, line 251
> Traceback (most recent call last):
>   File "mlp_training.py", line 62, in <module>
>     num_iter = model.train(train, cv2.ml.ROW_SAMPLE, np.float32(train_labels.reshape(1,4)))
> cv2.error: /Users/jenkins/miniconda/1/x64/conda-bld/work/opencv-3.1.0/modules/ml/src/data.cpp:251: error: (-215) samples.type() == CV_32F || samples.type() == CV_32S in function setData

The Shape of my trainData is (1, 307200)
The Shape of my trainLable is (1,4)
The layer size is setted as: 

> layer_sizes = np.int32([307200, 32, 4])

Here is the whole code:

skywong1230, 2017-02-15:
OK, I find the solution finally, the trainData should be convert to np.float32 also.

> num_iter = model.train(np.float32(train), cv2.ml.ROW_SAMPLE, np.float32(train_labels))

> Loading training data...
> ['train_labels', 'train']
> (1, 307200)
> (1, 4)
> (1, 307200)
> (1, 4)
> Loading image duration: 0.025519053
> Training MLP ...
> Training duration: 0.762251955
> Ran for 1 iterations
> Prediction: [1]
> True labels: [1]
> Testing...
> Train rate: 100.000000:

Thank you hamuchiwa

hamuchiwa, 2017-02-16:
@skywong1230 Thank you for solving the issue 👍

geek1111, 2017-03-05:
@skywong1230  Can you send me all the modifications that you have done in your code
Thanks in advance!

closed 2018-03-19 by hamuchiwa