hamuchiwa/AutoRCCar #21: problem in streaming

varun333, 2017-03-05:


hamuchiwa, 2017-03-06:
@varun333 Did you run server first? Have you tried changed port?

varun333, 2017-03-06:
hello hamuchiwa 

thanks for the help but here are some problem which I am currently  facing while running these

1)while using rapistill to capture image the preview display opens up but mine is not opening and I had enabled it. Generally in place of image some box opens for the time I am running the code.  

2)Now second problem here as you have mentioned Id run the rc control test.py then camera callibration then collect training .py and then stream server .py and then  stream client.py , the error mentioned above got resolved.

2)But due to my camera is not showing the preview of the image it throws large sum of errors as of which earlier I was not able to capture any kind of image or video. Any kind of help will be great help.

3)  As i had said RC control test.py was working properly but it is again throwing error

File "C:/Users/USER/PycharmProjects/AutoRCCar-master/test/rc_control_test.py", line 73
    if name == 'main':
                     ^
IndentationError: unindent does not match any outer indentation level

As i have no idea about working on pycharm so help me here.
I had adapted this as my project and now I am stuck, help me out.

hamuchiwa, 2017-03-07:
@varun333 Are you trying to display video stream on raspberry pi or pc? 
It says IndentationError, remove extra spaces before "if name == ''main'" if there are any.

varun333, 2017-03-07:
For sure on PC.

Just wanna ask, at first when I run this code RC control test.py was working properly but after running collection_training.py, the car which I was running through keyboard has opted the self steer function.

Is this what causing the problem, when I am again running the RC control test.py and it just showing 
Traceback (most recent call last):
  File "C:/Users/USER/PycharmProjects/AutoRCCar/test/rc_control_test.py", line 73, in <module>
    RCTest()
  File "C:/Users/USER/PycharmProjects/AutoRCCar/test/rc_control_test.py", line 12, in __init__
    self.steer()
AttributeError: 'RCTest' object has no attribute 'steer'

So now what If I am initiating my project so I again i have to run this control test.py or it was for just one time as I have run the collection training .py code.

hamuchiwa, 2017-03-08:
All test.py files are just for testing purpose. When collecting images, no need to run these test files. Same thing here, start server first, then client.

varun333, 2017-03-09:
how did you do the video streaming
gstreamer or vlc meadia player

varun333, 2017-03-10:
and these are my camera matrix
[[  2.40406352e+03   0.00000000e+00   1.06956635e+03]
 [  0.00000000e+00   2.45713525e+03   8.87405747e+02]
 [  0.00000000e+00   0.00000000e+00   1.00000000e+00]]
Ay: 2457.13524515
u0: 1069.56635117
v0: 887.405746804

hamuchiwa, 2017-03-10:
@varun333 
Video streaming is handled by OpenCV.
Images are captured by pi camera. If you are using the exact same pi camera module, then calibration is not needed. 
As OpenCV doc says, it needs "at least 10 images". Go with the second one.

varun333, 2017-03-10:
I am streaming through vlc, will that work fine for streaming and any idea about the latency rate.

varun333, 2017-03-11:
thats my vlc error

and stream client.py when i am shows nothing

Connection failed:
VLC could not connect to "192.168.2.12:8554".
Your input can't be opened:
VLC is unable to open the MRL 'rtsp://192.168.2.12:8554/'. Check the log for details.

varun333, 2017-03-11:
first i run my vlc command on pi camera and started playing the video on vlc player then i run the code of server on pc and of client on pi, the server is showing me the streaming while I am running the client code but after it throws out the error

does it has anything to do with pi camera resolution as you have written  to the one that of raspivid command that I have passed for vlc streaming
raceback (most recent call last):
  File "/home/pi/Desktop/raspberryPi/stream_client.py", line 23, in <module>
    with picamera.PiCamera() as camera:
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 433, in __init__
    self._init_preview()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 513, in _init_preview
    self, self._camera.outputs[self.CAMERA_PREVIEW_PORT])
  File "/usr/lib/python3/dist-packages/picamera/renderers.py", line 558, in __init__
    self.renderer.inputs[0].connect(source).enable()
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 2212, in enable
    prefix="Failed to enable connection")
  File "/usr/lib/python3/dist-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to enable connection: Out of resources

hamuchiwa, 2017-03-11:
I didn't use vlc before, but as long as you can turn streaming image into an array, the goal is accomplished. Resolution is set to low to achieve "smooth" video streaming.

varun333, 2017-03-11:
the video is coming to my vlc screen on pc using raspivid command on raspberry pi, so does that mean  again I have to run your client code as well or not.

have you done this streaming through gstreamer or using only python code.

varun333, 2017-03-11:
if you can go through this video 

https://www.youtube.com/watch?v=JjPsW-7FUng

you can get me what I am trying to tell.

here the module shown in the link apart from that I am trying to run your client code.

hamuchiwa, 2017-03-11:
How did you save image array through vlc?

varun333, 2017-03-11:
As about saving image , I was gonna ask from you.

It will become easy for me ,  if you would help me by telling me  how did you stream the video over your mac using picamera. So that I should seek for it and just run those streaming code.

varun333, 2017-03-11:
I think i need to install python picamera on my raspberry to run the code of yours

varun333, 2017-03-11:
just run the code of yours on PC and raspi, just thrown out this error

File "/home/pi/Desktop/raspberryPi/stream_client.py", line 26, in <module>
    connection.write(stream.read())
  File "/usr/lib/python3.4/socket.py", line 391, in write
    return self._sock.send(b)
ConnectionResetError: [Errno 104] Connection reset by peer

help me out here.
i just wanna know the whole procedure so as to run the code of yours.

varun333, 2017-03-11:
this time i am not running vlc but initiated raspivid command to start the camera

after running server test.py throws out the output 

Connection from:  ('192.168.2.12', 43102)
Streaming...
Press 'q' to exit

and client.py throws out the error
File "/home/pi/Desktop/raspberryPi/stream_client.py", line 14, in <module>
    with picamera.PiCamera() as camera:
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 433, in __init__
    self._init_preview()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 513, in _init_preview
    self, self._camera.outputs[self.CAMERA_PREVIEW_PORT])
  File "/usr/lib/python3/dist-packages/picamera/renderers.py", line 558, in __init__
    self.renderer.inputs[0].connect(source).enable()
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 2212, in enable
    prefix="Failed to enable connection")
  File "/usr/lib/python3/dist-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to enable connection: Out of resources

hamuchiwa, 2017-03-11:
I think you misunderstood on who to start server and client. First you start the server_test.py on your pc, second run stream_client.py on raspberry pi. There are no vlc and raspivid involved.

varun333, 2017-03-12:
First I run the server.py on pc , as I hit enter on server nothing appear but as i run client on raspi 
error on server side 
Connection from:  ('192.168.2.12', 36988)
Streaming...
Press 'q' to exit

Traceback (most recent call last):
  File "C:\Python27\stream_server_test.py", line 41, in <module>
    VideoStreamingTest()
  File "C:\Python27\stream_server_test.py", line 13, in __init__
    self.streaming()
  File "C:\Python27\stream_server_test.py", line 31, in streaming
    image = cv2.imdecode(np.fromstring(jpg, dtype=np.uint8), cv2.CV_LOAD_IMAGE_UNCHANGED)
NameError: global name 'np' is not defined

and on client site

File "/home/pi/Desktop/raspberryPi/stream_client.py", line 24, in <module>
    connection.flush()
  File "/usr/lib/python3.4/socket.py", line 391, in write
    return self._sock.send(b)
ConnectionResetError: [Errno 104] Connection reset by peer

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/Desktop/raspberryPi/stream_client.py", line 33, in <module>
    connection.close()
  File "/usr/lib/python3.4/socket.py", line 391, in write
    return self._sock.send(b)
BrokenPipeError: [Errno 32] Broken pipe
>>>

hamuchiwa, 2017-03-13:
You got an error: "NameError: global name 'np' is not defined"

varun333, 2017-03-14:
@hamuchiwa 
Still not running,as i am trying to run your codes individually
server side :
File "C:/Users/USER/PycharmProjects/AutoRCCar/test/stream_server_test.py", line 44, in <module>
    VideoStreamingTest()
  File "C:/Users/USER/PycharmProjects/AutoRCCar/test/stream_server_test.py", line 16, in __init__
    self.streaming()
  File "C:/Users/USER/PycharmProjects/AutoRCCar/test/stream_server_test.py", line 34, in streaming
    image = cv2.imdecode(np.fromstring(jpg, dtype=np.uint8), cv2.CV_LOAD_IMAGE_UNCHANGED)
AttributeError: 'module' object has no attribute 'CV_LOAD_IMAGE_UNCHANGED'

client side:
File "/home/pi/Desktop/raspberryPi/stream_client.py", line 26, in <module>
    connection.write(stream.read())
  File "/usr/lib/python3.4/socket.py", line 391, in write
    return self._sock.send(b)
ConnectionResetError: [Errno 104] Connection reset by peer

varun333, 2017-03-14:
is it necessary to run collection.training.py before the streaming of video, are these both code interrelated and somehow causing these error.

varun333, 2017-03-16:
hamuchiwa thanx for the help, just now my both data are streaming fine but just wanna know in ultrasonic the output comes for 15 to 20 and then it says broken pipe error. 

And now i am heading for next module i.e. mlp training data.py so just guide me early with the module and requirements  that  I shoud seek to make it work in one go.

varun333, 2017-03-16:
@hamuchiwa I just wanna ask earlier my stream_client was not working but i run the collection training.py and not make any key movement, so now my stream client is working and again I want to start  from the begining so how I am gonna do it as my key function or movement were not collected that time and while I perform this it should be on track or random moves count.

hamuchiwa, 2017-03-17:
@varun333 Does it throw any errors when using keyboard to control the rc car with the test file?

varun333, 2017-03-17:
earlier kind of def self() related error but now when I am pressing my keyboard key the command are getting printed but my car is not moving.

varun333, 2017-03-17:
well when i will be running collect training data.py then what kind of output I am gonna be expecting and for how much duration I have to run this module on specified track  and after that in mlp training , if you will be able to tell me output ,would be a great help.

hamuchiwa, 2017-03-17:
Check this [thread](https://github.com/hamuchiwa/AutoRCCar/issues/2) @ghost commented on this issue, hope it helps. 
Well, when running collect_training_data.py, every time you press the key, the rc car moves, one frame is saved, when you exit, all images will be stacked together and saved into a .npz file which you will need for neural network training.

varun333, 2017-03-18:
@hamuchiwa
 again thanx for the help 
now for colleect training.py i will get instant output as i move car on the track or will it take time to generate the output and 
should I also have to create the folder or it will get save to the empty folder of yours and also tell me about the folder where it will get saved.

varun333, 2017-03-19:
@hamuchiwa 
i just place the track and first run server then collect_training then server client and then rc control test py for movement of my car and when i stop the collect_training, I just got SAVED  image in training image and at the time when I WAS DRIVING MY CAR NO OUTPUT  GENERATED IN  COLLECT TRAINING.PY COMPILER

ONE THING MORE I WANNA ASK TO MOVE CAR I have to run collect training.py first  or rc control.py.

HELP!!!! AND ALSO DIDNT RECEIVE TRAINING DATA, WHAT CAN BE THE REASON.

varun333, 2017-03-20:
@hamuchiwa to run collect_training data .py did I have to run stream_ server.py or not and if yes the address will be same for both as i just run collect training data.py and on pi stream_client.py and pygame window open for sort duration and same with cam and yes also I have to change the path in collect training.py for training data. 

please tell me the procedure fast as I have to submit this very soon 
thanks

varun333, 2017-03-20:
@hamuchiwa  i have fixed the error as i have increased the camera resolution but how i can come to know about the roi for this 8mp camera and now before collect training.py  i have to run stream server or not thats what i wanna ask.

hamuchiwa, 2017-03-20:
collect training data.py contains server side code already. Try tilting down the camera angle a bit to  take full frame at first (so frame is filled with track only), in this case the initial image array will be np.zeros((1, 76800)) if your resolution is 320x240.

varun333, 2017-03-20:
roi = image[120:240, :] this is for 320 *240, 
and what if i set resolution to 640*480 then in that case what will be my initial image array and roi as we are taking lower half image, how are we gonna decide

and one more thing to save testing data i have to create folder of .npz format or the data generated will be automatically of type .npz

varun333, 2017-03-21:
this what i have received as output

Streaming duration: 28.5937477815
(25L, 38400L)
(25L, 4L)
Total frame: 605
Saved frame: 25
Dropped frame 580

varun333, 2017-03-21:
error while running mlp_predict.test.py

 File "C:/Users/USER/PycharmProjects/AutoRCCar/computer/mlp_predict_test.py", line 36, in <module>
    ret, resp = model.predict(test)
cv2.error: ..\..\..\..\opencv\modules\ml\src\ann_mlp.cpp:279: error: (-5) Both input and output must be floating-point matrices of the same type and have the same number of rows in func

varun333, 2017-03-22:
@hamuchiwa 
i am just training and saving in different npz file, so in order to increase train rate what i need to do as you were saying merging the npy file how are we gonna do it or it just keep saving the data in same npz file format.

hamuchiwa, 2017-03-22:
Each time you collect, a npz file will be saved. The training can load multiple npz-s at one time as long as they are in the same folder.

varun333, 2017-03-23:
@hamuchiwa
It like just , I first specified the track and collect the data as test01 and if i want to define the new track and train it then I am gonna overwrite the test01 or i have to save it as test02 npz file in the training_data, and also tell me about the train rate i.e 64, is it workable, how i am gonna increase the train rate.

hamuchiwa, 2017-03-23:
You can use one or multiple npz files. You can tell whether it's good or not once you let the car drive itself, 80% accuracy certainly works better than 64%. You could do a straight line track first, and see what happens.

varun333, 2017-03-24:
yeah i got the accuracy of 80% but as after i run the mlp_predict.test.py

it throws an error why so,

File "C:/Users/USER/PycharmProjects/AutoRCCar/computer/mlp_predict_test.py", line 36, in 
ret, resp = model.predict(test)
cv2.error: ........\opencv\modules\ml\src\ann_mlp.cpp:279: error: (-5) Both input and output must be floating-point matrices of the same type and have the same number of rows in func

varun333, 2017-03-25:
@hamuchiwa Its urgent plz help.

hamuchiwa, 2017-03-25:
I remember you had the same error while you were doing the training, it's caused by OpenCV version difference. If you are always using the same track, you don't even need map_predict_test.py.

varun333, 2017-04-08:
@hamuchiwa please help me out with simple movement of car on the track autonomously as i just place the straight track and  its not following the path as after not running prediction.py part, what i need to do to make work the prediction module work

varun333, 2017-04-10:
@hamuchiwa how do you have obtained the testing data? the test cases of trained data can be used as testing data

varun333, 2017-04-10:
As after cut copy paste of npz file from training data to testing data
['train_labels', 'train']
(53L, 38400L)
(53L, 4L)
(53L, 38400L)
(53L, 4L)
Prediction time per frame: 0.00167296483453
Prediction: [1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1
 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1]
True labels: [2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 1 2 1 2 1 2 1 1 2 1 2 3 2 2 2 1 1 1 1 2 1 2
 2 0 2 2 2 2 2 2 2 1 2 2 2 2 2 2]
Testing...
12
Test rate: 22.641509

Process finished with exit code 0

hamuchiwa, 2017-04-10:
You can either have separate files for training and testing, or random split (for example, 7:3) a data set into training and testing. What's your train rate?

varun333, 2017-04-10:
@hamuchiwa Train rate: 55.752212:

varun333, 2017-04-10:
@hamuchiwa
 i just only wanna know ,we get the training data from collect training data.py in npz format but how we get the testing data (because when i copy the npz file from training data to testing data file ) then only i was able to get the above prediction as an output.

Is this how its need to be done

hamuchiwa, 2017-04-10:
Okay, try this if you haven't done so. 
1. collect data using collect_training.py
2. randomly split collected data into training and testing sets (ratio is 7:3 or 8:2)
3. copy each files to its corresponding folder
Your training set accuracy is low, make sure the data you collected is good quality.

varun333, 2017-04-11:
@hamuchiwa

 i have been working on it and i get the stop sign as a classifier but the problem is my car gets deviated from the track and always go first straight then turn right i dont know what the problem is...
&
 i have made this traffic light from led where red gets detected few times when i twist or turn it but green  is showing no sign of response 
&
when I place object infront of car it get stopped but after few seconds

Did i have to make some changes to rc driver.py with calculation like height and distance or  is it ok

hamuchiwa, 2017-04-12:
A simple straight line should give you a accuracy at least 90%. Try to tilt down the camera so it only sees the road, and use entire image as train data. It usually gives you a better result than using half of the image.

varun333, 2017-04-13:
@hamuchiwa So you are saying i should go with 76800 instead of 38400 while collecting the training data as well as in rc driver.py.

And one time training is effecient or I have train it for several times over the same track

varun333, 2017-04-13:
@hamuchiwa  modified code would be like this

image_array = np.zeros((1,76800 ))
        label_array = np.zeros((1, 4), 'float')

 roi = image[320:240, :]
                    cv2.imwrite('training_images/frame{:>05}.jpg'.format(frame), image)
                    cv2.imshow('image', image)
                    temp_array = roi.reshape(1, 76800).astype(np.float32)

varun333, 2017-04-13:
@hamuchiwa to use entire image as train data what parameter do i need to change?

hamuchiwa, 2017-04-14:
Change array size to reflect resolution. You could just train once and using the same track over and over again without doing any testing, or as I mentioned earlier, you could also do cross validation. I would suggest you to get familiar with the flow of machine learning first.

varun333, 2017-04-14:
@hamuchiwa 
as to work with complete resolution i just double the figure of 38400 to 76800 in image array for 320*240 and for temp array also done the same 
but for roi it will be same or it would be somewhat like this 
 roi = image[320:240, :]

hamuchiwa, 2017-04-17:
Since we are not slicing the image, "roi" is no longer needed, or you could set "roi = image".

[12 more replies]

closed 2018-06-29 by hamuchiwa