hamuchiwa/AutoRCCar #22: Error in `mlp_training.py`

geek1111, 2017-03-05:
After running `mlp_training.py` it shows an error...

```
Loading training data...
(0, 38400)
(0, 4)
Loading image duration: 0.017612816
Training MLP ...

Traceback (most recent call last):
  File "/home/pi/CAR/computer/mlp_training.py", line 49, in <module>
    num_iter = model.train( train, train_labels, None, params = params)
error: /home/pi/opencv/modules/ml/src/ann_mlp.cpp:690: error: (-5) input training data should be a floating-point matrix withthe number of rows equal to the number of training samples and the number of columns equal to the size of 0-th (input) layer in function CvANN_MLP::prepare_to_train

>>> 
```

hamuchiwa, 2017-03-06:
@geek1111 
import cv2
cv2.__version__
>> '2.4.10.1'

geek1111, 2017-03-08:
hi @hamuchiwa 
I'm still facing this issue i have tried as you said in issue #20 also tried, as _skywong1230_ said in that issue still getting errors! please help me! do you face same issue as I'm facing?

hamuchiwa, 2017-03-09:
@geek1111 Why the number of rows is 0?
Loading training data...
(0, 38400)
(0, 4)

swapnillubal, 2017-03-09:
Hi @hamuchiwa 
Im having the same issue 
I changed the framerate of vid streaming does that affect the number of rows?
Any solution ?

geek1111, 2017-03-10:
In a code written 

```
image_array = np.zeros((1, 38400))
label_array = np.zeros((1, 4), 'float')
```
but after running `mlp_training.py`
it shows 
```
Loading training data...
(0, 38400)
(0, 4)
```
so I've replaced 1's  to 2's  now it can create model but

```
Loading training data...
(1, 38400)
(1, 4)
Loading image duration: 0.001579994
Training MLP ...
Training duration: 29.613473582
Ran for 500 iterations
Prediction: [2]
True labels: [0]
Testing...
Train rate: 0.000000
```
train rate is "0.00000"

hamuchiwa, 2017-03-10:
@geek1111 @swapnillubal 
Image array is initialized to 0s as np.zeros((1, 38400)). Then each image(row) is appended to the array. At last the first row (0s) is removed. Now the array contains all images where row number equals to the number of images loaded. In your case, images are not loaded somehow.

geek1111, 2017-03-10:
while im training data using collect_training_data.py
****************output***********************
Streaming duration: 42.939297537
(61, 38400)
(61, 4)
Total frame: 180
Saved frame: 61
Dropped frame 119
**************************************************
The rows are equal to the saved frames.....Is that correct ???
@hamuchiwa

hamuchiwa, 2017-03-10:
@geek1111 The error says it didn't find input data which is a npz file. Make sure the path is correct and the file exists. 
Yes, it's correct. Each row is an image.

geek1111, 2017-03-10:
@hamuchiwa 
Can you give me trained model with images and other files to test it on my Pi?
P.S. I'm using PI as server as well as client by giving the same ip address in both files.

varun333, 2017-03-11:
suraj can you tell me the procedure how to stream the file using vlc as I am facing some problem
did I have to first run the code or vlc player.

hamuchiwa, 2017-03-11:
@geek1111 You could do it by yourself actually, you can create a Numpy array filled with random numbers, for example, the dimension could be (100, 38400)  or (100, 76800) if you are using all pixels.

varun333, 2017-03-16:
geek1111 can you help me out with streaming module, as i am facing bit problem with python interface for both video as well as ultrasonic sensor data streming.

varun333, 2017-03-20:
@geek1111 hello Suraj I just wanna ask about collect_training_data module, here  first i ran server then collect training then at client stream server module and to move car I had run rc control test.py but at last i just got only single image and no training data in npz format, I think I have provided the wrong path for training data that the folder is empty or there is any other reason.

chandrakant-github, 2017-05-29:
after running mlp training no file is saved ?? 

pls suggest what should i do

nimasmbm, 2017-06-08:
Connected to pydev debugger (build 171.4424.42)
Loading training data...
['train_labels', 'train']
(241, 38400)
(241, 4)
(241, 38400)
(241, 4)
Loading image duration: 0.414502338308
Training MLP ...
Training duration: 53.73782154
Ran for 1 iterations
Prediction: [0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0]
True labels: [2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2
 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2
 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2
 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2
 2 2 2 2 2 2 2 0 0 0 1 2 2 2 2 2 2 2 2 2 2 2 2 1 2 2 2 2 2 2 2 2 2 2 2 2 2
 2 2 2 1 1 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2
 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2]
Testing...
Train rate: 1.244813:

How to increase the Train rate???

hamuchiwa, 2017-06-09:
@nimasmbm I think your training data quality needs to be improved first.

IsmoilovMuhriddin, 2017-11-19:
@nimasmbm  Have you solved the issue I have the same problem. When i run collect_training_data and train mlp , Predicting trained data output is all 0s or all 1s.

varun333, 2017-11-23:
Yes my code ran fine, everything thing is right with the code just due to
some lagging connectivity you might not be able to train the data. Keep
training the data you will get the train rate.

Or later if there is any problem just reply me I will surely help you out.

Machine learning in itself is a very wide concept just keep on going you
will enjoy.

closed 2018-03-19 by hamuchiwa