hamuchiwa/AutoRCCar #23: Issue with collect_training_data.py

Cosmonaut65, 2017-03-14:
Hi @hamuchiwa !

I've been trying to recreate your project and ran into a problem I was hoping you might be able to help me with. When I run the "collect_training_data.py" the line "cv2.imshow('image', image)" does not appear to do anything. It doesn't open a preview window at all and as a result, PyGame cannot capture any key entries either.

I've run the "stream_server_test.py" without any issue and the images from the RaspberryPi stream onto the PC, so it definitely isn't a connection issue.

Do you have any idea why this might be happening?

I'm using OpenCV 2.4.9 on Ubuntu 14.04

Thanks in advance! :)

trymelz, 2017-03-17:
add the following after pygame.init

pygame.display.set_mode((640,480))

hamuchiwa, 2017-03-17:
@Cosmonaut65 That's strange, it should pop up the window as you see using the test files. Does it throw any errors? For PyGame check this [thread](https://github.com/hamuchiwa/AutoRCCar/issues/2), hope you can find your answer.

varun333, 2017-03-20:
@Cosmonaut65 here I think u have exceeded the camera resolution in stream_client.py on raspi thats y  ur module is not getting run, if u r using high pixel camera then u have to adjust ur roi or just go with the program as Hamuchiwa has done

Cosmonaut65, 2017-03-20:
Thanks for the replies guys!
I managed to get it to work by adding the line cv2.waitKey(25) right after cv2.imshow. I'm not sure why this is necessary as it works without it on the rc_driver and stream_server_test scripts. This still doesn't allow key capture, so I added the line "pygame.display.set_mode((500,500),HWSURFACE|DOUBLEBUF|RESIZABLE)" after "pygame.init()" so that I have a small window to send control signals over. This seems to work fine for the time being, though it does freeze after I run it for a bit of time.

Due to the constant freezing I wanted to ask, if I break up the training regularly, and make multiple .npz file, will the mlp_training script be able to read all of them as long as they are in the same folder?

And on a less related note, does it matter if the track surface is white with the edge being dark as opposed to your example where the edge was white? I imagine not as it's a machine learning algorithm that just picks out patterns.

Thanks again for the help!

hamuchiwa, 2017-03-20:
@Cosmonaut65 You can have multiple .npz files, they will be merged together. The actual saved images are not used for training, just for reference purpose. For the track layout, you are absolutely correct.

Cosmonaut65, 2017-03-23:
Thanks a lot for the help @hamuchiwa !
I have another question: How did you calculate the 38,400 input layer nodes? With an image of 120x240 being fed into the neural network, wouldn't it be 28,800?

Also, what steps could I take to improve the neural network training? I wanted to feed a larger image, perhaps the full image, into the neural network. Would this work to improve the predictions, and if so, what changes would I need to make to the code? Would it just be a case of changing the input layer to 76,800 (320x240 - pixels in the full image)?

hamuchiwa, 2017-03-23:
it's 320x120
Yes, take the full frame (320x240) if you can, it works better if you tilt down the camera (frame is filled with track(ground) only). The trade off is you cannot do object detection.

Cosmonaut65, 2017-03-23:
Thanks @hamuchiwa, I've decided to crop the image so that it sees the whole track but not beyond. I'll train and see if it improves predictions. Is the reason for the object detection not working that there is too much data to process with a larger frame in real-time?

hamuchiwa, 2017-03-25:
Not that reason. It's just because when you tilt down the camera angle, it only sees the road, objects such as stop sign won't appear in the image.

chandrakant-github, 2017-05-29:
can you resolve this please 
im getting error during collecting data

chandrakant-github, 2017-05-29:
can you resolve this please
im getting error during collecting data  
 
 temp_array = roi.reshape(1, 38400).astype(np.float32)
ValueError: cannot reshape array of size 76800 into shape (1,38400)

chandrakant-github, 2017-05-29:
hello  @hamuchiwa , i can take and store frames but couldnt save npz file it shows me 
Streaming duration: 206.096424455
(105L, 38400L)
(105L, 4L)
Total frame: 800
Saved frame: 105
Dropped frame 695

but no npz file were saved  in training_data_temp  please help

hamuchiwa, 2017-05-30:
@chandrakant-github Check file path see if that's the issue.

ANISHCAPRICORN, 2017-06-01:
[  **SOLVED** ]   PyGame cannot capture any key entries either. 

Add following code session to collect_training_data.py 

        pygame.init()
        size = [70, 50]
        screen = pygame.display.set_mode(size)

        pygame.display.set_caption("RC_ROBO_CAR")
        self.collect_image()

chandrakant-github, 2017-06-02:
thank you so much

marcusvn, 2018-03-15:
How to change the resolution of a stream from 320x240 to 640x480?  @hamuchiwa I had the problem when I put first "Forward". 
[image]

closed 2018-03-19 by hamuchiwa