hamuchiwa/AutoRCCar #24: RC Car Control Unit

trymelz, 2017-03-17:
what type of on/off switch type controller do you use? our remoter controller use tactile switch.
but it seems the arduino is not able to control the tactile switch directly. we have to use 4n35 to replace the ticticle switch, but it cause other issues (maybe drain too much power from arduino )....

hamuchiwa, 2017-03-17:
A button on/off switch, the rc car costs less than $20.

trymelz, 2017-03-21:
Thanks.  We found the issue is because we used 2.4G rc car, which interference with wifi. After changing to 49MHz rc car, everything works fine.

darshansm, 2017-03-24:
I'm having issue with driving car with keyboard in windows 10. I flashed rc_keyboard_control.ino to my Arduino uno, and i installed all required Dependencies to my windows... tn i run rc_control_test.py but can't drive from keyboard.

hamuchiwa, 2017-03-25:
@darshansm Please check [this thread](https://github.com/hamuchiwa/AutoRCCar/issues/2). Comments from @ghost may be helpful to you.

darshansm, 2017-03-28:
Thank you for your replay. I didn't noticed your message earlier. I solved that issue myself. I am very thank full to you because my model is completely working now. Some little calibrations required i can manage it. If any droughts i'll definitely contact you.

floerer, 2017-04-12:
I almost have the same RC car as you but a newer version I think. there are a lot more resistors and stuff. And the RC car is controlled by a chip in the center. I tried a lot of connecting the jumpingwires to different places, but the only thing I get a bit working is when I connect the ground wire and the forward jumpingwire, then the car moves foward, but without any press on the keyboard. No other things work. The python files and arduino files are working fine and the data is send to the arduino. (PS. does you receiving LED on the arduino flashes or keeps turned on all of the time?). I made a picture of my RC controller board. the VCC and ground are on the exact same spots as your controller board. I'm a noob in electronics but pretty well in programming so I hope you can help me out with connecting the jumpingwires. 
[http://imgur.com/vcYngRQ,hU7WzbB](url)
[http://m.imgur.com/hU7WzbB](url)

jameskkxvil, 2017-04-20:
Hey Would the brains of my rc car is an arduino. I built it myself and i wanna know if i would be able to control my rc from your code? if so what would i need to change?

jameskkxvil, 2017-04-20:
Or could i transfer the controls over to the raspberry pi and control it using the code you have for the arduino? and maybe use something like Bluetooth to communicate? or whatever you think is best.

hamuchiwa, 2017-04-23:
@jameskkxvil It really depends on what you would like to achieve here. Nonetheless, you still need to modify or re-write the code anyway.

farazhussain2015, 2017-08-08:
i am directly connecting my motors with arduino via motor driver. everything is working but motors are not moving. any solution?

lakshand8, 2018-01-11:
@hamuchiwa I'm trying to replicate this project using just a standalone raspberry Pi with in house neural network processing. I was just wondering would I be able to use a Raspberry Pi motor Hat 
( https://www.adafruit.com/product/2348 ) to run the motors or would that be putting too much stress on the Pi processor?

hamuchiwa, 2018-01-11:
@lakshand8 I'm not familiar with motor hat, but I do believe raspberry pi can handle that. Also I still recommend you to train neural network on a computer, only deploy it on raspberry pi.

kittinan, 2018-03-01:
@floerer Did you solve the problem about RC controller? i got the same model RC controller like you.

Gooose1909, 2018-03-01:
I tried to solder wire and try different combinations as input
But it didnt work
Help if u find any solution

floerer, 2018-03-01:
@kittinan  I found another rc controller on the same frequency. I used that one since I was not able to fix the problem with the controller. After all I finished to project and everything worked so great job @hamuchiwa  Detecting the traffic light was the most difficult part for me but after all it was a nice project

Gooose1909, 2018-03-01:
I really want to finish this project

floerer, 2018-03-01:
@Gooose1909 Do you have a controller which uses different voltages for the steering and forward and backward? or does your controller just gives full power when the circuit is closed and nothing when it is open? the last one is easy, you just have to solder the wire from the arduino onto the spot where the circuit normally gets closed when you push the lever forward for example

Gooose1909, 2018-03-01:
Different power levels

floerer, 2018-03-01:
@Gooose1909 I was not able to fix that problem, I found another rc car controller on the same frequency and I used that one

Gooose1909, 2018-03-01:
Ok

closed 2018-03-19 by hamuchiwa

kxrf97, 2019-02-21:
@floerer can you share the car