hamuchiwa/AutoRCCar #25: Issue with mlp_predict_test.py

darshansm, 2017-03-27:
After running the code, I am getting Following error...

C:\Python27\python.exe E:/Downloads/Compressed/AutoRCCar-master/computer/mlp_predict_test.py
(0L, 38400L)
(0L, 4L)
OpenCV Error: Bad argument (Both input and output must be floating-point matrices of the same type and have the same number of rows) in CvANN_MLP::predict, file ..\..\..\..\opencv\modules\ml\src\ann_mlp.cpp, line 279
Traceback (most recent call last):
  File "E:/Downloads/Compressed/AutoRCCar-master/computer/mlp_predict_test.py", line 36, in <module>
    ret, resp = model.predict(test)
cv2.error: ..\..\..\..\opencv\modules\ml\src\ann_mlp.cpp:279: error: (-5) Both input and output must be floating-point matrices of the same type and have the same number of rows in function CvANN_MLP::predict

hamuchiwa, 2017-03-28:
Based on the log and error,  your test data is not present. Make sure file path is correct and you put npz files in your test folder.

darshansm, 2017-03-28:
Thank you so much.... it is path issue..... it is solved now

closed 2017-03-28 by darshansm