hamuchiwa/AutoRCCar #27: Memory Error

rh01, 2017-04-08:
Error Message:
 File "C:\Python27\lib\site-packages\numpy\core\shape_base.py", line 234, in vstack
    return _nx.concatenate([atleast_2d(_m) for _m in tup], 0)
MemoryError

what's wrong? Have you ever met this situation? What I should do? how do you control data size? thank you!

hamuchiwa, 2017-04-10:
I did run into this issue. But I think it's a fair point, if it throws a memory error then some operations need to be optimized. What's the resolution are you using? How large is your data?

rh01, 2017-04-12:
thanks! I think it 's should be OpenCV version issue or Operation System? I use 240*329 ewsolution and My data is mostly 670M+.

hamuchiwa, 2017-04-12:
It could be a Numpy issue since internally OpenCV utilizes Numpy, or it could just your data exceeds your machine's memory. Check the memory consumption while loading the data, see what happens.

rh01, 2017-04-13:
Thank you.I think it not be my memory not enough, My machine have 8G RAM, It's Ok when I check my laptop.

closed 2018-03-19 by hamuchiwa