hamuchiwa/AutoRCCar #28: Getting error in ULTRASONIC_CLINT.py

darshansm, 2017-04-08:
Traceback (most recent call last):
  File "/home/pi/Downloads/AutoRCCar-master/raspberryPi/ultrasonic_client.py", line 52, in <module>
    distance = measure()
  File "/home/pi/Downloads/AutoRCCar-master/raspberryPi/ultrasonic_client.py", line 31, in measure
    elapsed = stop-start
UnboundLocalError: local variable 'stop' referenced before assignment

darshansm, 2017-04-08:
Traceback (most recent call last):
  File "/home/pi/Downloads/AutoRCCar-master/raspberryPi/ultrasonic_client.py", line 52, in <module>
    distance = measure()
  File "/home/pi/Downloads/AutoRCCar-master/raspberryPi/ultrasonic_client.py", line 31, in measure
    elapsed = stop-start
UnboundLocalError: local variable 'stop' referenced before assignment

hamuchiwa, 2017-04-10:
It seems the loop is not executed: while GPIO.input(GPIO_ECHO)==1
Do you get any input from GPIO pin?

darshansm, 2017-04-10:
i am getting distance.......i can see "obstacle in front" message in computer. once i remove obstacle, i am getting that error

hamuchiwa, 2017-04-10:
Have you tried to declare variables at the beginning of the measure() function:
start, stop = 0, 0
I wonder what would happen.

darshansm, 2017-04-10:
no...i ll try once
i didn't change any thing in program except IP

darshansm, 2017-04-10:
i tried....getting same error

chandrakant-github, 2017-06-02:
@hamuchiwa how to train the car for obstacle ahead how ultrasonic data is tought  ??

can you explain pls

chandrakant-github, 2017-06-03:
@hamuchiwa  can you provide me code through which you have taken cascade_xml 

for stop and traffic sign

hamuchiwa, 2017-06-05:
@chandrakant-github Hi, ultrasonic sensor just measures the distance, if the distance is below some threshold, you make the car stop. Regarding your second question, are you referring to how to train cascade classier? I followed a great tutorial, link is on project home page, or in your another thread.

chandrakant-github, 2017-06-05:
@hamuchiwa thank you so much  for great help

closed 2018-03-19 by hamuchiwa