hamuchiwa/AutoRCCar #29: Problem with ultrasonic & stream tests

razRzz, 2017-04-22:
Hello!

I have a weird problem with the ultrasonic & stream tests.
Every time I run either ultrasonic_server_test.py or stream_server_test.py on my laptop with PyCharm, and then run ultrasonic_client.py or stream_client.py on the RPi, nothing happens. They both stay like they're waiting for something to happen. No warning/error, no nothing.
I made sure I had the correct local IP of the laptop in the scripts; I tried multiple ports, still the same outcome. Do you have any idea what I'm doing wrong?
I've read through all the issues risen, but I have not seen my problem. Sorry if I've missed something already solved here.

Thanks a lot for this awesome project!

hamuchiwa, 2017-04-23:
That's strange... what about just running a basic sample server client program, does it display anything?

razRzz, 2017-05-11:
Sorry for the late reply. 
Even with basic client/server examples, it would still behave the same.
Seems like the problem was my damn firewall, blocking everything. I've added a couple of exceptions for the ports, and it's working just fine now.
Thanks for your help!

closed 2017-05-11 by razRzz

ayesha-manzoor, 2019-02-03:
hey, i am having the exact same problem.. can you please mention the exceptions that you added? @razRzz