hamuchiwa/AutoRCCar #32: Understanding RC_Driver and thinking behind your approach.

coeshack, 2017-05-03:
Thank you for this fantastic project, I've been working to replicate something similar and add addition functionality and have been learning supervised machine learning on the go. 
 
Just to test my understanding. 

1. so we've collected our data (frames and assigned them keyboard inputs) 
2. we reshape and process this data into our neural net 
3. the neural is trained to make predictions and outputs 1 2 3 for driving and steering. 

Coud you explain this part of the code 
I'm not clear on how the .getTickCount() is used to generate predictions from the MLP.  
`# create MLP
layer_sizes = np.int32([38400, 32, 4])
model = cv2.ANN_MLP()
model.create(layer_sizes)
model.load('mlp_xml/mlp.xml')

# generate predictions
e0 = cv2.getTickCount()
ret, resp = model.predict(test)
prediction = resp.argmax(-1)
e00 = cv2.getTickCount()
time0 = (e00 - e0)/cv2.getTickFrequency()
print 'Prediction time per frame:', time0/(test.shape[0])`

On the actual RC_Driver  
I want to understand your approach on how you were able to use the trained MLP to drive the vehicle. 
if i put this this trained model onto a new track how does it different a left tun from a right turn? 
My understanding it the model has been trained to look at each frame and predict forward left or right based on the frame it sees. 

I'd like to understand how your thinking how you added the stop sign classifiers to an already trained neural net. 

Thanks for your time ! 
James

hamuchiwa, 2017-05-04:
.getTickCount() is a OpenCV function, just used to calculate the time or duration. You could also use built-in Python time.time() to do the calculation as well. 

After the training, the weights/parameters of the network are saved. In prediction, the weights are loaded and the network generates a prediction based on that. Basically, the purpose of training is to find the proper weights that minimize the cost. 

Your understanding is correct, of course, the track can vary but should be in a similar pattern. 
Stop sign and the neural net are different classifiers, they are not conflicting with each other.

AiBuddy, 2017-08-24:
I am getting this error. "F:/SDC CAR Project/AutoRCCar-1-aleonnet-stand-alone/computer/mlp_predict_test.py", line 38, in <module>
    prediction = resp.argmax(-1)
AttributeError: 'NoneType' object has no attribute 'argmax' any help? @coeshack @hamuchiwa

NithinSS, 2017-10-01:
It is probably because your predict() returned None

closed 2018-03-19 by hamuchiwa