hamuchiwa/AutoRCCar #33: Version OpenCV

MihaiSandor, 2017-05-05:
Hello, what do you use version of opencv? And please share your schema for connections raspberry pi and arduino (sensors). Thank you so much. It”s most important version opencv.

hamuchiwa, 2017-05-08:
OpenCV: 2.4.10.1 

Connect ultrasonic sensor to Raspberry pi:
[image]

MihaiSandor, 2017-05-08:
Thank you so much!

closed 2017-05-08 by MihaiSandor

MihaiSandor, 2017-05-08:
Can you please give me a schema for Arduino and controller?

hamuchiwa, 2017-05-08:
It really depends on your controller, here is mine:
[image]

MihaiSandor, 2017-05-08:
Is like yours. In which pins the threads are inserted? In Arduino of course.

hamuchiwa, 2017-05-10:
It doesn't matter that much what pins you selects, in m case I have 6, 7, 9 10 (simply because 8 is broken). I originally planed to power the controller through Arduino, black wire connects to negative. It's not used, you can ignore that black wire.