hamuchiwa/AutoRCCar #34: Error in "stream_server_test.py"and "collect_training_data.py"

Noorulain15, 2017-06-01:
Hello...
Sir first i run "stream_server_test.py" on laptop and then "stream_client.py" on raspberry.It shows:
"connection from: 
Streaming. . .
press 'q' to exit"
and soon error occurs
"AttributeError: 'module' object has no attribute 'CV_LOAD_IMAGE_UNCHANGED' "
and when i run "collect_training_data.py" and then "stream_client.py"...It show:
collecting images. . .
and the same above AttributeError occurs again.
I've done everything in the same manner as you said so where is the issue??how to solve it??

Noorulain15, 2017-06-01:
[image]

hamuchiwa, 2017-06-02:
This might be one of the namespace changes in OpenCV 3. Back then, this project is based on OpenCV 2.4.10.1. Take a look at parameters from [here](http://docs.opencv.org/3.0-beta/modules/imgcodecs/doc/reading_and_writing_images.html#imread).

Noorulain15, 2017-06-03:
Thank You so much Sir for your reply. I'll let you know,if it works for me.

closed 2017-07-24 by Noorulain15