hamuchiwa/AutoRCCar #35: Understanding the MLP output & label format

Cosmonaut65, 2017-06-06:
Hi Hamuchiwa!

I've revisited this project in my quest to better understand machine learning and I have a question regarding the output format. 

In the project, you use a 1x4 array to determine what action the car should do. Is there a reason this can't instead be one value with a range from 0 to 3 instead?
Does the output have to be between 0 and 1?

Using the same code for the neural network as yours, if the input labels had arrays which had elements ranging from 0 to 100 (for example) to indicate turn angle and drive speed (assuming we have a car which has variable turn/speed) would the code still work? (assuming the dimensions of the array holding the labels is still 1x4)

Thanks again for this great project and all the help!

hamuchiwa, 2017-06-06:
The activation function (type of neuron) defines the output range, for example, for sigmoid function, the output is mapped to 0-1. 

I think you brought up a really interesting question, honestly I don't really exactly know why we need four labels instead of 2-bits or even 1-bit. Intuitively, simplicity represents reality better...? Of course there is also a ["ultimate justification" which is "empirical"](http://neuralnetworksanddeeplearning.com/chap1.html#a_simple_network_to_classify_handwritten_digits) (scroll down a few paragraphs).

Certainly you can use neural network for regression, then there will be no activation function in the output layer, therefore square error is used as cost function and output layer only contains one neuron.
There are two great references for in case you are interested, [here](http://www.holehouse.org/mlclass/08_Neural_Networks_Representation.html) and [here](http://neuralnetworksanddeeplearning.com/index.html).  

Actually I have the plan to collect steering angle from a real car, in this case, it's a regression problem, both input and output are continuous.

Cosmonaut65, 2017-06-12:
Thank your for your reply and the resources. So it seems the empirical evidence shows that having one output node for each possibility works better. I will try this approach with my project and see what the outcome is.
With the set up that I have, I'm planning on having 4 output parameters. One to specify whether it is left or right, one to specify whether it is forwards or reverse, another to specify on a scale of 0-100 what the turn input should be and one last one on a scale of 0-100 what the throttle input should be.

All in all, this would lead to over 200 output nodes. With that many output nodes, would I need to adjust the number of nodes in the hidden layer? I've read that the mean of the input and output nodes would be a good value to start with. Would you agree with that statement?

I'm also considering supplying all 3 colour channels into the neural network input layer. Is there any reason to believe this would reduce reliability compared to using just a black & white image?

Thanks again for the help!

hamuchiwa, 2017-06-12:
It's hard to say how many hidden layers or how many neurons should be used in the network, but I usually have more neurons in a hidden layer than input.  
Honestly speaking, throttle control is beyond my current knowledge. Most resource I found is to map input image to steering angle. This project was done in 2015, recent works start to adapting convolutional neural networks(CNNs) which is more powerful in handling image tasks. 
There is a great paper from Nvidia [End-to-End Deep Learning for Self-Driving Cars](https://devblogs.nvidia.com/parallelforall/deep-learning-self-driving-cars/), which uses CNNs to map raw color image to steering angle.

closed 2018-03-19 by hamuchiwa