hamuchiwa/AutoRCCar #36: Syntax error in mlp.training

iamEmma, 2017-06-09:
Hello sir

I am working on your project but i am new to python, and it is really great to have your codes and they are quite helpful. Sir, i am facing a syntax error while running mlp.training.

e2 = cv2.getTickCount
SyntaxError : invalid syntax

I am also attaching the snap, It would be nice to have your guidance and suggestion in this regard.

Thank you so much
Emma Rosa

iamEmma, 2017-06-09:
[image]

hamuchiwa, 2017-06-09:
Hi Emma, on line 49, add ")" to the end.

iamEmma, 2017-06-10:
Sir, i have added ")" on line 49 as you suggested, the syntax error on line 52 has been removed, but it caused another error to occur.

**line49   num_iter = model.train(train, train_labels, None, params = params)
Error  cv2.error: ..\..\..\..\opencv\modules\ml\src\ann_mlp.cpp:690: error: (-5) input training data should be a floating-point matrix withthe number of rows equal to the number of training samples and the number of columns equal to the size of 0-th (input) layer in function CvANN_MLP::prepare_to_trai**n

When I replaced the line 49, with another one, an attribute error took its place

**line 49  num_iter = model.train(np.float32(train), cv2.ml.ROW_SAMPLE, np.float32(train_labels))
Error     num_iter = model.train(np.float32(train), cv2.ml.ROW_SAMPLE, np.float32(train_labels))
AttributeError: 'module' object has no attribute 'ml'
(0, 38400)
(0, 4)
Loading image duration: 0.00151298991803
Training MLP ...

Process finished with exit code 1**

kindly suggest me something to get this code worked. And i have one more question:
what does it mean by 'module' object has no attribute 'ml'? and how can i resolve the errors like this?

I would be grateful to you for your kind response.  I need help.

Thank you sir
Emma Rosa

closed 2017-06-12 by iamEmma