hamuchiwa/AutoRCCar #38: Error in collect_training_data.py "training_data_temp" no such file and directly

iamEmma, 2017-06-10:
Hello sir

I am facing another issue, when i run collect_training_data,py, an error occurred which says: there is no such file or directory like training_data_temp/test08, and this file doesn't even exist in your repository too. Therefore i created this files, by opening up pycharmprojects in my PC and creating a new folder in a required place then again open the project in pycharm from PC.  I want to ask, if this is the correct way to create a file? or the following error in mlp_training.py is because of this problem?

**line49   num_iter = model.train(train, train_labels, None, params = params)
Error  cv2.error: ..\..\..\..\opencv\modules\ml\src\ann_mlp.cpp:690: error: (-5) input training data should be a floating-point matrix withthe number of rows equal to the number of training samples and the number of columns equal to the size of 0-th (input) layer in function CvANN_MLP::prepare_to_train**

iamEmma, 2017-06-10:
There is no such file or directory like training_data_temp in your repository
[image]

iamEmma, 2017-06-10:
There is no such folder or directory like training_data_temp

[image]

hamuchiwa, 2017-06-10:
You are correct, back then I collected train and test data separately rather than split one data set into train and test, so I had another folder as staging area then manually copy files into training_data and testing_data folders. 
Thanks for the comment, I'll combine train/validation/test into one file and update repository soon.

iamEmma, 2017-06-10:
Yes sir, Kindly update the repository as soon as possible, i am thankful to you for this. And what should i do for this error 

**line49 num_iter = model.train(train, train_labels, None, params = params)
Error cv2.error: ........\opencv\modules\ml\src\ann_mlp.cpp:690: error: (-5) input training data should be a floating-point matrix withthe number of rows equal to the number of training samples and the number of columns equal to the size of 0-th (input) layer in function CvANN_MLP::prepare_to_train**

Any soultion?

hamuchiwa, 2017-06-10:
It looks like no training data loaded. I have updated the project, collected data will be saved into "training_data" directory.

iamEmma, 2017-06-10:
Oh thanks a lot sir. I am trying the updated repository, i will inform you if any other error occurs.

iamEmma, 2017-06-11:
Hello sir

I have tried the updated repository, i have a problem regarding scikit-learn package, i have tried to install it, but it always shows a warning, i don't understand what does it mean although my model is still being trained.
i want to ask, will this error cause any other problem afterwards? is my model being trained perfectly? or i need to resolve this issue, if yes then how can i resolve this problem.

Waiting for your kind response
Thank you

**mlp_training.py

Loading training data...
C:\Python27\lib\site-packages\sklearn\cross_validation.py:44: DeprecationWarning: This module was deprecated in version 0.18 in favor of the model_selection module into which all the refactored classes and functions are moved. Also note that the interface of the new CV iterators are different from that of this module. This module will be removed in 0.20.
  "This module will be removed in 0.20.", DeprecationWarning)
Image array shape:  (49, 38400)
Label array shape:  (49, 4)
Loading image duration: 0.0887369190671
Training MLP ...
Training duration: 55.0710580002
Train accuracy:  100.00%
Test accuracy:  60.00%

Process finished with exit code 0  **
[image]

hamuchiwa, 2017-06-12:
sklearn.cross_validation was depreciated in 0.18. I have changed it to sklearn.model_selection module to removes the warning.

iamEmma, 2017-06-14:
Hello sir,

There is an error i am getting in Ultrasonic_client.py, i don't know what does it say, Please help
[image]

Thanks a lot
Emma Rosa

hamuchiwa, 2017-06-14:
It seems an indentation error, just make sure on each indentation level, you have enough and consistent number of spaces at the beginning.

iamEmma, 2017-06-16:
Thank you so much sir, for your kind replies. I wouldn't have done this without your help.
I am grateful to you

Regards
Emma Rosa

hamuchiwa, 2017-06-16:
No problem, this is what is open source for. If you have made any improvements, please feel free to drop some codes.

iamEmma, 2017-06-16:
No sir, i am new to python, i don't know much about it therefore i didn't make any changes. All you did per my request was enough. I got all my errors resolved with your help.

Thanks.

closed 2017-06-21 by iamEmma

kxrf97, 2019-03-06:
@iamEmma  did you manage to run  your car well on track?