hamuchiwa/AutoRCCar #39: What RC cars are people using?

094459, 2017-06-19:
Not really an issue, but I have got everything working in this project but I am struggling with one thing. The car will not move!!

It looks like it is down to weight / poor choice in RC car.

I have tried pairing down, and am just waiting for the PiZero motor hat before I swap from Pi3 to PiZero, but suspect that it will still be too much weight. 

So what are other people using? I might need to go out and buy something more powerful.

MayusYT, 2018-11-13:
What car did you use lastly?

krukhlis, 2018-11-13:
In order to understand if the overweight of your car+platform is causing the problem -- just lift it by hand. If you see that motor rotates your wheels based on directives of your robot -- yes, the issue is weak motor/servos. If there are no changes -- it's not your RC car issues, but yours-- you have wired/programmed something wrong.