hamuchiwa/AutoRCCar #40: please help!!

ahmaddaghstani, 2017-06-23:
hi
i just keep having this error whenever i run "stream_server_test.py" it shows me this error 

" Traceback (most recent call last):
  File "C:\Users\Suzy\Desktop\AutoRCCar-master\test\stream_server_test.py", line 44, in <module>
    VideoStreamingTest()
  File "C:\Users\Suzy\Desktop\AutoRCCar-master\test\stream_server_test.py", line 12, in __init__
    self.server_socket.bind(('192.168.1.24',5400))
  File "C:\Python27\lib\socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 10049] The requested address is not valid in its context "

what's the proplem?!

094459, 2017-06-23:
You need to make sure the IP address of the computer you are using as well as the raspberry Pi are set correctly.

If your laptop is 192.168.1.100 and your Pi is 192.168.1.200, then in the stream_server_test you need to make sure you have it set to 192.168.1.100 and you configure the client on the Pi to use that address too.

Also, I did initially try Windows and then gave up. My firewall was blocking ports so you should check that too.

ahmaddaghstani, 2017-06-24:
thank you very much for your help
the fire wall was the proplem so i turned it off and boom ... it's working

094459, 2017-06-24:
Cool. dont forget to close this issue.

closed 2017-06-25 by ahmaddaghstani