hamuchiwa/AutoRCCar #42: Issues with the Neural network training

Abdalrazak-Mansour, 2017-06-26:
I was wondering if I could get the network trained please, I need it for my year project soon.

hamuchiwa, 2017-06-27:
First you need to collect good quality data, then you can use mlp_training.py to start training.

Abdalrazak-Mansour, 2017-07-16:
I'm running the collect training data program but I can't figure out how to control the car using the arrow's keypresses.
I get the video streaming window and it starts saving pictures automatically without me doing anything
Can you show me a video of how you trained it,

Abdalrazak-Mansour, 2017-07-16:
you said: "frames are saved only when there is a key press action"
but as soon as I start the collect_training_data it starts saving frames in the training_images folder.

hamuchiwa, 2017-07-17:
Training data is saved as numpy array format, frames are saved only if you press an arrow key. Actually images are just for your reference, you could comment out line 65 so no images will be saved into the folder.

Abdalrazak-Mansour, 2017-07-17:
thanks, it's working now, I had to make this adjustment:
`import numpy as np
import cv2
import serial
import pygame
from pygame.locals import *
import socket
import time
import os
pygame.init()
display = pygame.display.set_mode((100,100))

class CollectTrainingData(object):`

closed 2018-03-19 by hamuchiwa