hamuchiwa/AutoRCCar #43: Some questions about the camera calibration

MigicMice, 2017-07-15:
hi,man. Your project is awesome. I wanna make it by myself,but I have some problems on camera calibration.The "picam_calibration.py" should run on pi or computer? It seems to run on pi but u put it in the "computer" folder,that confuse me.Hoping ur reply.

hamuchiwa, 2017-07-17:
Certainly you can run it on raspberry pi. It's just because in this project, pi only handles streaming, so naturally calibration is done on computer.

MigicMice, 2017-07-18:
多谢回复，我就用中文好了，233.我拍了棋盘的照片，跑了相机标定的代码但是报错了。
`/System/Library/Frameworks/Python.framework/Versions/2.7/bin/python2.7 /Users/mice/PycharmProjects/RC_Car/picam_calibration.py
OpenCV Error: Assertion failed (nimages > 0) in calibrateCamera, file /tmp/opencv-20170224-1869-10nlf6f/opencv-2.4.13.2/modules/calib3d/src/calibration.cpp, line 3417
Traceback (most recent call last):
  File "/Users/mice/PycharmProjects/RC_Car/picam_calibration.py", line 47, in <module>
    retval, cameraMatrix, distCoeffs, rvecs, tvecs = cv2.calibrateCamera(object_points, image_points, (w, h), None, None)
cv2.error: /tmp/opencv-20170224-1869-10nlf6f/opencv-2.4.13.2/modules/calib3d/src/calibration.cpp:3417: error: (-215) nimages > 0 in function calibrateCamera

Process finished with exit code 1
`ps.照片上棋盘的背景是黄色的椅子还有杂物，会不会有影响？

hamuchiwa, 2017-07-19:
代码其实是从[opencv文档](http://opencv-python-tutroals.readthedocs.io/en/latest/py_tutorials/py_calib3d/py_calibration/py_calibration.html)直接搬过来的，标定的时候会对每张图片做corner检测，找到的话会显示出来。会不会是文件路径问题或者是格子数和设定的不一样。

MigicMice, 2017-07-19:
我开始没做灰度，做了灰度后报错不一样了，（我的棋盘是8*8的，涉及到的全改成了8*8，）`
OpenCV Error: Assertion failed (nimages > 0) in calibrateCamera, file /tmp/opencv-20170224-1869-10nlf6f/opencv-2.4.13.2/modules/calib3d/src/calibration.cpp, line 3417
Traceback (most recent call last):
  File "/Users/mice/PycharmProjects/RC_Car/picam_calibration.py", line 47, in <module>
    retval, cameraMatrix, distCoeffs, rvecs, tvecs = cv2.calibrateCamera(object_points, image_points, (w, h), None, None)
cv2.error: /tmp/opencv-20170224-1869-10nlf6f/opencv-2.4.13.2/modules/calib3d/src/calibration.cpp:3417: error: (-215) nimages > 0 in function calibrateCamera

Process finished with exit code 1
`刚接触OpenCV，不太懂，麻烦了。
PS.（github应该怎么叫，hub主吗）hub主实在是nice，两年多的项目一直回复，能加个联系方式吗，膜拜大佬

hamuchiwa, 2017-07-19:
OpenCV自己带的（格子图片+文档）有没有试过，看看报不报错。文档里用的是6x7，我用的是6x9。
其实我还在摸索阶段啦，咱们一起互相学习互相进步吧。方便的话可以留个邮箱或是微信，我加上你。

MigicMice, 2017-07-20:
哈哈哈 吼啊 微信：MXH1024xx

closed 2018-03-19 by hamuchiwa