hamuchiwa/AutoRCCar #44: RC Car Speed Control

ss1306, 2017-07-29:
Firstly, great work with the project and on the extensive documentation you have provided. I am making a similar project taking inspiration from your work. I have managed to successfully collect training data and train my network weights. But, when I let the car run for itself, its moving too quickly (i.e. its got too much speed) and ends up moving out of the track before its got a chance to correct itself Did you face a similar problem? If so, how did you slow down the car. I see from your video that your car is moving in a more slow and controlled manner. I'm failing to achieve that with my car. Any pointers or advice would be very helpful

Furthermore, from my NN training (of layers 38400, 32, 1) I get a train error % ~ 2% and test set error % ~ 15%. Could it be that my network isn't generalising very well? I may have to move the sides of the road a bit closer since they are a bit widely spread at the moment?

ss1306, 2017-07-29:
I see from https://github.com/hamuchiwa/AutoRCCar/issues/5 that someone already raised this issue. You mentioned that you used the potentiometer and alternatively a rubber band to slow down the speed. Could you please elaborate how you used both of them? Potentiometer for each wire going to Forward/ Reverse pins?

hamuchiwa, 2017-07-31:
Hi, 15% error rate is not too bad. You may also use all pixels (320x240) for training data which generates better results in my experience. 
Either one works, I took a short cut and wrap a rubber band on the wheel to increase friction. If you plan to use a potentiometer, pick a small one.

ss1306, 2017-07-31:
I decided to increase the # units in my hidden layer (32 -> 64) and the # iterations from 100 -> 400. Additionally, I also added PWM for motor speed control. That seemed to have resolved most of the issues I was having. However, my car still behaves oddly every now and then on the track. Did you perform any model selection on your Network to decided your hyperparam values? Also, do you remember how much data you collected for training, and how you decided if you needed more data or if the amount was sufficient enough? I currently have around 2500 frames.

hamuchiwa, 2017-08-06:
Not really, the track pattern is simple, most default parameters perform well in my case. I only collect a couple of hundreds. In general you might want to balance the data, for example, there are roughly equal amount of straight lines and corners. 
There are other people tried this project and had success with a longer track than mine, here is one with a [circular track](https://www.youtube.com/watch?v=KrvvmLEQk8o) which looks pretty good.

farazhussain2015, 2017-08-10:
Use motor driver instead of RC to control speed of motors by PWM function.

closed 2017-08-31 by ss1306

AnoRebel, 2019-06-17:
@hamuchiwa Sorry, but can you explain to me exactly how you used the rubber band to decrease the car speed, it seems like the only feasible method for me.. Please help..

hamuchiwa, 2019-06-17:
@AnoRebel Something you can try, wrap a rubber band around the axle or the wheel to make it harder to turn, or use a potentiometer.

AnoRebel, 2019-06-17:
Ok, thank you, also, i'm using a really bad webcam(was all i could find), so i figured i should use the whole image during training, how do i go about it?? What do i change in the code?? Or do you have a better alternative??

AnoRebel, 2019-06-17:
Also, one more thing, do you have a sample code of streaming using a webcam, i cant seem to find good code, using imagezmq fails when converting the images with numpy(i believe), so i'm using UDP, but on changing the rc_drive.py to use the UDP request handler, the program just closes itself with no error.. Any ideas or suggestions??

hamuchiwa, 2019-06-17:
@AnoRebel If you'd like to use the whole image, remove slicing and save the whole images as your training data, also make sure the dimensions match in training. I don't have the code of using a webcam. Take a look at #103, see if it helps.

AnoRebel, 2019-06-17:
Sorry, but where and how exactly do i remove the slicing??