hamuchiwa/AutoRCCar #46: Camera placement for distance measurement

Noorulain15, 2017-08-17:
Hey sir!

I have a question,  does camera placement affects the distance measured by raspberry pi? If yes, then how should I place the Camera to get accurate distance measurement. 

Waiting for your response. Thank you so much.

Muhammed287, 2017-08-17:
Sorry . I don't understand it ??

hamuchiwa, 2017-08-18:
@Noorulain15 Yes, from the equation, the height, angle affect the calculation. But please keep in mind it's not accurate, if your intention is to measure distance, you might want to chose stereo vision.

Noorulain15, 2017-08-24:
@hamuchiwa sir I have one more question, what kind of features does haar cascade classifier extract from a traffic light and stop sign for specific detection of both? 

Waiting for your response sir.
Thanks

hamuchiwa, 2017-08-25:
@Noorulain15 Please refer to [opencv tutorial](http://docs.opencv.org/3.0-beta/doc/py_tutorials/py_objdetect/py_face_detection/py_face_detection.html#face-detection). It works in the same way.

Noorulain15, 2017-09-04:
Respected sir

While reading your blog and understanding the neural network, i am stuck on a line which is " Only lower half of the input image is used for training and prediction purposes". Sir can you please explain what does you mean by "Lower half input images"?

Thanks

hamuchiwa, 2017-09-15:
@Noorulain15 Draw a line from left to right and split the image into upper and lower parts. You could also just take the entire image as input which works as well.

closed 2018-03-19 by hamuchiwa