hamuchiwa/AutoRCCar #47: Delay Response of Arduino to control motors Plz Help

AiBuddy, 2017-08-19:
I am using arduino and motor driver l238N , while executing collect training data file i am facing problem of delay response from arduino. The lights of arduino TX/RX are on continuously while executing code. Motors are moving about after 5 to 8 seconds of key press. please help me as quick as possilble.. thank you in advance

hamuchiwa, 2017-08-23:
Hi, I'm not familiar with motor shield, do you see any delay using the rc_control_test.py file?

AiBuddy, 2017-08-23:
thank you for reply . i am back to remote control now. i dont use motor driver shield anymore . but i have another problem now. I cant control my rc car if i plug both right and left buttons. Once i remove any left or right wire from arduino . its works like charm. why is it happening? is it because my remote control architecture may be different from yours?

AiBuddy, 2017-08-23:
everything is working now but i am facing delay in training file. when i hit forward key my car is moving after some seconds. any solution ? @hamuchiwa

AiBuddy, 2017-08-23:
i fix it by myself. one more question. how big should be my npz file for test? how much should i train my car. i have no curve path. i am only training it on straight path. will it effect automation?

hamuchiwa, 2017-08-25:
If it's just a straight line, then you don't need much data. You can try it out to see how many images is good enough.

xmoose, 2017-09-07:
hi guys ,@Aibuddy how did you manage to get rid of the delay?even i am having 2..3s delay..any help appreciated

AiBuddy, 2017-09-08:
@xmoose  try to put your remote antenna at 90 degree

AiBuddy, 2017-09-08:
@xmoose check if there is any unwanted input in arduino code

xmoose, 2017-09-08:
@AiBuddy did you complete this project? i'm having issues running the RC_driver.py did you run it?

AiBuddy, 2017-09-09:
what issues you are having in rc_driver.py . post errors or issues @xmoose

M-Salman1337, 2017-10-05:
hey AiBuddy, do u know how to interface arduino controls for the car into python program?

closed 2018-06-29 by hamuchiwa