hamuchiwa/AutoRCCar #48: How did you stream from Pi camera using Python?

M-Salman1337, 2017-09-04:
**Trying to be successful to get the streaming part correct for now.**
I am using '**Motion**' to stream from pi camera onto an ip address... now I changed the IP address(192.168.0.6:8081) in the 2 codes.. (_stream_server_test.py_ which runs on laptop and _stream_client.py_ which runs on RPi) I am getting errors on both the programs..

I want to know what software you used in Pi to stream on the IP address? I used Motion and it clearly doesn't work :( Please help

AiBuddy, 2017-09-04:
what errors are u getting ? post them here @M-Salman1337

M-Salman1337, 2017-09-05:
I first make sure I am streaming on the ip then I run the computer and then the pi application (but computer and pi programs are giving errors)
Error on computer:
Traceback (most recent call last):
  File "C:/Users/tiger/Desktop/take_the_stream_from_pi.py", line 43, in <module>
    VideoStreamingTest()
  File "C:/Users/tiger/Desktop/take_the_stream_from_pi.py", line 11, in __init__
    self.server_socket.bind(('192.168.0.6', 8081))
OSError: [WinError 10049] The requested address is not valid in its context

Error on Pi: 
mmal: mmal_vc_port_enable: failed to enable port vc.null_sink:in:0(OPQV): ENOSPC
mmal: mmal_port_enable: failed to enable connected port (vc.null_sink:in:0(OPQV))0x11c2330 (ENOSPC)
mmal: mmal_connection_enable: output port couldn't be enabled
Traceback (most recent call last):
  File "stream_from_pi.py", line 21, in <module>
    with picamera.PiCamera() as camera:
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 433, in __init__
    self._init_preview()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 513, in _init_preview
    self, self._camera.outputs[self.CAMERA_PREVIEW_PORT])
  File "/usr/lib/python3/dist-packages/picamera/renderers.py", line 558, in __init__
    self.renderer.inputs[0].connect(source).enable()
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 2212, in enable
    prefix="Failed to enable connection")
  File "/usr/lib/python3/dist-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to enable connection: Out of resources

What did u use to stream and use his code ?

M-Salman1337, 2017-09-05:
@hamuchiwa  Please reply how did u manage to use that code? like what streamer did u use in the Raspberry Pi

M-Salman1337, 2017-09-05:
I used this code after some corrections...

```
import numpy as np
import cv2
import socket

class VideoStreamingTest(object):
    def __init__(self):

        #self.server_socket = socket.socket()
        #self.server_socket.bind(('192.168.0.6', 8081))
        #self.server_socket.listen(0)
        
        #self.connection, self.client_address = self.server_socket.accept()
        #self.connection = self.connection.makefile('rb')
        #self.streaming()
        self.socket = socket.socket()
        self.connection = self.socket.connect(('192.168.0.6', 8081))
        #self.socket.connect(('192.168.0.6', 8081))
        self.streaming()

    def streaming(self):

        try:
            #print ("Connection from: ", self.client_address)
            print ("Streaming...")
            print ("Press 'q' to exit")

            stream_bytes = b' '
            while True:
                
                stream_bytes += self.socket.recv(1024)
                first = stream_bytes.find(b'\xff\xd8')
                last = stream_bytes.find(b'\xff\xd9')
                if first != -1 and last != -1:
                    jpg = stream_bytes[first:last + 2]
                    stream_bytes = stream_bytes[last + 2:]
                    #image = cv2.imdecode(np.fromstring(jpg, dtype=np.uint8), cv2.IMREAD_GRAYSCALE)
                    image = cv2.imdecode(np.fromstring(jpg, dtype=np.uint8), cv2.IMREAD_COLOR)
                    cv2.imshow('image', image)

                    if cv2.waitKey(1) & 0xFF == ord('q'):
                        break
        finally:
            #self.connection.close()
            self.socket.close()

if __name__ == '__main__':
    VideoStreamingTest()

```
Will ask further doubts as they arise!

M-Salman1337, 2017-09-05:
**# Answer these 3 questions please! :-)**
**1**.But why do you have 2 programs running 1 from pi and 1 on computer? 
**2**.and also why `self.connection = self.connection.makefile('rb')` is this required?
**3**. (are you maybe saving stream on the computer? I'm not sure..)
I am currently always set stream option when pi turns on.. at 192.168.0.6:8081
and I run the above code to get it with python/OpenCV. (So turn pi on and run 1 program on computer)
Please give your insights!

closed 2017-09-05 by M-Salman1337

AiBuddy, 2017-09-06:
@M-Salman1337  
1. One program is for server which in our case is our pc and other is client which is pi. We are getting streams of images from client and get them through server
2.makefile means to use the program make which updates targets from newer dependencies, rb is for reading it in binary mode where r is for reading and b is for binary mode because we are expecting data in bytes.
3. Yes we can save stream, we are already saving them as images. See code of training in computer folder.

In your case 192.168.06 is your local ip. if you are on windows type ipconfig in cmd to confirm your local ip. 8081 is port which u have to open from firewall if you are using windows. i am not sure about others OSes.