hamuchiwa/AutoRCCar #49: Writing 1,800 jpeg images of streamed video in 1 minute ?

M-Salman1337, 2017-09-06:
```
import numpy as np
import cv2
import socket

class VideoStreamingTest(object):
    def __init__(self):
        
        # self.server_socket = socket.socket()
        # self.server_socket.bind(('192.168.0.6', 8081))
        # self.server_socket.listen(0)
        # self.connection, self.client_address = self.server_socket.accept()
        # self.connection = self.connection.makefile('rb')
        # self.streaming()
        # self.socket.connect(('192.168.0.6', 8081))
        
        self.socket = socket.socket()
        self.connection = self.socket.connect(('192.168.0.6', 8081))
        self.streaming()

    def streaming(self):

        try:
            #print ("Connection from: ", self.client_address)
            print ("Streaming...")
            print ("Press 'q' to exit")
            frame = 0
            stream_bytes = b' '
            while True:
                
                stream_bytes += self.socket.recv(1024)
                first = stream_bytes.find(b'\xff\xd8')
                last = stream_bytes.find(b'\xff\xd9')
                if first != -1 and last != -1:
                    jpg = stream_bytes[first:last + 2]
                    stream_bytes = stream_bytes[last + 2:]
                    # Grayscale image
                    #image = cv2.imdecode(np.fromstring(jpg, dtype=np.uint8), cv2.IMREAD_GRAYSCALE)
                    # Color image
                    image = cv2.imdecode(np.fromstring(jpg, dtype=np.uint8), cv2.IMREAD_COLOR)
                    # To write/save the frames
                    # cv2.imwrite("stream_images/frame_%d.jpg" % count, image)
                    cv2.imwrite('stream_images/frame{:>05}.jpg'.format(frame), image)
                    
                    cv2.imshow('image', image)
                    
                    frame += 1
                    
                    if cv2.waitKey(1) & 0xFF == ord('q'):
                        break
        finally:
            #self.connection.close()
            self.socket.close()

if __name__ == '__main__':
    VideoStreamingTest()

```
Thats the code im using.... now I'm able to have 1800 images(frames) saved in 1 minute, is this correct? everytime I run the program I over write them, is that correct as well?

hamuchiwa, 2017-09-15:
Hi, I was only able to save about 10~15 frames/second. Yes, files are overwritten each time.

M-Salman1337, 2017-09-15:
First of all thanks for taking the time to reply :-) 
what algorithm did u use for the machine learning part ?

hamuchiwa, 2017-09-16:
neural network

M-Salman1337, 2017-09-16:
[image]
reso 320x240 , 50fps 
I am trying to get rid of the date and time.. but I am unable to.. do u think I can make it work with the date and time on the frames?

hamuchiwa, 2017-09-16:
How did you add the date and time to frames?

M-Salman1337, 2017-09-16:
its from the motion.conf file given by the webserver for pi camera

R0na1d, 2017-09-24:
Hello, I want to make this self driving car for my college project . Please help me how to do it...please guide me on this.

evanrich, 2017-10-03:
@R0na1d copying someone's work isn't a college project, it's copying someone's work.  if you can't  figure out from the instructions, you might want to pick a different project...like a baking soda volcano...

closed 2017-10-05 by M-Salman1337