hamuchiwa/AutoRCCar #50: How to get the commands from Arduino into Python program ?

M-Salman1337, 2017-10-05:
Hey, so after getting the car to work with (in my case i used A W S D for movement) how do u interface Arduino controls with the python program running with the streamed images converted to numpy arrays ?

hamuchiwa, 2017-10-06:
If you know the size of the object in advance, then yes. You can calculate the distance by geometric relationships.

M-Salman1337, 2017-10-07:
Oh wait , so the calibration is for the stop sign/signal part? and not for the car actually moving accross the tracks? :o

hamuchiwa, 2017-10-10:
correct, it's for measuring the distance between stop sign and camera.

M-Salman1337, 2017-10-10:
ohh alright thanks! And..

The pygame interface program (rc_car_test.py) isnt running my RC car with COM3 as port and 9600 baud rate... heres the program..

```
import serial
import pygame
from pygame.locals import *

class RCTest(object):
    def __init__(self):
        pygame.init()
        screen = pygame.display.set_mode((400, 300))
        self.ser = serial.Serial('COM3', 9600, timeout = 1)
        self.send_inst = True
        self.steer()
    
    def steer(self):
        
        while self.send_inst:
            for event in pygame.event.get():
                print('entered for loop')
                if event.type == KEYDOWN:
                    
                    key_input = pygame.key.get_pressed()
                    # complex orders
                    if key_input[pygame.K_UP] and key_input[pygame.K_RIGHT]:
                        print("Forward Right")
                        self.ser.write(chr(6))
                    
                    elif key_input[pygame.K_UP] and key_input[pygame.K_LEFT]:
                        print("Forward Left")
                        self.ser.write(chr(4))

                    # simple orders
                    elif key_input[pygame.K_UP]:
                        print("Forward")
                        self.ser.write(chr(8))
                    
                    elif key_input[pygame.K_DOWN]:
                        print("Reverse")
                        self.ser.write(chr(2))

                    elif key_input[pygame.K_x] or key_input[pygame.K_q]:
                        print('Exit')
                        self.send_inst = False
                        self.ser.write(chr(0))
                        self.ser.close()
                        break
                
                elif event.type == pygame.KEYUP:
                     self.ser.write(chr(0))

if __name__ == '__main__':
    RCTest()
```

M-Salman1337, 2017-10-10:
Like the only difference is COM3 / 9600 / and no reverse left , no reverse right.
[Arduino is running @COM3 and @9600 baud rate...]
this program opens up pygame interface but pressing arrowkeys dont do anything to the car (it should operate ideally with arrow keys)

hamuchiwa, 2017-10-10:
Does it display the output string in the console? Do the char-s sending to Arduino match the pin numbers defined in ino file?

M-Salman1337, 2017-10-11:
Nope, it doesn't even display in console... the ino program that runs the RC car with Arduino works perfectly on COM3... its just the python file

M-Salman1337, 2017-10-11:
Altho it says Forward and Reverse for those 2 cases, the car however doesn't move.

M-Salman1337, 2017-10-11:
I changed the program.. It now displays output string in the console like this.... but theres nothing happening to the car...like its not responding to the commands but I see Arduino LED sending commands...
https://gyazo.com/c4284a9bc6a89ae9836149c9ea20604d
[screenshot of the output strings in the console]

I really do appreciate the time you are taking to reply me @hamuchiwa :)

M-Salman1337, 2017-10-11:
I got it working now :)

M-Salman1337, 2017-10-11:
how many laps or how many times did you train the machine learning model? or neural network

9600, 2017-10-11:
@M-Salman1337 you called?

M-Salman1337, 2017-10-11:
omg sorry I used @ as in 'at'.... my bad

hamuchiwa, 2017-10-11:
Training is done in OpenCV, if your track is short then you don't need many images, even 100 images can work fine. Here is a video from aha124226675 showing self-driving: https://www.youtube.com/watch?v=KrvvmLEQk8o

M-Salman1337, 2017-10-11:
```
class CollectTrainingData(object):
    def __init__(self):
        
        self.socket = socket.socket()
        self.connection = self.socket.connect(('192.168.0.6', 8081))
        # self.streaming()
        
        
        # connect to a seral port
        self.ser = serial.Serial('COM3', 9600, timeout = 1)
        self.send_inst = True
        
        # create labels
        self.k = np.zeros((4, 4), 'float')
        for i in range(4):
            self.k[i, i] = 1
        self.temp_label = np.zeros((1, 4), 'float')
        
        pygame.init()
        screen = pygame.display.set_mode((400, 300))
        self.collect_image()
    
    def collect_image(self):
        
        saved_frame = 0
        total_frame = 0
        
        # collect images for training
        print('Start collecting images...')
        e1 = cv2.getTickCount()
        image_array = np.zeros((1, 38400))
        label_array = np.zeros((1, 4), 'float')
        
        # stream video frames one by one
        try:
            stream_bytes = b' '
            frame = 1
            while self.send_inst:
                stream_bytes += self.socket.recv(1024)
                first = stream_bytes.find(b'\xff\xd8')
                last = stream_bytes.find(b'\xff\xd9')
                if first != -1 and last != -1:
                    jpg = stream_bytes[first:last + 2]
                    stream_bytes = stream_bytes[last + 2:]
                    image = cv2.imdecode(np.fromstring(jpg, dtype = np.uint8), cv2.IMREAD_COLOR)
                    
                    # select lower half of the image
                    roi = image[120:240, :]
                    
                    # save streamed images
                    cv2.imwrite('training_images/frame{:>05}.jpg'.format(frame), image)
                    
                    # cv2.imshow('roi_image', roi)
                    cv2.imshow('image', image)
                    
                    # reshape the roi image into one row array
                    temp_array = roi.reshape(1, 38400).astype(np.float32)
                    
                    frame += 1
                    total_frame += 1
                    
                    # get input from human driver
                    for event in pygame.event.get():
                        if event.type == KEYDOWN:
                            key_input = pygame.key.get_pressed()
                            
                            # complex orders
                            if key_input[pygame.K_RIGHT]:
                                print("Forward Right")
                                image_array = np.vstack((image_array, temp_array))
                                label_array = np.vstack((label_array, self.k[1]))
                                saved_frame += 1
                                self.ser.write("6".encode())
                            
                            elif key_input[pygame.K_LEFT]:
                                print("Forward Left")
                                image_array = np.vstack((image_array, temp_array))
                                label_array = np.vstack((label_array, self.k[0]))
                                saved_frame += 1
                                self.ser.write("4".encode())
                            
                            
                            # simple orders
                            elif key_input[pygame.K_UP]:
                                print("Forward")
                                saved_frame += 1
                                image_array = np.vstack((image_array, temp_array))
                                label_array = np.vstack((label_array, self.k[2]))
                                self.ser.write("8".encode())
                            
                            elif key_input[pygame.K_DOWN]:
                                print("Reverse")
                                saved_frame += 1
                                image_array = np.vstack((image_array, temp_array))
                                label_array = np.vstack((label_array, self.k[3]))
                                self.ser.write("2".encode())
                            
                            
                            elif key_input[pygame.K_x] or key_input[pygame.K_q]:
                                print('exit')
                                self.send_inst = False
                                self.ser.write("0".encode())
                                break
                        
                        elif event.type == pygame.KEYUP:
                            self.ser.write("0".encode())
            
            # save training images and labels
            train = image_array[1:, :]
            train_labels = label_array[1:, :]
            
            # save training data as a numpy file
            file_name = str(int(time.time()))
            directory = "training_data"
            if not os.path.exists(directory):
                os.makedirs(directory)
            try:
                np.savez(directory + '/' + file_name + '.npz', train = train, train_labels = train_labels)
            except IOError as e:
                print(e)
            
            e2 = cv2.getTickCount()
            # calculate streaming duration
            time0 = (e2 - e1) / cv2.getTickFrequency()
            print('Streaming duration:', time0)
            
            print(train.shape)
            print(train_labels.shape)
            print('Total frame:', total_frame)
            print('Saved frame:', saved_frame)
            print('Dropped frame', total_frame - saved_frame)
        
        finally:
            self.connection.close()
            self.server_socket.close()

if __name__ == '__main__':
    CollectTrainingData()
```

After running this it says "Start collecting images..."
but then it says ValueError: cannot reshape array of size 115200 into shape (1,38400)

M-Salman1337, 2017-10-11:
I'm able to control the car with arrow keys and view the camera in grayscale..... but It is **not** saving the training_images and training_data..

M-Salman1337, 2017-10-11:
For mlp_training.py ... I am not getting mlp.xml because theres error with ANN_MLP..

AttributeError: module 'cv2.cv2' has no attribute 'ANN_MLP'

maybe new version of OpenCV uses different than ANN_MLP ?

closed 2017-11-08 by M-Salman1337