hamuchiwa/AutoRCCar #52: Stream.client.py / stream_server_test.py connection issue

bryce0515, 2017-11-02:
After uploading the correct files and connecting the PI and PC to the same I.P. I am getting the following error:

runfile('C:/Users/bsimmons/Stream_test.py', wdir='C:/Users/bsimmons')
Connection from:  ('192.168.43.130', 56770)
Streaming...
Press 'q' to exit
Traceback (most recent call last):

  File "<ipython-input-4-7e98b7b95164>", line 1, in <module>
    runfile('C:/Users/bsimmons/Stream_test.py', wdir='C:/Users/bsimmons')

  File "C:/Users/bsimmons/Anaconda3/envs/virtual_platform/lib/site-packages/spyder/utils/site/sitecustomize.py", line 710, in runfile
    execfile(filename, namespace)

  File "C:/Users/bsimmons/Anaconda3/envs/virtual_platform/lib/site-packages/spyder/utils/site/sitecustomize.py", line 101, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Users/bsimmons/Stream_test.py", line 44, in <module>
    VideoStreamingTest()

  File "C:/Users/bsimmons/Stream_test.py", line 16, in __init__
    self.streaming()

  File "C:/Users/bsimmons/Stream_test.py", line 27, in streaming
    stream_bytes += self.connection.read(1024)

TypeError: Can't convert 'bytes' object to str implicitly

hamuchiwa, 2017-11-05:
Can you post the error on raspberry pi as well?

lakshand8, 2018-01-12:
Hi @hamuchiwa  I am getting a similar error when i try to run the stream test file.
PC:
Connection from:  ('192.168.0.151', 33762)
Streaming...
Press 'q' to exit
Traceback (most recent call last):
  File "C:/Users/ldedi/Documents/GitHub/AutoRCCar/test/stream_server_test.py", line 44, in <module>
    VideoStreamingTest()
  File "C:/Users/ldedi/Documents/GitHub/AutoRCCar/test/stream_server_test.py", line 16, in __init__
    self.streaming()
  File "C:/Users/ldedi/Documents/GitHub/AutoRCCar/test/stream_server_test.py", line 34, in streaming
    image = cv2.imdecode(np.fromstring(jpg, dtype=np.uint8), cv2.CV_LOAD_IMAGE_UNCHANGED)
AttributeError: 'module' object has no attribute 'CV_LOAD_IMAGE_UNCHANGED'

Pi:
Traceback (most recent call last):
  File "stream_client.py", line 40, in <module>
    connection.close()
  File "/usr/lib/python2.7/socket.py", line 283, in close
    self.flush()
  File "/usr/lib/python2.7/socket.py", line 307, in flush
    self._sock.sendall(view[write_offset:write_offset+buffer_size])
socket.error: [Errno 32] Broken pipe

Any idea why that is?

lakshand8, 2018-01-12:
never mind, turned out in OpenCV 3, cv2.CV_LOAD_IMAGE_UNCHANGED changed to cv2.IMREAD_UNCHANGED.

closed 2018-03-19 by hamuchiwa