hamuchiwa/AutoRCCar #53: Hi! I have a some questions

kwon9man, 2017-11-03:
Hi, I am very impressed with your project. So I am also trying your project.
There are some questions.

1. In my view, your RCCar system seems to give RaspberryPi and Arduino each command on the computer. Is RaspberryPi sending ultrasound sensor and image information to computer, Computer telling Arduino according to the received information and Arduino controlling RCCar through remote control?

2. If the first question is correct, I control four DC motors with my RCCar motor as the motor driver (motor-motor driver-Arduino). So my Arduino should be attached to my RC car. In this case, do I have to connect to my computer using the Bluetooth module? Or do I have to issue commands in serial communication with RaspberryPi?

3. After finishing the neural network learning on the practice track, should I study again when driving to the practice track in another place?

I will be waiting for your answer.

P.S. The sentence may be awkward because it was a question by a translate.

hamuchiwa, 2017-11-05:
Hi, your first question is correct, actually there are many people have successfully tried using raspberry pi solely to control rc car, so arduino can be eliminated if you wish. As for neural network learning, a well trained model should be generalized well, if it doesn't perform well on another track then it implies your model is overfitting.

kwon9man, 2017-11-05:
@hamuchiwa 
Thank you for answer.
I have one more problem.
collect_training_data.py/stream_client.py Error.

[collect_training_data.py error]
Traceback (most recent call last):
  File "C: /Users/Administrator/Desktop/AutoRCCar-master/computer/collect_training_data.py", line 176, in <module>
    CollectTrainingData ()
  File "C: /Users/Administrator/Desktop/AutoRCCar-master/computer/collect_training_data.py", line 39, in __init__
    self.collect_image ()
  File "C: /Users/Administrator/Desktop/AutoRCCar-master/computer/collect_training_data.py", line 63, in collect_image
    image = cv2.imdecode (np.fromstring (jpg, dtype = np.uint8), cv2.CV_LOAD_IMAGE_GRAYSCALE)
AttributeError: 'module' object has no attribute 'CV_LOAD_IMAGE_GRAYSCALE'

[stream_client.py error]
Traceback (most recent call last):
    File "stream_client.py", line 40, in <module>
        connection.close ()
    File "/usr/lib/python2.7/socket.py", line 279, in close
        self.flush ()
    File "/usr/lib/python2.7/socket.py", line 303, in flush
        self._sock.sendall (view [write_offset: write_offset + buffer_size])
socket.error: [Errno 32] Broken pipe

Is there a problem with the module?
I don't know why.

hamuchiwa, 2017-11-05:
It's due to version difference. In OpenCV 3, image reading [flags](https://docs.opencv.org/3.0-beta/doc/py_tutorials/py_gui/py_image_display/py_image_display.html) have been changed.

kwon9man, 2017-11-05:
Hi! @hamuchiwa 
OK, finally I solve this problem. 
I used OpenCV3, so i changed code 'cv2.CV_LOAD_IMAGE_GRAYSCALE' -> 'cv2.IMREAD_GRAYSCALE'

Thank you for your reply : )

kwon9man, 2017-11-06:
@hamuchiwa 

I'm sorry for many questions.
If I don't use traffic signals, I don't know which part to remove from rc_driver.py
So I deleted what I thought was a traffic signal.
Can you take a look at it?

`import threading
import SocketServer
import serial
import cv2
import numpy as np
import math

# distance data measured by ultrasonic sensor
sensor_data = " "

class NeuralNetwork(object):
    def __init__(self):
        self.model = cv2.ANN_MLP()

    def create(self):
        layer_size = np.int32([38400, 32, 4])
        self.model.create(layer_size)
        self.model.load('mlp_xml/mlp.xml')

    def predict(self, samples):
        ret, resp = self.model.predict(samples)
        return resp.argmax(-1)

class RCControl(object):
    def __init__(self):
        self.serial_port = serial.Serial('/dev/tty.usbmodem1421', 115200, timeout=1)

    def steer(self, prediction):
        if prediction == 2:
            self.serial_port.write(chr(1))
            print("Forward")
        elif prediction == 0:
            self.serial_port.write(chr(7))
            print("Left")
        elif prediction == 1:
            self.serial_port.write(chr(6))
            print("Right")
        else:
            self.stop()

    def stop(self):
        self.serial_port.write(chr(0))

class DistanceToCamera(object):
    def __init__(self):
        # camera params
        self.alpha = 8.0 * math.pi / 180
        self.v0 = 119.865631204
        self.ay = 332.262498472

    def calculate(self, v, h, x_shift, image):
        # compute and return the distance from the target point to the camera
        d = h / math.tan(self.alpha + math.atan((v - self.v0) / self.ay))
        if d > 0:
            cv2.putText(image, "%.1fcm" % d,
                        (image.shape[1] - x_shift, image.shape[0] - 20), cv2.FONT_HERSHEY_SIMPLEX, 0.7, (255, 255, 255),
                        2)
        return d

class SensorDataHandler(SocketServer.BaseRequestHandler):
    data = " "

    def handle(self):
        global sensor_data
        try:
            while self.data:
                self.data = self.request.recv(1024)
                sensor_data = round(float(self.data), 1)
                # print "{} sent:".format(self.client_address[0])
                print sensor_data
        finally:
            print "Connection closed on thread 2"

class VideoStreamHandler(SocketServer.StreamRequestHandler):
    # h1: stop sign
    h1 = 15.5 - 10  # cm
    # h2: traffic light
    h2 = 15.5 - 10

    # create neural network
    model = NeuralNetwork()
    model.create()

    rc_car = RCControl()

    def handle(self):

        global sensor_data
        stream_bytes = ' '
        stop_flag = False
        stop_sign_active = True

        # stream video frames one by one
        try:
            while True:
                stream_bytes += self.rfile.read(1024)
                first = stream_bytes.find('\xff\xd8')
                last = stream_bytes.find('\xff\xd9')
                if first != -1 and last != -1:
                    jpg = stream_bytes[first:last + 2]
                    stream_bytes = stream_bytes[last + 2:]
                    gray = cv2.imdecode(np.fromstring(jpg, dtype=np.uint8), cv2.CV_LOAD_IMAGE_GRAYSCALE)
                    image = cv2.imdecode(np.fromstring(jpg, dtype=np.uint8), cv2.CV_LOAD_IMAGE_UNCHANGED)

                    # lower half of the image
                    half_gray = gray[120:240, :]

                    cv2.imshow('image', image)
                    # cv2.imshow('mlp_image', half_gray)

                    # reshape image
                    image_array = half_gray.reshape(1, 38400).astype(np.float32)

                    # neural network makes prediction
                    prediction = self.model.predict(image_array)

                    # stop conditions
                    if sensor_data is not None and sensor_data < 30:
                        print("Stop, obstacle in front")
                        self.rc_car.stop()

                    else:
                        self.rc_car.steer(prediction)
                        self.stop_start = cv2.getTickCount()

                    if cv2.waitKey(1) & 0xFF == ord('q'):
                        self.rc_car.stop()
                        break

            cv2.destroyAllWindows()

        finally:
            print "Connection closed on thread 1"

class ThreadServer(object):
    def server_thread(host, port):
        server = SocketServer.TCPServer((host, port), VideoStreamHandler)
        server.serve_forever()

    def server_thread2(host, port):
        server = SocketServer.TCPServer((host, port), SensorDataHandler)
        server.serve_forever()

    distance_thread = threading.Thread(target=server_thread2, args=('192.168.35.63', 8002))
    distance_thread.start()
    video_thread = threading.Thread(target=server_thread('192.168.35.63', 8000))
    video_thread.start()

if __name__ == '__main__':
    ThreadServer()`

It seems that what I deleted is not accurate. please check
Help me! Perhaps this is the last question.

hamuchiwa, 2017-11-07:
Do you intend to remove stop sign and traffic light or traffic light only?

kwon9man, 2017-11-07:
@hamuchiwa
I solved all the problems by modifying some code!(Open cv2->Open cv3)
It was very difficult for me. 
But finally, my RCcar operates like your project! :)
Thank you so much!

M-Salman1337, 2017-11-08:
For the neural network program I think you need to use OpenCV 2 (I also have a question as to which version to use) as I was unable to make the changes to OpenCV 3... program name is mlp_traning.py

M-Salman1337, 2017-11-08:
Reading your comment now @kwon9man  What changes did you do in the mlp_training.py program ? 

We are stuck at this point for a while, please reply asap!

kwon9man, 2017-11-09:
Hi! @M-Salman1337 This is my 'mlp_training.py' code!

And this link will help you -> https://docs.opencv.org/3.0.0/d0/dce/classcv_1_1ml_1_1ANN__MLP.html

`import cv2
import numpy as np
import glob
import sys
from sklearn.model_selection import train_test_split

print 'Loading training data...'
e0 = cv2.getTickCount()

# load training data
image_array = np.zeros((1, 38400))
label_array = np.zeros((1, 4), 'float')
training_data = glob.glob('training_data/*.npz')

# if no data, exit
if not training_data:
    print "No training data in directory, exit"
    sys.exit()

for single_npz in training_data:
    with np.load(single_npz) as data:
        print data.files
        train_temp = data['train']
        train_labels_temp = data['train_labels']
        print train_temp.shape
        print train_labels_temp.shape
    image_array = np.vstack((image_array, train_temp))
    label_array = np.vstack((label_array, train_labels_temp))

train = image_array[1:, :]
train_labels = label_array[1:, :]
print train.shape
print train_labels.shape

e00 = cv2.getTickCount()
time0 = (e00 - e0)/ cv2.getTickFrequency()
print 'Loading image duration:', time0

# train test split, 7:3

# set start time
e1 = cv2.getTickCount()

# create MLP
layer_sizes = np.int32([38400, 32, 4])
model = cv2.ml.ANN_MLP_create()
model.setLayerSizes(layer_sizes)
model.setTrainMethod(cv2.ml.ANN_MLP_BACKPROP)
model.setBackpropMomentumScale(0.0)
model.setBackpropWeightScale(0.001)
model.setTermCriteria((cv2.TERM_CRITERIA_COUNT, 20, 0.01))
model.setActivationFunction(cv2.ml.ANN_MLP_SIGMOID_SYM, 2, 1)

criteria = (cv2.TERM_CRITERIA_COUNT | cv2.TERM_CRITERIA_EPS, 500, 0.0001)
criteria2 = (cv2.TERM_CRITERIA_COUNT, 100, 0.001)

params = dict(term_crit = criteria,
               train_method = cv2.ml.ANN_MLP_BACKPROP,
               bp_dw_scale = 0.001,
               bp_moment_scale = 0.0 )

print 'Training MLP ...'
num_iter = model.train(np.float32(train), cv2.ml.ROW_SAMPLE, np.float32(train_labels))

# set end time
e2 = cv2.getTickCount()
time = (e2 - e1)/cv2.getTickFrequency()
print 'Training duration:', time
#print 'Ran for %d iterations' % num_iter

# save model
model.save('mlp_xml/mlp.xml')

print 'Ran for %d iterations' % num_iter

ret, resp = model.predict(train)
prediction = resp.argmax(-1)
print 'Prediction:', prediction
true_labels = train_labels.argmax(-1)
print 'True labels:', true_labels

print 'Testing...'
train_rate = np.mean(prediction == true_labels)
print 'Train rate: %f' % (train_rate*100)`

M-Salman1337, 2017-11-09:
I've been on the link you gave me, but is your program for OpenCV latest version or is it for 2? @kwon9man help is much appreciated man :)

kwon9man, 2017-11-09:
@M-Salman1337 
I used the OpenCV3.3.0 version of this project.
The 'ANN_MLP' function used in this project does not work in OpenCV3.
The link will help you because you need to use 'cv2.ml.ANN_MLP' to run 'ANN_MLP' in OpenCV3.
I'm glad I helped you! :)

M-Salman1337, 2017-11-09:
yea but even after the ml being added it has lot of other things showing errors 🈂️

kwon9man, 2017-11-09:
@M-Salman1337 
Errors are probably due to the language of OpenCV2.
You must search for the OpenCV3 language corresponding to the error code.

M-Salman1337, 2017-11-09:
the code above you mentioned is mlp_training.py for OpenCV3 ?

yea we did search some places we got stuck..

kwon9man, 2017-11-09:
@M-Salman1337 
Yes, my mlp_training.py code for OpenCV3.
Can you send me error message?

M-Salman1337, 2017-11-09:
I'll send you when I get home because right now I'm outside; by today.

thanks for your concern and help man

kwon9man, 2017-11-09:
@M-Salman1337 
OK, I hope your project works well!
Good Luck :)

M-Salman1337, 2017-11-09:
OMG THIS IS WORKING DUDE@@ @kwon9man

kwon9man, 2017-11-10:
Hi @M-Salman1337

This is my rc_driver.py code.
I didn't use the traffic light and stop sign :(
So I don't know if this code will be of any help to you.
But the ultrasonic sensor worked well.

'import threading
import SocketServer
import serial
import cv2
import numpy as np
import math

# distance data measured by ultrasonic sensor
sensor_data = " "

class NeuralNetwork(object):

    def __init__(self):
        self.model = cv2.ml.ANN_MLP_create()

    def create(self):
        layer_size = np.int32([38400, 32, 4])
        self.model.setLayerSizes(layer_size)
        self.model = cv2.ml.ANN_MLP_load('mlp_xml/mlp.xml')

    def predict(self, samples):
        ret, resp = self.model.predict(samples)
        return resp.argmax(-1)

class RCControl(object):

    def __init__(self):
        self.serial_port = serial.Serial('COM3', 115200, timeout=1)

    def steer(self, prediction):
        if prediction == 2:
            self.serial_port.write(chr(1))
            print("Forward")
        elif prediction == 0:
            self.serial_port.write(chr(4))
            print("Left")
        elif prediction == 1:
            self.serial_port.write(chr(3))
            print("Right")
        else:
            self.stop()

    def stop(self):
        self.serial_port.write(chr(0))

class DistanceToCamera(object):

    def __init__(self):
        # camera params
        self.alpha = 8.0 * math.pi / 180
        self.v0 = 119.865631204
        self.ay = 332.262498472

    def calculate(self, v, h, x_shift, image):
        # compute and return the distance from the target point to the camera
        d = h / math.tan(self.alpha + math.atan((v - self.v0) / self.ay))
        if d > 0:
            cv2.putText(image, "%.1fcm" % d,
                (image.shape[1] - x_shift, image.shape[0] - 20), cv2.FONT_HERSHEY_SIMPLEX, 0.7, (255, 255, 255), 2)
        return d

class ObjectDetection(object):

[... 195 lines cut ...]
                        break

            cv2.destroyAllWindows()

        finally:
            print "Connection closed on thread 1"

class ThreadServer(object):

    def server_thread(host, port):
        server = SocketServer.TCPServer((host, port), VideoStreamHandler)
        server.serve_forever()

    def server_thread2(host, port):
        server = SocketServer.TCPServer((host, port), SensorDataHandler)
        server.serve_forever()

    distance_thread = threading.Thread(target=server_thread2, args=('192.168.123.132', 8002))
    distance_thread.start()
    video_thread = threading.Thread(target=server_thread('192.168.123.132', 8000))
    video_thread.start()

if __name__ == '__main__':
    ThreadServer()
'

M-Salman1337, 2017-11-11:
This is rc_driver.py program ..... I'm getting a few errors..
```
import threading
import socketserver
import serial
import cv2
import numpy as np
import math
import socket

class NeuralNetwork(object):
    def __init__(self):
        self.model = cv2.ml.ANN_MLP_create()
    
    def create(self):
        layer_size = np.int32([38400, 32, 4])
        self.model.setLayerSizes(layer_size)
        self.model = cv2.ml.ANN_MLP_load('mlp_xml/mlp.xml')
    
    def predict(self, samples):
        ret, resp = self.model.predict(samples)
        return resp.argmax(-1)

class RCControl(object):
    def __init__(self):
        self.serial_port = serial.Serial('COM3', 9600, timeout = 1)
    
    def steer(self, prediction):
        if prediction == 2:
            self.serial_port.write("8".encode())
            print("Forward")
        elif prediction == 0:
            self.serial_port.write("4".encode())
            print("Left")
        elif prediction == 1:
            self.serial_port.write("6".encode())
            print("Right")
        else:
            self.stop()
    
    def stop(self):
        self.serial_port.write("0".encode())

class VideoStreamHandler(socketserver.StreamRequestHandler):
 
    # create neural network
    model = NeuralNetwork()
    model.create()
    
    rc_car = RCControl()
    
    def handle(self):
        
        stream_bytes = b' '
       
        
        # stream video frames one by one
        try:
            while True:
                stream_bytes += self.socket.recv(1024)
                first = stream_bytes.find(b'\xff\xd8')
                last = stream_bytes.find(b'\xff\xd9')
                if first != -1 and last != -1:
                    jpg = stream_bytes[first:last + 2]
                    stream_bytes = stream_bytes[last + 2:]
                    gray = cv2.imdecode(np.fromstring(jpg, dtype = np.uint8), cv2.IMREAD_GRAYSCALE)
                    image = cv2.imdecode(np.fromstring(jpg, dtype = np.uint8), cv2.IMREAD_UNCHANGED)
                    
                    # lower half of the image
                    half_gray = gray[120:240, :]
                    
                    cv2.imshow('image', image)
                    # cv2.imshow('mlp_image', half_gray)
                    
                    # reshape image
                    image_array = half_gray.reshape(1, 38400).astype(np.float32)
                    
                    # neural network makes prediction
                    prediction = self.model.predict(image_array)
                    
                    if cv2.waitKey(1) & 0xFF == ord('q'):
                        self.rc_car.stop()
                        break
            
            cv2.destroyAllWindows()
        
        finally:
            print("Connection closed on thread 1")

class ThreadServer(object):
    def server_thread(host, port):
        server = socketserver.TCPServer((host, port), VideoStreamHandler)
        server.serve_forever()
    
  
    video_thread = threading.Thread(target = server_thread('192.168.0.6', 8081))
    video_thread.start()

if __name__ == '__main__':
    ThreadServer()
```

Error message:
```
Traceback (most recent call last):
  File "C:/Users/.....", line 91, in <module>
    class ThreadServer(object):
  File "C:/Users/...", line 97, in ThreadServer
    video_thread = threading.Thread(target = server_thread('192.168.0.6', 8081))
  File ".....", line 93, in server_thread
    server = socketserver.TCPServer((host, port), VideoStreamHandler)
  File "C:\Users\....", line 440, in __init__
    self.server_bind()
  File "C:\Users\...", line 454, in server_bind
    self.socket.bind(self.server_address)
OSError: [WinError 10049] The requested address is not valid in its context
```

M-Salman1337, 2017-11-11:
I am **_## not_** using the red light, green light stop sign and obstacle detection.... Just plain lane following

kwon9man, 2017-11-11:
@M-Salman1337 
This code 'just include' traffic signal , stop sign and obstacle detection.
It doesn't matter if you don't use them.
If you want only lane following, you have to do neural network training by collect_training_data.py and mlp_training.py
If you have done your training well, your RCCar will follow the lane.
Despite not using traffic signal, stop signal, obstacle detection!

M-Salman1337, 2017-11-11:
If you want only lane following, you have to do neural network training by collect_training_data.py and mlp_training.py

But what program will execute the car to operate ?

kwon9man, 2017-11-11:
@M-Salman1337 
First, run collect_training_data.py on your computer and run stream_client.py on RaspberryPi.
Then pygame's GUI pops up and you can manipulate RCCar by typing the directional key on it.
If RCCar doesn't work, you should check the following:
1. Is the Arduino code of RCCar correct?
2. Did you correctly fill in the IP addresses of collect_training_data.py and stream_client.py?
3. Is pygame GUI code syntax added to collect_training_data.py?

check your code

M-Salman1337, 2017-11-11:
It works, collect_training_data.py opens 2 windows 1. pygame GUI 2.Pi camera (it saves the frames and makes NPZ file)
then, mlp_training.py creates the mlp.xml file...

So, now those 2 programs are done, now how do I let the car go in a track?

kwon9man, 2017-11-11:
@M-Salman1337 
You need to repeat neural network learning several times with collect_training_data.py.
For example, I repeated about 15 times on the same track (15 npz files).
RCCar learns itself by learning neural network several times.
Once the correct learning is done, the RCCar will look at the road on the camera and predict it on its own.

M-Salman1337, 2017-11-11:
Once the correct learning is done, the RCCar will look at the road on the camera and predict it on its own.

but how? because those 2 programs just create files

kwon9man, 2017-11-11:
@M-Salman1337 
Once you have completed the training and mapping, run rc_driver.py on your PC and stream_client.py from RaspberryPi to run it automatically.

M-Salman1337, 2017-11-11:
Yea but as I said, I'm getting those errors :(

kwon9man, 2017-11-11:
@M-Salman1337 
Oh, sorry I checked your rc_driver.py error now.
The error indicates the problem of serial communication.
Check your IP and port number again.
stream_client.py should be the same.

M-Salman1337, 2017-11-11:
IP and port is correct, but do you think it is missing anything? I removed lot of code that had anything related to ultrasonic sensor/stop sign/ signal lights...

kwon9man, 2017-11-11:
@M-Salman1337 
There may be some code that you have deleted that is important to your operation.
Run it with the code I sent.
My code includes traffic signals, stop sign, and obstruction detection.
However, the code can be executed without using them.

M-Salman1337, 2017-11-11:
Ok I'll try it out :D

AiBuddy, 2017-11-12:
@kwon9man  can you give me your training model please. I need it in order to use it with my car because i am running out of time and my project submission is very close

AiBuddy, 2017-11-12:
if you could give me training data or mlp data it would be very helpful from you.

KhyatiMehta3, 2018-02-24:
Hi, your project is very impressive! I'm trying to implement this and make it follow google maps too. Could you probably give me some insight? Also, this may sound like a silly question, but after the model is trained, can I just deploy it in the raspberry pi such that I don't have to use the computer anymore?

hamuchiwa, 2018-02-24:
@KhyatiMehta3 Hi, if your goal is to navigate the car only using gps, I could imagine the error would be very high. Certainly, you can train the model on computer and deploy it to the raspberry.

closed 2018-03-19 by hamuchiwa