hamuchiwa/AutoRCCar #56: Problem in assigning proper IP address to "server_socket"

smahajan07, 2017-11-22:
Hi , I am streaming the mjpg video stream from the pi in a different manner (using ffmpeg) and the IP address it streams on is " {address}:{port}/test.mjpg" but as given [here](https://github.com/hamuchiwa/AutoRCCar/blob/master/computer/collect_training_data.py#L18) self.server_socket.bind(('192.168.1.100', 8000)) , I am unable to make it listen to the address modified to address. Server_socket only seems to take two arguments , hence I am unable to add the "/test.mjpg" at the end.
Any suggestions? @hamuchiwa

hamuchiwa, 2017-11-22:
Hi, I'm not familiar with ffmpeg streaming, I think in your case raspberry pi becomes the server and you can access the stream through a browser. So you need to somehow make pc as a client to accept frames from raspberry pi. Could you post some code snippets on how you did it?

smahajan07, 2017-11-23:
@hamuchiwa , I figured out a possible solution. Basically ffmpeg is streaming over an HTTP server , but the socket works over a TCP/IP server. So I can either change the streaming method or use an HTTP based library to make my pc a client,

I will post a snippet as soon as I can get it working, so that it anyone wants to stream it using ffmpeg can use the script/library.

smahajan07, 2017-11-23:
Python has a library 'urllib' which can be used for the same purpose. Here is the sample code to just view the stream(from Pi) using opencv. 

'''
  
    import numpy as np
    import cv2
    import urllib
    
    url = 'http://192.168.1.186:8080/test.mjpg'
    stream = urllib.urlopen(url)
    bytes = ''
    
    while True:
        bytes += stream.read(1024)
        a = bytes.find('\xff\xd8')
        b = bytes.find('\xff\xd9')
        # print('getting a and b')
        if a != -1 and b != -1:
            # print('Got it')
            jpg = bytes[a:b+2]
            bytes = bytes[b+2:]
            i = cv2.imdecode(np.fromstring(jpg, dtype=np.uint8), cv2.CV_LOAD_IMAGE_COLOR)
            cv2.imshow('Cam from Pi', i)
    
            if cv2.waitKey(1) == 27:
                exit(0)
    
    cv2.destroyAllWindows()
'''
Here is an excellent explanation of the code [stackoverflow](https://stackoverflow.com/questions/21702477/how-to-parse-mjpeg-http-stream-from-ip-camera)

hamuchiwa, 2017-11-23:
Very nice, thanks :)

closed 2018-03-19 by hamuchiwa