hamuchiwa/AutoRCCar #57: Training model  and accuracy.

IsmoilovMuhriddin, 2017-11-23:
So far I have managed to stream video to my computer and taking training data by saving them in .npz file. (Still I have more than 4000 pics and labels- training data)
train.shape - (1762, 38400)
train_label.shape - (1762, 4)

But now I have a problem with a model which returns as output always the same number: all 0s,1s or 2s.
Here I used two models- from opencv3: cv2.ml.ANN_MLP_create() and from Tensorlow.

**Question** 

- How to reach the best accuracy?

- Can you test my code on your training data, so if its accuracy is high enough my training data will be trash and I should collect absolutely different training data?

mlp_training.py
```
import cv2
import numpy as np
import glob
import sys
from sklearn.model_selection import train_test_split
import tensorflow as tf
import keras.backend
from keras.layers import Flatten, Dense, Dropout, Lambda, Convolution2D, Cropping2D, Conv2D, Conv1D, Reshape
from keras.callbacks import EarlyStopping
from keras.models import Sequential
import matplotlib.pyplot as plt

print 'Loading training data...'
e0 = cv2.getTickCount()

# load training data
image_array = np.zeros((1, 38400))
label_array = np.zeros((1, 4), 'float')
training_data = glob.glob('training_data/*.npz')

# if no data, exit
if not training_data:
    print "No training data in directory, exit"
    sys.exit()

for single_npz in training_data:
    with np.load(single_npz) as data:
        print data.files
        train_temp = data['train']
        train_labels_temp = data['train_labels']
        print train_temp.shape
        print train_labels_temp.shape
    image_array = np.vstack((image_array, train_temp))
    label_array = np.vstack((label_array, train_labels_temp))

train = image_array[1:, :]
train_labels = label_array[1:, :]

# train test split, 7:3
train_X, test_X, train_y, test_y = train_test_split(train, train_labels, test_size=0.3)

print train.shape
print train_labels.shape

show_images = False
export_images = False
should_train = True
use_tensorflow = True

def show_image(sample, delay=200):
    sample = sample.reshape((120, 320)).astype(np.uint8)
    cv2.imshow('image', sample)
    cv2.waitKey(delay=delay)

if show_images:
[... 158 lines cut ...]
 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
 0 0 0 0 0 0 0 0 0 0 0]
True labels: [2 1 2 0 2 2 0 1 2 2 2 2 2 2 1 2 0 2 2 2 0 0 2 2 1 0 0 0 0 1 2 2 2 2 0 2 1
 1 2 2 2 2 1 2 0 1 2 0 2 1 0 2 0 2 1 2 1 2 2 2 2 2 0 2 0 0 2 2 2 2 2 2 2 2
 2 2 1 2 2 2 2 2 1 1 2 2 2 1 2 1 2 0 2 2 2 1 2 2 1 2 0 0 2 2 2 2 2 2 2 0 2
 2 0 2 0 2 0 1 2 0 2 2 2 2 2 1 1 2 1 0 2 2 0 1 2 2 2 2 2 2 2 1 2 1 1 2 2 2
 2 2 1 1 2 2 2 2 1 1 0 2 0 2 2 2 0 2 2 2 2 1 2 0 2 0 2 2 1 2 1 1 2 0 2 2 0
 2 2 2 0 0 2 2 2 1 0 0 2 2 0 2 2 2 0 2 0 2 1 2 2 2 2 2 0 2 2 1 2 2 2 0 2 2
 1 2 2 0 2 2 0 0 2 2 2 2 2 2 2 0 0 2 2 0 2 2 2 2 1 0 2 2 2 2 0 2 1 2 0 2 2
 0 2 2 2 2 2 0 2 2 0 2 2 0 2 2 2 1 1 2 2 2 2 0 1 2 1 2 0 2 2 2 0 2 2 2 0 2
 1 2 1 2 0 0 1 2 0 2 0 0 1 2 2 2 1 2 2 2 2 2 2 1 2 3 2 2 0 1 2 1 1 2 2 2 2
 2 2 2 2 2 2 2 2 2 1 1 1 1 2 0 0 2 0 2 1 0 2 2 2 2 2 2 2 0 2 2 1 2 2 2 2 2
 2 2 2 2 0 2 2 1 1 0 2 0 2 0 1 2 2 2 2 0 2 2 2 2 2 2 2 2 1 2 2 2 2 1 2 2 0
 2 2 0 2 0 0 1 1 0 2 2 2 2 2 0 2 2 2 2 0 2 0 2 1 0 2 2 2 0 1 1 2 0 0 2 0 2
 2 0 1 0 0 2 2 2 1 0 2 1 2 2 2 2 0 2 2 2 2 2 0 2 2 2 2 2 2 0 2 2 0 2 2 2 1
 2 2 1 2 2 0 2 2 2 2 0 2 2 0 1 2 0 2 2 2 2 0 2 2 2 1 2 0 2 1 2 2 0 2 2 2 2
 0 2 2 2 2 1 2 1 1 2 1]
Testing...
Train rate: 19.848771:

```
Pls help!!!

AiBuddy, 2017-11-23:
@hamuchiwa  having slightly same problem . please help

hamuchiwa, 2017-11-23:
Training neural network is non trivial, try to use whole image which may give you a better result. To give you peace of mind, here is a [self-driving video](https://www.youtube.com/watch?v=KrvvmLEQk8o) from aha124226675.

smahajan07, 2017-11-26:
@hamuchiwa . I seem to have a similar problem, except that I have successfully trained the model (see below). 
But when I'm running rc_driver.py, it is predicting only one output , that is, 'left'. I am still debugging, but open to suggestions.

    Loading training data...
    ['train_labels', 'train']
    (219, 38400)
    (219, 4)
    ['train_labels', 'train']
    (74, 38400)
    (74, 4)
    ['train_labels', 'train']
    (16, 38400)
    (16, 4)
    ['train_labels', 'train']
    (34, 38400)
    (34, 4)
    ['train_labels', 'train']
    (75, 38400)
    (75, 4)
    ['train_labels', 'train']
    (57, 38400)
    (57, 4)
    (475, 38400)
    (475, 4)
    Loading image duration: 1.18788429317
    Training MLP ...
    Training duration: 57.7994991789
    Ran for 30 iterations
    Prediction: [2 2 2 3 2 0 2 0 3 2 2 2 0 0 0 0 0 2 2 0 0 0 2 0 2 2 2 2 0 0 0 0 0 2 1 2 0
     0 2 0 0 0 2 0 2 3 2 2 2 0 0 2 0 0 0 2 2 2 2 1 1 2 2 0 0 0 2 0 0 0 0 0 0 1
     2 0 2 3 2 1 1 2 2 0 0 0 0 0 0 0 0 2 2 3 2 0 0 2 2 1 0 2 0 0 0 0 0 0 2 2 3
     3 2 2 2 2 0 0 0 0 1 2 0 0 0 2 2 2 2 2 1 1 0 3 0 0 0 0 2 1 1 1 3 3 1 1 2 3
     2 0 0 0 0 2 2 1 1 2 2 2 1 1 2 1 1 1 2 2 2 2 2 2 2 1 2 1 1 1 1 1 1 2 1 1 2
     1 1 1 1 2 1 1 3 3 0 2 2 2 2 2 1 2 2 0 2 2 2 1 1 1 1 1 1 2 2 2 2 2 2 2 2 2
     2 2 1 1 2 2 2 2 1 1 2 2 2 1 1 1 1 2 2 2 3 2 2 2 1 2 2 2 2 2 2 1 1 1 1 1 1
     2 2 2 3 1 2 2 2 1 1 1 1 1 2 1 1 1 1 2 2 2 0 0 0 2 1 1 1 1 0 0 2 2 2 2 2 2
     3 2 2 2 0 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 0 3 2 2 2 2 2 3 2 2 2 2 2 2 2 3 2
     2 2 2 2 2 2 3 3 3 2 2 2 2 2 2 1 1 0 0 0 0 0 0 0 0 0 0 2 1 1 0 3 3 3 3 3 3
     2 2 0 0 0 0 0 2 0 0 2 3 1 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 3 1 1 0
     0 0 0 2 2 2 0 0 0 0 0 2 2 2 2 2 2 2 2 0 0 0 0 0 0 0 3 2 2 2 0 0 0 0 0 0 0
     2 0 0 0 0 0 3 3 3 3 3 3 3 3 1 3 2 0 1 0 0 0 0 0 0 0 0 0 0 0 0]
    True labels: [2 2 2 2 2 0 2 0 2 2 2 2 0 0 0 0 0 2 2 0 2 0 2 2 2 2 2 2 0 0 0 0 0 2 1 2 0
     0 2 0 0 0 0 0 2 2 2 2 2 0 0 2 0 0 2 2 2 2 2 1 1 2 0 0 0 0 2 0 0 0 0 0 0 1
     2 2 2 2 2 1 1 2 2 0 0 0 0 0 1 0 0 2 2 2 2 0 0 2 2 1 0 2 0 0 0 0 0 0 2 2 2
     2 2 2 2 2 0 0 0 0 1 2 0 0 2 2 2 2 2 2 1 1 0 0 0 0 0 0 2 1 1 1 1 1 1 1 2 2
     0 0 0 0 0 2 2 1 1 2 2 2 1 1 2 1 1 1 2 2 2 2 2 2 2 1 2 1 1 1 1 1 1 2 1 1 2
     1 1 1 1 2 1 1 0 0 2 2 2 2 2 2 1 2 2 0 2 2 2 1 1 1 1 1 1 2 0 2 2 2 2 2 2 2
     2 2 1 1 2 2 2 2 1 1 2 2 2 1 1 1 1 2 2 2 2 2 2 2 1 2 2 2 2 2 2 1 1 2 1 1 0
     2 2 2 1 1 2 2 2 1 1 1 1 1 2 1 1 1 1 2 2 2 0 0 2 2 1 1 1 1 1 1 2 2 2 2 2 2
     2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2
     2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 1 1 0 0 2 0 0 0 0 0 0 2 2 1 1 0 0 0 0 0 0 0
     2 2 0 0 0 0 1 2 1 0 2 2 1 1 0 2 0 0 0 0 0 0 1 0 2 2 0 0 0 0 2 0 0 2 1 1 0
     0 0 0 2 2 0 0 0 0 0 0 2 2 2 2 2 2 0 2 0 0 0 0 0 1 1 2 2 2 2 0 0 0 0 0 0 1
     2 0 0 0 0 1 2 0 1 2 1 2 1 1 1 2 2 0 1 0 0 0 0 0 0 0 0 0 0 0 0]
    Testing...
    Train rate: 84.631579:

smahajan07, 2017-11-26:
Just wanted to mention that I am not exactly using rc_driver.py but an equivalent script just to check predictions of my network and not worry about the ultrasonic sensor or stop signs. I have uploaded it as a gist [here](https://gist.github.com/smahajan07/8e0eb251e706e0e3dc6c32bba0b739db)

hamuchiwa, 2017-11-27:
@smahajan07 You could do cross validation to see how good the model is performed. Of course, if your keep using the same track then a high accuracy on training set is sufficient, overfitting is irrelevant in this case.

smahajan07, 2017-11-27:
@hamuchiwa I will try to cross validate my training set and get back, in the mean time could you tell me if the procedure I am following is faulty.

Before training it on dozens of different tracks, I am training it on 2 tracks (straight and then right/left respectively) back and forth. The data being collected is approximately 500 images in total. Using the now generated npz files I'm running the 'mlp_training.py' and getting an xml file.
What concerns me is the training is happening too fast (i didn't expect this on my i5 and integrated graphics card, I have a server but not currently using it for this) and that every time my xml file is approx 35 MB (even though I change my number of training images).

I know these are very trivial concerns,but let me know if you have any suggestions.

Also, what is the minimum number of images that we should train the neural network on?

smahajan07, 2017-11-27:
@hamuchiwa I am planning to create a branch for this repo where in those who are making this project using an external webcam (and not raspi cam) and use a different streaming method (and not socket server).

hamuchiwa, 2017-11-28:
@smahajan07 Hi, I think 500 images in total is good enough. If you are not entirely satisfied with your training data, you could "cheat" a bit meaning that you can manually put the car on the track where you think it should be, then create a sample, and then do ti over and over again. Sometimes I found this helpful as I do make mistakes while controlling the rc car, but it is really tedious doing this...

The xml file contains weights and biases for your trained model, the dimension of the weights matrix and bias matrix depends on the architecture of the neural network (layer size) rather than number of samples. If you have a deep neural network then you probably would see the file size increase.

smahajan07, 2017-11-28:
@hamuchiwa , I cross validated my data set and it looks promising, but I am unfortunately facing the same problem that my [script](https://gist.github.com/smahajan07/8e0eb251e706e0e3dc6c32bba0b739db) (an equivalent to your rc_driver.py, instead only focuses on driving, no signs or distance calculation) is causing some problem and the only output I am receiving is 'LEFT'.

And this cross validation was performed by inputting the complete image and not just half. Also I changed the layer size to [(76800, 64,4)].

        Loading training data...
        ['train_labels', 'train']
        (114, 76800)
        (114, 4)
        ['train_labels', 'train']
        (100, 76800)
        (100, 4)
        ['train_labels', 'train']
        (122, 76800)
        (122, 4)
        ['train_labels', 'train']
        (75, 76800)
        (75, 4)
        (411, 76800)
        (411, 4)
        Loading image duration: 1.81404559868
        Training MLP ...
        Training duration: 509.356298789
        Ran for 76 iterations
        Prediction: [2 2 2 2 0 0 0 0 0 0 0 0 0 0 0 0 2 0 0 0 0 0 0 0 0 2 1 1 0 0 2 2 0 0 0 0 0
         0 0 0 0 0 0 0 0 0 0 0 0 0 2 2 2 2 2 1 1 1 1 1 1 2 0 0 1 1 1 1 1 1 2 2 2 2
         2 1 1 1 0 0 2 2 2 2 0 0 2 2 0 0 0 2 2 2 2 0 0 0 0 3 3 2 0 0 0 0 0 0 0 0 0
         0 0 0 2 2 2 1 1 1 1 1 1 2 2 0 0 0 0 0 1 1 2 2 2 2 2 0 0 1 1 0 0 0 0 0 0 0
         1 1 1 1 0 0 0 0 0 0 2 0 0 0 0 0 0 0 0 0 0 0 2 0 0 1 1 1 1 1 1 1 1 1 1 1 1
         1 1 0 0 2 0 1 1 1 2 2 0 0 0 0 0 2 0 0 0 0 0 0 0 0 0 0 1 1 1 1 1 1 2 0 0 2
         2 1 1 0 1 2 2 2 1 2 2 0 0 0 0 1 1 0 0 0 0 0 0 0 0 0 0 0 0 1 1 0 0 1 1 1 1
         1 0 1 1 1 0 0 2 2 1 1 1 1 1 1 1 1 1 0 0 0 2 2 2 2 2 0 0 2 2 2 2 0 0 0 0 3
         2 2 1 1 1 1 1 1 2 2 2 2 2 2 0 0 1 1 0 0 0 2 2 2 0 0 0 0 1 0 0 1 1 1 1 1 1
         1 1 2 2 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 2 2 2 2 2 2 2 2 0 0 0 0 0 0 0 0
         0 0 0 0 0 0 0 0 0 0 1 1 1 2 2 2 2 2 2 2 2 1 1 1 1 1 1 1 1 1 1 1 1 2 2 2 2
         2 2 2 2]
        True labels: [2 2 2 2 0 0 0 0 0 0 0 0 0 0 0 0 2 0 0 0 0 0 0 0 2 2 1 1 0 0 2 2 0 0 0 0 0
         0 0 0 0 0 0 1 1 0 0 0 0 0 2 2 2 2 2 1 1 1 1 1 1 2 0 0 1 1 1 1 1 1 0 0 2 2
         2 1 1 0 0 0 2 2 2 2 0 0 2 2 0 0 3 2 2 2 2 0 0 0 0 1 1 2 0 0 0 0 0 0 0 0 0
         0 0 0 2 2 2 1 1 1 1 1 1 2 2 0 0 0 0 1 1 1 2 2 2 2 2 0 0 2 0 0 0 0 0 0 0 0
         1 1 1 1 0 0 0 1 1 0 2 0 0 0 0 0 0 0 0 0 0 1 2 0 0 1 1 1 1 1 1 1 1 1 1 1 1
         1 1 0 0 2 0 1 1 1 2 2 0 0 0 0 2 2 0 0 0 0 0 0 0 0 0 0 1 1 1 1 1 1 2 0 0 2
         2 1 1 0 1 2 2 2 2 2 2 0 0 0 0 1 1 0 2 0 0 0 0 0 0 0 0 0 0 1 1 0 0 1 1 1 1
         1 0 1 1 0 0 0 1 1 1 1 0 0 1 1 1 1 0 0 0 0 2 2 2 2 2 0 0 2 2 2 2 0 0 0 0 3
         2 2 1 1 1 1 1 1 2 2 0 2 2 2 0 0 1 1 0 0 1 2 0 2 0 0 0 0 1 0 0 1 1 1 1 1 1
         1 1 2 2 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 2 2 2 2 2 2 2 2 0 0 0 0 0 0 0 0
         0 0 0 0 0 0 0 0 2 2 1 1 2 2 2 2 2 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 2 2 2 2
         2 2 2 2]
        Train rate: 91.727494:
        ['train_labels', 'train']
        (75, 76800)
        (75, 4)
        (411, 76800)
        (411, 4)
        Testing Prediction: [2 2 2 2 0 0 0 0 0 0 0 0 0 0 0 0 2 0 0 0 0 0 0 0 0 2 1 1 0 0 2 2 0 0 0 0 0
         0 0 0 0 0 0 0 0 0 0 0 0 0 2 2 2 2 2 1 1 1 1 1 1 2 0 0 1 1 1 1 1 1 2 2 2 2
         2 1 1 1 0 0 2 2 2 2 0 0 2 2 0 0 0 2 2 2 2 0 0 0 0 3 3 2 0 0 0 0 0 0 0 0 0
         0 0 0 2 2 2 1 1 1 1 1 1 2 2 0 0 0 0 0 1 1 2 2 2 2 2 0 0 1 1 0 0 0 0 0 0 0
         1 1 1 1 0 0 0 0 0 0 2 0 0 0 0 0 0 0 0 0 0 0 2 0 0 1 1 1 1 1 1 1 1 1 1 1 1
         1 1 0 0 2 0 1 1 1 2 2 0 0 0 0 0 2 0 0 0 0 0 0 0 0 0 0 1 1 1 1 1 1 2 0 0 2
         2 1 1 0 1 2 2 2 1 2 2 0 0 0 0 1 1 0 0 0 0 0 0 0 0 0 0 0 0 1 1 0 0 1 1 1 1
         1 0 1 1 1 0 0 2 2 1 1 1 1 1 1 1 1 1 0 0 0 2 2 2 2 2 0 0 2 2 2 2 0 0 0 0 3
         2 2 1 1 1 1 1 1 2 2 2 2 2 2 0 0 1 1 0 0 0 2 2 2 0 0 0 0 1 0 0 1 1 1 1 1 1
         1 1 2 2 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 2 2 2 2 2 2 2 2 0 0 0 0 0 0 0 0
         0 0 0 0 0 0 0 0 0 0 1 1 1 2 2 2 2 2 2 2 2 1 1 1 1 1 1 1 1 1 1 1 1 2 2 2 2
         2 2 2 2 3 3 3 2 3 3 3 3 0 0 3 3 3 3 3 1 1 0 0 2 2 3 0 0 0 0 0 0 0 0 0 0 1
         1 1 2 2 3 3 3 3 2 2 2 2 2 1 1 1 1 1 1 1 2 2 3 3 0 0 0 2 0 2 0 0 1 1 1 1 1
         1 2 2 2 2]
        Testing True labels: [2 2 2 2 0 0 0 0 0 0 0 0 0 0 0 0 2 0 0 0 0 0 0 0 2 2 1 1 0 0 2 2 0 0 0 0 0
         0 0 0 0 0 0 1 1 0 0 0 0 0 2 2 2 2 2 1 1 1 1 1 1 2 0 0 1 1 1 1 1 1 0 0 2 2
         2 1 1 0 0 0 2 2 2 2 0 0 2 2 0 0 3 2 2 2 2 0 0 0 0 1 1 2 0 0 0 0 0 0 0 0 0
         0 0 0 2 2 2 1 1 1 1 1 1 2 2 0 0 0 0 1 1 1 2 2 2 2 2 0 0 2 0 0 0 0 0 0 0 0
         1 1 1 1 0 0 0 1 1 0 2 0 0 0 0 0 0 0 0 0 0 1 2 0 0 1 1 1 1 1 1 1 1 1 1 1 1
         1 1 0 0 2 0 1 1 1 2 2 0 0 0 0 2 2 0 0 0 0 0 0 0 0 0 0 1 1 1 1 1 1 2 0 0 2
         2 1 1 0 1 2 2 2 2 2 2 0 0 0 0 1 1 0 2 0 0 0 0 0 0 0 0 0 0 1 1 0 0 1 1 1 1
         1 0 1 1 0 0 0 1 1 1 1 0 0 1 1 1 1 0 0 0 0 2 2 2 2 2 0 0 2 2 2 2 0 0 0 0 3
         2 2 1 1 1 1 1 1 2 2 0 2 2 2 0 0 1 1 0 0 1 2 0 2 0 0 0 0 1 0 0 1 1 1 1 1 1
         1 1 2 2 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 2 2 2 2 2 2 2 2 0 0 0 0 0 0 0 0
         0 0 0 0 0 0 0 0 2 2 1 1 2 2 2 2 2 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 2 2 2 2
         2 2 2 2 2 2 2 2 0 0 2 2 2 2 1 1 1 1 1 1 1 1 1 0 2 2 2 2 1 1 1 1 1 1 1 1 1
         1 2 0 2 2 1 1 2 0 0 1 1 0 0 0 1 2 2 2 2 2 2 2 2 0 0 0 2 2 3 0 0 0 0 0 0 1
         1 1 1 0 2]
        Test rate: 81.481481:
        Process finished with exit code 0

Could you look at my [script](https://gist.github.com/smahajan07/8e0eb251e706e0e3dc6c32bba0b739db) and see if something is wrong?

smahajan07, 2017-11-29:
@hamuchiwa 
I'm so sorry, there was a very minor bug in my code. It was not looking in the right folder. :/
I apologize for the confusion, and thanks for your help.

Everything is fine, except my the response is a little delayed. Do you have any suggestions to speed up the response?

hamuchiwa, 2017-11-29:
@smahajan07 You could lower the image resolution, or maybe create a buffer and use two threads, one for just getting image from camera, another for streaming. Another "quick" way is to make your car go slower...

closed 2018-03-19 by hamuchiwa

denisrexhepi, 2020-05-17:
> @hamuchiwa I am planning to create a branch for this repo where in those who are making this project using an external webcam (and not raspi cam) and use a different streaming method (and not socket server).

Did you manage to do it?