hamuchiwa/AutoRCCar #58: What is the basic idea/algorithm and approach here?

M-Salman1337, 2017-12-01:
Can anyone explain the neural network model?

As far as I know, the 38,400 (pixels) go into the input node but then there are 32 nodes (what are these? what do they carry?) and then 4 outputs (reverse forward left and right) Someone please help!

hamuchiwa, 2017-12-01:
@M-Salman1337 I would suggest you take a look the following resources:
1. A book, [Neural Network and Deep Learning](http://neuralnetworksanddeeplearning.com/chap1.html)
2. [Deep Learning on Coursera](https://www.coursera.org/specializations/deep-learning)

M-Salman1337, 2017-12-01:
Ya I’ll take a look at them, but can you breifly explain the working of how the camera takes pixels and converts to outputs ..

hamuchiwa, 2017-12-03:
@M-Salman1337 Think neural network as a black box that has lots of knobs(weights and biases) on it. You give it an image and it tells you what that image is (output). In order to make sure it works properly, those default knobs need to be adjusted. So in training, the neural network take lots of labeled images as input and adjusts the knobs until the outputs are mostly correct(output == label). Once it's trained, you use the same knob setting for prediction, and it will tell you what that image is.

M-Salman1337, 2017-12-03:
Oh I see, thats interesting analogy! 
I’ll ask a few questions..
1.output will be an operation and not an image? (Forward, reverse right left) I dont get it how image becomes an output again. 
2.Based on what are the weights and biased adjusted?
3. How does activation function (sigmoid here) help in the model because it gives either -1 and 1 as an S in graph format?

Thanks for the help @hamuchiwa :)

hamuchiwa, 2017-12-04:
@M-Salman1337 
1. Output is a number, for example, given a image to tell if it's a car or not, the output is 1 (cat) or 0(not a cat). For multiple-classes problem, the output is a vector.
2. Minimizing the cost function, you want the cost as small as possible so the prediction is close to the actual value, for example, the function is L = (output - label)^2, the goal is to minimize that.
3. Sigmoid gives you the y range from 0 to 1. Activation functions make your model non-linear that produces non-linear decision boundary.

M-Salman1337, 2017-12-04:
Oh, and the sigmoid function that we're using is Symmetrical sigmoid so it varies from -1 to 1... (SIGMOID_SYS) like how does that account for the pixel values? is it like white pixel = greater than 0 and black pixel = less than 0? (depending on intensity)

hamuchiwa, 2017-12-04:
@M-Salman1337 I see, in OpenCV the default activation is [SIGMOID_SYS](https://docs.opencv.org/2.4/modules/ml/doc/neural_networks.html#neural-networks). The pixel value range is still from 0 to 255, it's the output that changes to [-1, 1]. OpenCV [normalizes](https://docs.opencv.org/2.4/modules/ml/doc/neural_networks.html#neural-networks) y labels by default in training to match its activation function. 

I'm not sure how OpenCV implements multi-class classification with Sigmoid_sys, but in general what people do is using ReLU or tanh(sigmoid_sys) in hidden layers, and using softmax (mutti-class or binary, sigmoid can also be used for binary) for the output layer. Unfortunately, OpenCV doesn't provide you this kind of flexibility.

closed 2018-03-19 by hamuchiwa