hamuchiwa/AutoRCCar #59: mlp_training error

OmairSalahat, 2017-12-13:
Hello!!
I got this error after running mlp_training.py:

Traceback (most recent call last):
  File "C:/Python27/mlp_training.py", line 7, in <module>
    from sklearn.model_selection import train_test_split
  File "C:\Python27\lib\site-packages\sklearn\__init__.py", line 134, in <module>
    from .base import clone
  File "C:\Python27\lib\site-packages\sklearn\base.py", line 13, in <module>
    from .utils.fixes import signature
  File "C:\Python27\lib\site-packages\sklearn\utils\__init__.py", line 11, in <module>
    from .validation import (as_float_array,
  File "C:\Python27\lib\site-packages\sklearn\utils\validation.py", line 18, in <module>
    from ..utils.fixes import signature
  File "C:\Python27\lib\site-packages\sklearn\utils\fixes.py", line 144, in <module>
    from scipy.sparse.linalg import lsqr as sparse_lsqr  # noqa
  File "C:/Python27\scipy\sparse\linalg\__init__.py", line 114, in <module>
    from .isolve import *
  File "C:/Python27\scipy\sparse\linalg\isolve\__init__.py", line 6, in <module>
    from .iterative import *
  File "C:/Python27\scipy\sparse\linalg\isolve\iterative.py", line 7, in <module>
    from . import _iterative
ImportError: DLL load failed: The specified module could not be found.
>>> 
I tried to reinstall scipy but it doesn't work.
Hopefully you can help me solving this problem..
Thank you!

hamuchiwa, 2017-12-13:
Check this on [stackoverflow](https://stackoverflow.com/questions/20201868/importerror-dll-load-failed-the-specified-module-could-not-be-found).

OmairSalahat, 2017-12-23:
Hello, thanks I solved it by installing scipy and numpy again, but I got this error while training..

Traceback (most recent call last):
  File "C:\Python27\mlp_training.py", line 81, in <module>
    model.save('mlp_xml/mlp.xml')
error: ..\..\..\..\opencv\modules\ml\src\inner_functions.cpp:71: error: (-2) Could not open the file storage. Check the path and permissions in function CvStatModel::save

hamuchiwa, 2017-12-26:
It seems it's the path, is there a folder 'mlp_xml' in your directory?

OmairSalahat, 2018-01-01:
Thank you so much Hamuchiwa it's working perfectly!

closed 2018-01-02 by hamuchiwa