hamuchiwa/AutoRCCar #60: mlp_training.py issue with model.create

Kaulinan, 2017-12-21:
Hello hamuchiwa, 

I have an error occuring when i try to do the mlp_training.py.

i made some changes to the code since some things changed with new versions of OpenCV.

Some of the changes are : 

`model = cv2.ml_ANN_MLP(training_data)`

Also i changed the value in image_array in order to match the resolution i set on the picamera.

Below you can see the full script with all changes i made.

`__author__ = 'zhengwang'

import cv2
import numpy as np
import glob
import sys
from sklearn.cross_validation import train_test_split

print 'Loading training data...'
e0 = cv2.getTickCount()

# load training data
image_array = np.zeros((1, 28800))
label_array = np.zeros((1, 4), 'float')
training_data = glob.glob('training_data/*.npz')

# if no data, exit
if not training_data:
    print "No training data in directory, exit"
    sys.exit()

for single_npz in training_data:
    with np.load(single_npz) as data:
        train_temp = data['train']
        train_labels_temp = data['train_labels']
    image_array = np.vstack((image_array, train_temp))
    label_array = np.vstack((label_array, train_labels_temp))

X = image_array[1:, :]
y = label_array[1:, :]
print 'Image array shape: ', X.shape
print 'Label array shape: ', y.shape

e00 = cv2.getTickCount()
time0 = (e00 - e0)/ cv2.getTickFrequency()
print 'Loading image duration:', time0

# train test split, 7:3
train, test, train_labels, test_labels = train_test_split(X, y, test_size=0.3)

# set start time
e1 = cv2.getTickCount()

# create MLP
layer_sizes = np.int32([28800, 32, 4])
model = cv2.ml_ANN_MLP(training_data)
model.create(layer_sizes)
criteria = (cv2.TERM_CRITERIA_COUNT | cv2.TERM_CRITERIA_EPS, 500, 0.0001)
#criteria2 = (cv2.TERM_CRITERIA_COUNT, 100, 0.001)
params = dict(term_crit = criteria,
               train_method = cv2.ANN_MLP_TRAIN_PARAMS_BACKPROP,
               bp_dw_scale = 0.001,
               bp_moment_scale = 0.0 )

print 'Training MLP ...'
num_iter = model.train(np.float32(train), cv2.ml.ROW_SAMPLE, np.float32(train_labels))
#num_iter = model.train(train, train_labels, None, params = params)

# set end time
e2 = cv2.getTickCount()
time = (e2 - e1)/cv2.getTickFrequency()
print 'Training duration:', time
#print 'Ran for %d iterations' % num_iter

# train data
ret_0, resp_0 = model.predict(train)
prediction_0 = resp_0.argmax(-1)
true_labels_0 = train_labels.argmax(-1)

train_rate = np.mean(prediction_0 == true_labels_0)
print 'Train accuracy: ', "{0:.2f}%".format(train_rate * 100)

# test data
ret_1, resp_1 = model.predict(test)
prediction_1 = resp_1.argmax(-1)
true_labels_1 = test_labels.argmax(-1)

test_rate = np.mean(prediction_1 == true_labels_1)
print 'Test accuracy: ', "{0:.2f}%".format(test_rate * 100)

# save model
model.save('mlp_xml/mlp.xml')
`

Maybe its something simple i'm just missing or some sort of bug.

Hope you can help me.

Best Regards,

Jon

hamuchiwa, 2017-12-26:
Hi, could you post the error?

Kaulinan, 2017-12-26:
Traceback (most recent call last):
  File "C:\Users\Jon\Desktop\AutoRCCar-master\computer\mlp_training.py", line 47, in <module>
    model.create(layer_sizes)
SystemError: error return without exception set

hamuchiwa, 2017-12-27:
Could it be a numpy issue? What is your numpy version?

Kaulinan, 2017-12-27:
1.14.0rc1

Kaulinan, 2018-01-05:
Hey hamuchiwa,

hope you had some amazing holidays.

did you found something that could help to solve the problem?

hamuchiwa, 2018-01-06:
Hi, I'm not able to recreate your error, but I still think this could be something relates to packages. Have you tried create a virtual environment and start fresh?

Kaulinan, 2018-01-07:
i can try that. May take some time.

Kaulinan, 2018-01-07:
can i send you my training data to test it with your script? maybe the training data is corrupt

hamuchiwa, 2018-01-08:
I doubt it but I can try that , can you provide a link that I can download from

Kaulinan, 2018-01-08:
https://we.tl/u1vtDiLhrz

the link is active for one day

hamuchiwa, 2018-01-08:
Your data works fine

Kaulinan, 2018-01-08:
Hmm. okay then i will try it with a fresh start

Kaulinan, 2018-01-10:
I made a fresh restart of the packages, yet i still get the same error. Any other solutions?

Kaulinan, 2018-01-10:
Numpy also went from an 1.14rc0 version to just 1.14

Kaulinan, 2018-01-10:
i also tried switching 

`layer_sizes = np.int32([28800, 32, 4])
model = cv2.ml_ANN_MLP(training_data)`

but then it just says its a read only attribute

hamuchiwa, 2018-01-10:
I noticed you put a param in MLP function, remove that from cv2.ml.ANN_MLP_create()

Kaulinan, 2018-01-10:
removed the parameter. it still shows the same error.

Traceback (most recent call last):
  File "C:\Users\Jon\Desktop\AutoRCCar-master\computer\mlp_training.py", line 47, in <module>
    model.create(layer_sizes)
SystemError: error return without exception set

hamuchiwa, 2018-01-10:
Okay, I tested your data in Python3 environment, it works on my mac, below is the script:
```python
import cv2
import numpy as np
import glob
import sys

print ("OpenCV:",  cv2.__version__)
print ("Numpy : ", np.__version__)
print ("Python:",  sys.version)

# load training data
dim = 28800
X = np.empty((0, dim))
y = np.empty((0, 4))
training_data = glob.glob('temp/*.npz')

for single_npz in training_data:
    with np.load(single_npz) as data:
        train = data['train']
        train_labels = data['train_labels']
    X = np.vstack((X, train))
    y = np.vstack((y, train_labels))

print ('Image array shape: ', X.shape)
print ('Label array shape: ', y.shape)

# create model
model = cv2.ml.ANN_MLP_create()
layer_sizes = np.int32([dim, 32, 32, 4])
model.setLayerSizes(layer_sizes)
model.setTrainMethod(cv2.ml.ANN_MLP_BACKPROP)
model.setActivationFunction(cv2.ml.ANN_MLP_SIGMOID_SYM, 2, 1)
model.setBackpropWeightScale(0.001)
#model.setBackpropMomentumScale(0.1)
#model.setTermCriteria((cv2.TERM_CRITERIA_COUNT, 20, 0.01))

# train
model.train(np.float32(X), cv2.ml.ROW_SAMPLE, np.float32(y))

# predict
ret, resp = model.predict(X)
prediction = resp.argmax(-1)
true_labels = y.argmax(-1)

train_rate = np.mean(prediction == true_labels)
print (prediction)
print ('Train accuracy: ', "{0:.2f}%".format(train_rate * 100))
```

Output:
```
OpenCV: 3.3.0
Numpy :  1.13.3
Python: 3.6.3 (default, Nov 24 2017, 16:59:30) 
[GCC 4.2.1 Compatible Apple LLVM 9.0.0 (clang-900.0.38)]

Image array shape:  (79, 28800)
Label array shape:  (79, 4)

[2 3 2 2 2 2 3 3 3 2 2 2 2 2 0 0 2 0 2 0 2 2 3 3 3 2 2 0 0 0 0 2 1 1 0 2 2
 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2
 0 0 0 0 0]
Train accuracy:  87.34%
```

Kaulinan, 2018-01-11:
well the error massage disappeared with python3, but the shell just restarts after telling me the image and the label array shape

OpenCV: 3.4.0
Numpy :  1.14.0
Python: 3.6.4 (v3.6.4:d48eceb, Dec 19 2017, 06:04:45) [MSC v.1900 32 bit (Intel)]
Image array shape:  (0, 28800)
Label array shape:  (0, 4)

=============================== RESTART: Shell ===============================
>>> 

also i noticed the 0 at image and laber array shape (0, 28800), (0, 4)

Kaulinan, 2018-01-11:
IT WORKED.

i just had to change the directory of the training data in the script

 RESTART: C:\Users\Jon\Desktop\AutoRCCar-master\computer\hamuchiwa github.py 
OpenCV: 3.4.0
Numpy :  1.14.0
Python: 3.6.4 (v3.6.4:d48eceb, Dec 19 2017, 06:04:45) [MSC v.1900 32 bit (Intel)]
Image array shape:  (43, 28800)
Label array shape:  (43, 4)
[2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 0 0 0 0 0
 0 0 0 0 0 0]
Train accuracy:  100.00%

Kaulinan, 2018-01-17:
I think that i need to do some changes to the rc_driver.py too. can i just compare the cv2 stuff that changed between rc_driver.py and mlp_training.py?

Kaulinan, 2018-01-17:
the rc_driver.py now sends out the following error probably due to him not knowing what to do with the model:

Traceback (most recent call last):
  File "C:\Users\Jon\Desktop\AutoRCCar-master\computer\rc_driver.py", line 150, in <module>
    class VideoStreamHandler(socketserver.StreamRequestHandler):
  File "C:\Users\Jon\Desktop\AutoRCCar-master\computer\rc_driver.py", line 159, in VideoStreamHandler
    model.create()
  File "C:\Users\Jon\Desktop\AutoRCCar-master\computer\rc_driver.py", line 26, in create
    self.model.create(layer_size)
SystemError: <built-in method create of type object at 0x69A914E0> returned NULL without setting an error

lakshand8, 2018-01-17:
@Kaulinan How did you exactly fix the 'model.create(layer_sizes) SystemError: error return without exception set' error?

Kaulinan, 2018-01-17:
@lakshand8 I switched from Python 2.7.10 to 3.6 and just used the script @hamuchiwa pasted in this issue

Kaulinan, 2018-01-17:
import cv2
import numpy as np
import glob
import sys

print ("OpenCV:",  cv2.__version__)
print ("Numpy : ", np.__version__)
print ("Python:",  sys.version)

# load training data
dim = 28800
X = np.empty((0, dim))
y = np.empty((0, 4))
training_data = glob.glob('temp/*.npz')

for single_npz in training_data:
    with np.load(single_npz) as data:
        train = data['train']
        train_labels = data['train_labels']
    X = np.vstack((X, train))
    y = np.vstack((y, train_labels))

print ('Image array shape: ', X.shape)
print ('Label array shape: ', y.shape)

# create model
model = cv2.ml.ANN_MLP_create()
layer_sizes = np.int32([dim, 32, 32, 4])
model.setLayerSizes(layer_sizes)
model.setTrainMethod(cv2.ml.ANN_MLP_BACKPROP)
model.setActivationFunction(cv2.ml.ANN_MLP_SIGMOID_SYM, 2, 1)
model.setBackpropWeightScale(0.001)
#model.setBackpropMomentumScale(0.1)
#model.setTermCriteria((cv2.TERM_CRITERIA_COUNT, 20, 0.01))

# train
model.train(np.float32(X), cv2.ml.ROW_SAMPLE, np.float32(y))

# predict
ret, resp = model.predict(X)
prediction = resp.argmax(-1)
true_labels = y.argmax(-1)

train_rate = np.mean(prediction == true_labels)
print (prediction)
print ('Train accuracy: ', "{0:.2f}%".format(train_rate * 100))

if you are using windows you need to change the path of the training data

lakshand8, 2018-01-17:
@Kaulinan thanks for the reply. So I managed to get @hamuchiwa's code working on Python 2.7 with the changes you made in the above comments. but when I run the program i get the output below: 

`Loading training data...
Image array shape:  (384, 38400)
Label array shape:  (384, 4)
Loading image duration: 0.872863756846
Training MLP ...
Training duration: 34.5171404085
Train accuracy:  28.36%
Test accuracy:  25.00%
Ran for 1 iterations
Prediction: [2 3 0 2 0 2 2 0 2 2 0 3 2 2 0 2 0 0 0 0 2 2 2 0 0 2 0 0 2 2 0 2 2 0 2 2 0
 2 2 0 2 2 0 2 2 0 2 3 2 3 2 2 3 2 2 2 2 2 0 2 2 2 2 2 2 0 0 2 3 3 0 3 2 2
 2 2 2 0 2 2 0 2 3 3 2 2 2 0 2 2 2 0 2 0 0 0 2 0 2 2 0 2 0 0 0 2 0 3 0 0 3
 0 2 0 2 2 0 2 0 2 3 0 0 2 2 2 2 0 2 2 2 0 2 2 0 3 2 2 3 2 3 0 2 2 2 0 2 0
 2 2 0 0 2 0 2 2 0 2 0 3 2 0 2 2 2 2 0 2 3 2 2 0 0 0 2 0 0 2 2 2 2 0 2 2 2
 2 2 0 0 2 2 0 3 0 2 0 2 2 0 2 2 0 3 2 2 2 3 2 2 2 0 0 0 2 0 3 2 0 2 2 0 0
 3 2 0 0 2 2 2 2 2 2 0 2 0 0 2 3 0 0 0 2 0 0 3 3 0 2 0 2 2 2 2 0 3 2 0 0 3
 0 3 2 0 0 2 0 2 2]
True labels: [1 0 2 1 2 1 2 2 2 1 3 0 1 2 2 2 2 2 2 1 2 2 1 1 1 2 2 1 1 2 3 2 1 1 0 2 2
 1 2 2 2 1 1 0 2 2 0 3 1 3 2 1 3 0 2 2 0 1 2 1 2 0 2 2 2 1 3 2 0 0 3 0 2 0
 1 0 2 2 2 2 1 0 0 0 2 1 1 2 0 2 0 1 1 3 3 3 2 3 1 0 1 2 2 1 2 2 1 0 2 2 0
 3 2 3 1 2 2 2 2 1 3 3 3 0 2 0 1 3 2 1 2 2 0 1 3 0 1 0 3 0 0 3 0 1 2 1 0 1
 2 1 3 3 2 1 1 2 2 2 1 0 1 1 2 0 2 0 1 0 0 2 1 1 2 2 2 3 2 1 2 1 0 1 2 0 2
 1 2 3 1 1 2 2 0 3 1 1 2 1 2 2 1 1 0 0 0 2 3 0 0 2 1 1 2 2 3 0 2 1 2 2 3 2
 0 2 1 3 2 0 2 0 1 1 3 2 2 3 2 0 3 2 2 2 2 2 0 0 1 0 2 1 0 1 2 2 3 2 1 2 0
 1 3 2 1 2 2 2 0 0]
Testing...
Train rate: 28.358209:`

These are just random pictures i captured (24 .npz files) just to test out the code, so is that why the accuracy is very low? and also I'm using image_array = np.zeros((1, 38400)) and label_array = np.zeros((1, 4), 'float'). Do I need to change these values or can I keep them the same? Thank you.

Kaulinan, 2018-01-17:
i think you can keep the values, the accuracy might be because of the random pictures taken for the training.

lakshand8, 2018-01-17:
@Kaulinan  Thank you.. Just roughly how many pictures should I aim for increasing the accuracy? and also If I run the script couple of more times while driving around the track for higher accuracy, can I use all the mlp.xml files I've gathered in the rc_drive.py script?

aakashm301, 2018-03-19:
**I am also having an error in rc_drive.py**
C:\Python27\python.exe C:/Users/Aakash/Desktop/AutoRCCar-master/computer/rc_driver.py
Traceback (most recent call last):

  File "C:/Users/Aakash/Desktop/AutoRCCar-master/computer/rc_driver.py", line 145, in <module>
    class VideoStreamHandler(SocketServer.StreamRequestHandler):

  File "C:/Users/Aakash/Desktop/AutoRCCar-master/computer/rc_driver.py", line 153, in VideoStreamHandler
    model = NeuralNetwork()
  
File "C:/Users/Aakash/Desktop/AutoRCCar-master/computer/rc_driver.py", line 17, in __init__
    self.model = cv2.ANN_MLP()
AttributeError: 'module' object has no attribute 'ANN_MLP'

Process finished with exit code 1
 
**Any suggestions ?**

aakashm301, 2018-03-20:
I did as you said and changed it to opencv3 but then again I am getting this error

C:\ProgramData\Anaconda3\python.exe "C:/Users/Aakash/Desktop/AutoRCCar-master/computer/rc_driver 3.py"
Traceback (most recent call last):
  File "C:/Users/Aakash/Desktop/AutoRCCar-master/computer/rc_driver 3.py", line 145, in <module>
    class VideoStreamHandler(socketserver.StreamRequestHandler):
  File "C:/Users/Aakash/Desktop/AutoRCCar-master/computer/rc_driver 3.py", line 153, in VideoStreamHandler
    model = NeuralNetwork()
  File "C:/Users/Aakash/Desktop/AutoRCCar-master/computer/rc_driver 3.py", line 17, in __init__
    self.model = cv2.ANN_MLP()
AttributeError: module 'cv2' has no attribute 'ANN_MLP'

Process finished with exit code 1

hamuchiwa, 2018-03-21:
If you plan to do it in OpenCV3, and the model has been trained (refer to my comment above), the wrapper class needs modifications too, for example, something like this:
```python
class NeuralNetwork(object):
    def __init__(self, file_path):
        self.model = cv2.ml.ANN_MLP_load(file_path)

    def predict(self, samples):
        ret, resp = self.model.predict(samples)
        return resp.argmax(-1)

# create neural network
nn = NeuralNetwork('mlp_xml/mlp.xml')
prediction = nn.predict(X)
```

tkajbaje, 2018-03-30:
I am getting below error while executing the code which hamuchiwa posted.
`('OpenCV:', '3.4.1-dev')
('Numpy : ', '1.14.1')
('Python:', '2.7.14 (default, Sep 23 2017, 22:06:14) \n[GCC 7.2.0]')
Traceback (most recent call last):
  File "./mlp_training.py", line 20, in <module>
    X = np.vstack((X, train))
  File "/home/tanmay/.local/lib/python2.7/site-packages/numpy/core/shape_base.py", line 234, in vstack
    return _nx.concatenate([atleast_2d(_m) for _m in tup], 0)
ValueError: all the input array dimensions except for the concatenation axis must match exactly
`

dheerendra-arc, 2018-04-03:
Traceback (most recent call last):
  File "rc_driver.py", line 146, in <module>
    class VideoStreamHandler(socketserver.StreamRequestHandler):
  File "rc_driver.py", line 155, in VideoStreamHandler
    prediction = nn.predict(X)
NameError: name 'X' is not defined

MuhammadAnwarKhan, 2018-04-12:
@hamuchiwa @dheerendra-arc 
How to solve this error? I'm using openCV3.
Traceback (most recent call last):
  File "C:\Users\anwar\Desktop\AutoRCCar-master\computer\rc_driver.py", line 152, in <module>
    prediction = nn.predict(X)
NameError: name 'X' is not defined
[Finished in 1.5s]

hamuchiwa, 2018-04-12:
@dheerendra-arc @dheerendra-arc 
Sorry for the confusion, I was just making an example on how to make a wrapper, `X` is the input data that defined somewhere else.

Moha-alfridi, 2018-04-18:
So, I change NeuralNetwork class to the above one that you provide, however, I had this error 

```
from utils.utils import *
ModuleNotFoundError: No module named 'utils.utils'
```

hamuchiwa, 2018-04-18:
@Moha-alfridi There is no `utils.py` in this project.

Moha-alfridi, 2018-04-19:
Thank you for the fast respond, and I am so sorry for the confusion, I was accidentally running my sketch.

I switched back to your sketch and I have the same error that @aakashm301 had which is 

```
Traceback (most recent call last):
  File "/Users/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py", line 146, in <module>
    class VideoStreamHandler(SocketServer.StreamRequestHandler):
  File "/Users/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py", line 153, in VideoStreamHandler
    model = NeuralNetwork()
  File "/Users/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py", line 17, in __init__
    self.model = cv2.ANN_MLP()
AttributeError: 'module' object has no attribute 'ANN_MLP'
```

Line 17 : 
`self.model = cv2.ANN_MLP()`
Line 146 : 
`class VideoStreamHandler(SocketServer.StreamRequestHandler):`
Line 153 : 
`model = NeuralNetwork()`

This is on the `rc_driver.py`
I am running python2.7

and If I did this 

> class NeuralNetwork(object):
> 
>     def __init__(self):
>         self.model = cv2.ml.ANN_MLP_load('mlp_xml/mlp.xml')
> 
>     #def create(self):
>         #layer_size = np.int32([38400, 32, 4])
>         #self.model.create(layer_size)
>         #self.model.load('mlp_xml/mlp.xml')
> 
>     def predict(self, samples):
>         ret, resp = self.model.predict(samples)
>         return resp.argmax(-1)

I get this error 

```
Traceback (most recent call last):
  File "/Users/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py", line 146, in <module>
    class VideoStreamHandler(socketserver.StreamRequestHandler):
  File "/Users/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py", line 154, in VideoStreamHandler
    model.create()
AttributeError: 'NeuralNetwork' object has no attribute 'create'
```

Thanks in advance

Moha-alfridi, 2018-04-19:
So, I was apply to modify the code. the code is working, however it show some error and the car doesn't move, but the distance measurement is kept on updating until I stooped the project
here is the error 

> /Users/moha/.virtualenvs/stream_server_test.py/bin/python /Users/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py
> 4.8
> 229.4
> 231.1
> [ INFO:0] Initialize OpenCL runtime...
> 229.0
> 3199.5
> OpenCV Error: Assertion failed ((type == 5 || type == 6) && inputs.cols == layer_sizes[0]) in predict, file /Users/travis/build/skvark/opencv-python/opencv/modules/ml/src/ann_mlp.cpp, line 411
> Traceback (most recent call last):
>   File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/SocketServer.py", line 295, in _handle_request_noblock
>     self.process_request(request, client_address)
>   File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/SocketServer.py", line 321, in process_request
>     self.finish_request(request, client_address)
>   File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/SocketServer.py", line 334, in finish_request
>     self.RequestHandlerClass(request, client_address, self)
>   File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/SocketServer.py", line 655, in __init__
>     self.handle()
>   File "/Users/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py", line 212, in handle
>     prediction = self.model.predict(image_array)
>   File "/Users/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py", line 25, in predict
>     ret, resp = self.model.predict(samples)
> error: /Users/travis/build/skvark/opencv-python/opencv/modules/ml/src/ann_mlp.cpp:411: error: (-215) (type == 5 || type == 6) && inputs.cols == layer_sizes[0] in function predict
> 
> Connection closed on thread 1
> ----------------------------------------
> Exception happened during processing of request from ('192.168.1.195', 60744)
> ----------------------------------------
> 5.2
> 230.7
> 229.8
> 230.2
> 377.9
> 229.4
> 230.7
> 3197.8
> 5.2
> 229.8
> 229.8
> 3202.4
> 5.2
> 230.2

Moha-alfridi, 2018-04-19:
I tried to run it on python 3.6 it worked but the image frame is lagging I didn't show anything or listen when I press 'q' to exit
Both error are pointing to `socketserver.py` file, I believe :\

[image]

Moha-alfridi, 2018-04-19:
The error keep getting larger and larger, so the distance is coming from the Rpi however the camera is lagged as the above picture are showing and this is the new error

```
Connected to pydev debugger (build 181.4203.547)
215.7
180.3
179.0
[ INFO:0] Initialize OpenCL runtime...
179.5
180.7
Connection closed on thread 1
OpenCV Error: Assertion failed ((type == 5 || type == 6) && inputs.cols == layer_sizes[0]) in predict, file /Users/travis/build/skvark/opencv-python/opencv/modules/ml/src/ann_mlp.cpp, line 411
----------------------------------------
Exception happened during processing of request from ('192.168.1.195', 36340)
Traceback (most recent call last):
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/socketserver.py", line 317, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/socketserver.py", line 348, in process_request
    self.finish_request(request, client_address)
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/socketserver.py", line 361, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/socketserver.py", line 696, in __init__
    self.handle()
  File "/Users/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py", line 212, in handle
    prediction = self.model.predict(image_array)
  File "/Users/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py", line 25, in predict
    ret, resp = self.model.predict(samples)
cv2.error: /Users/travis/build/skvark/opencv-python/opencv/modules/ml/src/ann_mlp.cpp:411: error: (-215) (type == 5 || type == 6) && inputs.cols == layer_sizes[0] in function predict

----------------------------------------
178.3
179.5
179.0
179.0
180.7
180.3
179.1
Connection closed on thread 2
----------------------------------------
Exception happened during processing of request from ('192.168.1.195', 57568)
Traceback (most recent call last):
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/socketserver.py", line 317, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/socketserver.py", line 348, in process_request
    self.finish_request(request, client_address)
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/socketserver.py", line 361, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/socketserver.py", line 696, in __init__
    self.handle()
  File "/Users/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py", line 139, in handle
    sensor_data = round(float(self.data), 1)
ValueError: could not convert string to float: b'216.48567915179.473149776'
----------------------------------------
Traceback (most recent call last):
  File "/Applications/PyCharm CE.app/Contents/helpers/pydev/pydevd.py", line 1664, in <module>
    main()
  File "/Applications/PyCharm CE.app/Contents/helpers/pydev/pydevd.py", line 1658, in main
    globals = debugger.run(setup['file'], None, None, is_module)
  File "/Applications/PyCharm CE.app/Contents/helpers/pydev/pydevd.py", line 1068, in run
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "/Applications/PyCharm CE.app/Contents/helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "/Users/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py", line 275, in <module>
    class ThreadServer(object):
  File "/Users/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py", line 287, in ThreadServer
    video_thread = threading.Thread(target=server_thread('192.168.1.188', 6000))
  File "/Users/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py", line 279, in server_thread
    server.serve_forever()
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/socketserver.py", line 236, in serve_forever
    ready = selector.select(poll_interval)
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/selectors.py", line 376, in select
    fd_event_list = self._poll.poll(timeout)
KeyboardInterrupt

Process finished with exit code 137 (interrupted by signal 9: SIGKILL)
```

The sketch was not working at the beginning, so I have made changes to this part and couple more on the way. "The changes were all related to the cv2.ANN_MLP"

```
class NeuralNetwork(object):

    def __init__(self):
        self.model = cv2.ANN_MLP()

    def create(self):
        layer_size = np.int32([38400, 32, 4])
        self.model.create(layer_size)
        self.model.load('mlp_xml/mlp.xml')

    def predict(self, samples):
        ret, resp = self.model.predict(samples)
        return resp.argmax(-1)
```

to 

```
class NeuralNetwork(object):

    def __init__(self):
        self.model = cv2.ml.ANN_MLP_create()

    def create(self):
        layer_size = np.int32([38400, 32, 4])
        self.model.setLayerSizes(layer_size)
        self.model = cv2.ml.ANN_MLP_load('mlp_xml/mlp.xml')

    def predict(self, samples):
        ret, resp = self.model.predict(samples)
        return resp.argmax(-1)
```

hamuchiwa, 2018-04-20:
@Moha-alfridi So after the changes, it's working fine now...?

Moha-alfridi, 2018-04-20:
@hamuchiwa No.
So, I was apply to make it work on OpenCV3 by changing ANN_MLP and couple of line here and there.
The code operate, but show's me some error and I believe it is because a deference in the array size.
here is the code I am working on ..

[rc_driver_copy.py.zip](https://github.com/hamuchiwa/AutoRCCar/files/1933610/rc_driver_copy.py.zip)

and this is the error that its show ..

```
[ INFO:0] Initialize OpenCL runtime...
OpenCV Error: Assertion failed ((type == 5 || type == 6) && inputs.cols == layer_sizes[0]) in predict, file /Users/travis/build/skvark/opencv-python/opencv/modules/ml/src/ann_mlp.cpp, line 411
----------------------------------------
Exception happened during processing of request from ('192.168.1.195', 42420)
Traceback (most recent call last):
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/socketserver.py", line 317, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/socketserver.py", line 348, in process_request
    self.finish_request(request, client_address)
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/socketserver.py", line 361, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/socketserver.py", line 696, in __init__
    self.handle()
  File "/Users/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py", line 217, in handle
    prediction = self.model.predict(image_array)
  File "/Users/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py", line 25, in predict
    ret, resp = self.model.predict(samples)
cv2.error: /Users/travis/build/skvark/opencv-python/opencv/modules/ml/src/ann_mlp.cpp:411: error: (-215) (type == 5 || type == 6) && inputs.cols == layer_sizes[0] in function predict

----------------------------------------
(1, 28800)
Connection closed on thread 1
```

so the image Fram come up, but its lagging here is a picture that show it

[image]

I hope that you can help with it & thank you for keep coming back 😄

hamuchiwa, 2018-04-20:
@Moha-alfridi You are right, assertion error usually means shape doesn't match, what's the resolution of incoming frame?

dheerendra-arc, 2018-04-21:
Providing the path of the image folder in the place of X? Is that correct to predict.

Moha-alfridi, 2018-04-21:
@hamuchiwa 
Is it possible that the shape unmatched problem is from my trained .mxl file ? The car was really fast, and unclear picture, here is a sample 

[image]

Excuse the wires and the sensor, after I made sure everything is working will start on fixing the design. "I hope that is not the problem :( "

I started to thinks that way after I run out of ideas to fix the code 😄

Moha-alfridi, 2018-04-22:
@hamuchiwa 
so, I just made sure the resolution is (240x320)

by this couple lines
```
h, w =  image.shape[:2] 
print("The resolution is : "+str(h)+" "+str(w)) 
```
the output was 
`The resolution is : 240 320`

Moha-alfridi, 2018-04-23:
So, I have changed 

```
gray = cv2.imdecode(np.fromstring(jpg, dtype=np.uint8), cv2.cv2.IMREAD_GRAYSCALE)
image = cv2.imdecode(np.fromstring(jpg, dtype=np.uint8), cv2.cv2.IMREAD_GRAYSCALE)
```

to 

```
gray = cv2.imdecode(np.fromstring(jpg, dtype=np.uint8), cv2.cv2.IMREAD_LOAD_GDAL)
image = cv2.imdecode(np.fromstring(jpg, dtype=np.uint8), cv2.cv2.IMREAD_LOAD_GDAL)

```

and the previous error has disappear, instead this one came out

```
The resolution is : 240 320
[ INFO:0] Initialize OpenCL runtime...
Traceback (most recent call last):
Connection closed on thread 1
----------------------------------------
Exception happened during processing of request from ('192.168.1.195', 44360)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/SocketServer.py", line 295, in _handle_request_noblock
----------------------------------------
    self.process_request(request, client_address)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/SocketServer.py", line 321, in process_request
    self.finish_request(request, client_address)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/SocketServer.py", line 334, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/SocketServer.py", line 655, in __init__
    self.handle()
  File "/Users/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py", line 213, in handle
    image_array = half_gray.reshape(1, 38400).astype(np.float32)
ValueError: cannot reshape array of size 115200 into shape (1,38400)
```

Can you help please ..

hamuchiwa, 2018-04-23:
@Moha-alfridi Check your image size, if resolution is 240x320, then the image array shape is (1, 240x320=76800) or (1, 120x320=38400) if you use lower half of the image.

Moha-alfridi, 2018-04-23:
I am taking the lower half 

`half_gray = gray[120:240, :]`
and set it up to 
`image_array = half_gray.reshape(1, 28800).astype(np.float32)`

and tried your values, and tried 240x320=76800 before and still trapped :(

Moha-alfridi, 2018-04-23:
I train a new .xml file. it generate a good accuracy +%95
This is a sample of one of the image

[image]

and even though, the array size didn't match

```
The resolution is : 240 320
[ INFO:0] Initialize OpenCL runtime...
Connection closed on thread 1
Traceback (most recent call last):
----------------------------------------
Exception happened during processing of request from ('192.168.1.195', 42160)
----------------------------------------
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/SocketServer.py", line 295, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/SocketServer.py", line 321, in process_request
    self.finish_request(request, client_address)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/SocketServer.py", line 334, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/SocketServer.py", line 655, in __init__
    self.handle()
  File "/Users/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py", line 213, in handle
    image_array = half_gray.reshape(1, 76800).astype(np.float32)
ValueError: cannot reshape array of size 115200 into shape (1,76800)
```

and if set the `image_array` to 115200 it give me the `Assertion failed` error.

I believe I get these error because the modification from OpenCV2 to OpenCV3 since the `Assertion failed` error only disappear when I change 

```
gray = cv2.imdecode(np.fromstring(jpg, dtype=np.uint8), cv2.cv2.IMREAD_GRAYSCALE)
image = cv2.imdecode(np.fromstring(jpg, dtype=np.uint8), cv2.cv2.IMREAD_GRAYSCALE)
```
to

```
gray = cv2.imdecode(np.fromstring(jpg, dtype=np.uint8), cv2.cv2.IMREAD_LOAD_GDAL)
image = cv2.imdecode(np.fromstring(jpg, dtype=np.uint8), cv2.cv2.IMREAD_LOAD_GDAL)

```

what do you think

hamuchiwa, 2018-04-24:
@Moha-alfridi `image_array ` should be a gray image that has shape of `(120, 320)`. Check the shape of `image_array`, you are getting 115200 so I guess it's because 120x320x3=115200 which is a 3 channels color image.

Moha-alfridi, 2018-04-26:
So @hamuchiwa , back again but this time I have install Ubuntu OS in virtual box. I install OpenCV2.4.13
well I started to believe it's the `.xml` file that wont be loaded here is the error that I got

```
/home/moha/Desktop/AutoRCCar/venv/bin/python /home/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py
('OpenCV:', '2.4.13')
OpenCV Error: Unspecified error (The node does not represent a user object (unknown type?)) in cvRead, file /home/moha/opencv-2.4.13/modules/core/src/persistence.cpp, line 5008
Traceback (most recent call last):
  File "/home/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py", line 148, in <module>
    class VideoStreamHandler(SocketServer.StreamRequestHandler):
  File "/home/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py", line 156, in VideoStreamHandler
    model.create()
  File "/home/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py", line 24, in create
    self.model.load('/home/moha/Desktop/AutoRCCar/computer/mlp_xml/mlp.xml')
cv2.error: /home/moha/opencv-2.4.13/modules/core/src/persistence.cpp:5008: error: (-2) The node does not represent a user object (unknown type?) in function cvRead

Process finished with exit code 1
```

and the sketch that I am working on is the one that in github "Only changed the IP address & the directory of `.xml ` file"

I download the opencv2.4.13 on virtual box because I want to test it first, try to figure out the problem, and if possible solve it, and also because I am tired of formating my Mac -_-

any suggestion ?

Moha-alfridi, 2018-04-26:
so that is my NeuralNetwork class and I create the `file_path` before the class

```
file_path = 'Desktop/AutoRCCar/computer/mlp_xml/mlp.xml'
class NeuralNetwork(object):

    def __init__(self, file_path):
        self.model = cv2.ml.ANN_MLP_load(file_path)

    def create(self):
        layer_size = np.int32([38400, 32, 4])
        self.model.setLayerSizes(layer_size)

    def predict(self, samples):
        ret, resp = self.model.predict(samples)
        return resp.argmax(-1)
```

and this error came up
```
Traceback (most recent call last):
('OpenCV:', '3.4.0')
  File "/Users/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py", line 149, in <module>
    class VideoStreamHandler(socketserver.StreamRequestHandler):
  File "/Users/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py", line 156, in VideoStreamHandler
    model = NeuralNetwork()
TypeError: __init__() takes exactly 2 arguments (1 given)
```

Moha-alfridi, 2018-04-27:
@hamuchiwa
Should I just install OpenCV 2.4.13 and work it out, or keep looking for soultions with OpenCV3.4.1

This project has already take all of my time and I barely make any progress 😔
Please I seek your guidance 👏🏼 Help me out please

Moha-alfridi, 2018-04-27:
@hamuchiwa 
So, the NN class is

```
NeuralNetwork(object):

    def __init__(self, file_path):
        self.model = cv2.ml.ANN_MLP_load(file_path)

    def create(self):
        layer_size = np.int32([38400, 32, 4])
        self.model.setLayerSizes(layer_size)
        self.model.load('/Users/moha/Desktop/AutoRCCar/computer/mlp_xml/mlp.xml')

    def predict(self, samples):
        ret, resp = self.model.predict(samples)
        return resp.argmax(-1)
```

and according to your post issue #65 I included the path in
```
# create neural network
    model = NeuralNetwork('/Users/moha/Desktop/AutoRCCar/computer/mlp_xml/mlp.xml')
```

and now this is the error that show up

```
/Users/moha/.virtualenvs/project1/bin/python /Users/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py
OpenCV: 3.4.0
(240, 320)
(120, 320)
[ INFO:0] Initialize OpenCL runtime...
1
OpenCV Error: Unspecified error (The network has not been trained or loaded) in predict, file /Users/travis/build/skvark/opencv-python/opencv/modules/ml/src/ann_mlp.cpp, line 405
Connection closed on thread 1
----------------------------------------
Exception happened during processing of request from ('10.0.0.72', 44340)
Traceback (most recent call last):
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/socketserver.py", line 317, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/socketserver.py", line 348, in process_request
    self.finish_request(request, client_address)
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/socketserver.py", line 361, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/socketserver.py", line 696, in __init__
    self.handle()
  File "/Users/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py", line 222, in handle
    prediction = self.model.predict(image_array)
  File "/Users/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py", line 28, in predict
    ret, resp = self.model.predict(samples)
cv2.error: /Users/travis/build/skvark/opencv-python/opencv/modules/ml/src/ann_mlp.cpp:405: error: (-2) The network has not been trained or loaded in function predict

----------------------------------------

```

same as the error that was showing 2 days ago, do you have trained network that I can test it with? or any other ideas?

Moha-alfridi, 2018-04-29:
Just to make sure that I am working as you think
I didn't define `file_path` anywhere in the code and my `NeuralNetwork` class is

```
class NeuralNetwork(object):

    def __init__(self, file_path):
        self.model = cv2.ml.ANN_MLP_load(file_path)

    def create(self):
        layer_size = np.int32([38400, 32, 4])
        self.model.setLayerSizes(layer_size)
        self.model.load('/Users/moha/Desktop/AutoRCCar/computer/mlp_xml/mlp.xml')

    def predict(self, samples):
        ret, resp = self.model.predict(samples)
        return resp.argmax(-1)
```

I print my `half_image` shape and got `(1, 38400)` and printed the size which is `38400`
and I have commented 

`prediction = self.model.predict(image_array)`

and pasted 
```
nn = NeuralNetwork('mlp.xml')
prediction = nn.predict(image_array)
```

this modification is according to your last comment

and this is the error that I get ...

```
/Users/moha/.virtualenvs/project1/bin/python /Users/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py
OpenCV: 3.4.0
(240, 320)
76800
(120, 320)
38400
[ INFO:0] Initialize OpenCL runtime...
OpenCV Error: Assertion failed (fs.isOpened()) in load, file /Users/travis/build/skvark/opencv-python/opencv/modules/ml/src/ann_mlp.cpp, line 1584
----------------------------------------
Exception happened during processing of request from ('10.0.0.72', 38558)
Traceback (most recent call last):
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/socketserver.py", line 317, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/socketserver.py", line 348, in process_request
    self.finish_request(request, client_address)
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/socketserver.py", line 361, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/socketserver.py", line 696, in __init__
    self.handle()
  File "/Users/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py", line 223, in handle
    nn = NeuralNetwork('mlp.xml')
  File "/Users/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py", line 19, in __init__
    self.model = cv2.ml.ANN_MLP_load(file_path)
cv2.error: /Users/travis/build/skvark/opencv-python/opencv/modules/ml/src/ann_mlp.cpp:1584: error: (-215) fs.isOpened() in function load

----------------------------------------
Connection closed on thread 1
```

Should I define `file_path` anywhere in the code ?
I am not sure why the `Assertion` error keep coming up, I made sure that the shape and the sizes correct as what we discussed earlier.

hamuchiwa, 2018-04-30:
@Moha-alfridi Remove the `create` function, the model is already loaded. Initialize the object with file path `nn = NeuralNetwork('your_xml_file_path')`.

Moha-alfridi, 2018-04-30:
@hamuchiwa So,
I did as you said and I got this error

```
File "/Users/moha/Desktop/AutoRCCar/computer/rc_driver_copy.py", line 221, in handle
    nn = NeuralNetwork('mlp_xml/mlp.xml')
TypeError: __init__() takes 1 positional argument but 2 were given
```

here is the whole code so that you can see exactly where is the problem at 

```
#__author__ = 'zhengwang'

import threading
import socketserver
import serial
import cv2
import numpy as np
import math

# distance data measured by ultrasonic sensor
sensor_data = " "

print("OpenCV:",  cv2.__version__)

class NeuralNetwork(object):

    def __init__(self):
        self.model = cv2.ml.ANN_MLP_create()

    #def create(self):
        #layer_size = np.int32([38400, 32, 4])
        #self.model.setLayerSizes(layer_size)
        #self.model = cv2.ml.ANN_MLP_load('mlp_xml/mlp.xml')

    def predict(self, samples):
        ret, resp = self.model.predict(samples)
        return resp.argmax(-1)

class RCControl(object):

    def __init__(self):
        self.serial_port = serial.Serial('/dev/tty.usbmodem1421', 115200, timeout=1)

    def steer(self, prediction):
        if prediction == 2:
            self.serial_port.write(chr(1))
            print("Forward")
        elif prediction == 0:
            self.serial_port.write(chr(7))
            print("Left")
        elif prediction == 1:
            self.serial_port.write(chr(6))
            print("Right")
        else:
            self.stop()

    def stop(self):
        self.serial_port.write(chr(0))

class DistanceToCamera(object):

    def __init__(self):
        # camera params
        self.alpha = 8.0 * math.pi / 180
        self.v0 = 119.865631204
        self.ay = 332.262498472

    def calculate(self, v, h, x_shift, image):
[... 213 lines cut ...]
                        break

            cv2.destroyAllWindows()

        finally:
            print("Connection closed on thread 1")

class ThreadServer(object):

    def server_thread(host, port):
        server = socketserver.TCPServer((host, port), VideoStreamHandler)
        server.serve_forever()

    def server_thread2(host, port):
        server = socketserver.TCPServer((host, port), SensorDataHandler)
        server.serve_forever()

    distance_thread = threading.Thread(target=server_thread2, args=('10.0.0.70', 6002))
    distance_thread.start()
    video_thread = threading.Thread(target=server_thread('10.0.0.70', 6000))
    video_thread.start()

if __name__ == '__main__':
    ThreadServer()
```

hamuchiwa, 2018-04-30:
@Moha-alfridi Hi, see the `NeuralNetwork` class in my previous comment. Use `cv2.ml.ANN_MLP_load(file_path)` in driving, while use `cv2.ml.ANN_MLP_create()` in training.

[34 more replies]