hamuchiwa/AutoRCCar #61: Wiring diagram for RC remote.

ederghar, 2017-12-28:
Hi!
I have purchased the same RC car and remote as illustrated in the examples.  Can you provide a wiring diagram or instructions on soldering wires to Arduino?

Thank you,

Ed.

hamuchiwa, 2017-12-28:
This is one of the pictures I took before, pick four pins from arduino and connect them to the controller, please note the black wire is not used. 
[image]

ederghar, 2017-12-28:
Hi,With this wiring, I'm only able to turn left and move forward.  I can't go right or reverse.  Is this correct?
Thank you,
Ed.

hamuchiwa, 2017-12-29:
Yes, make sure four pins correspond to four direction.

ederghar, 2017-12-30:
Thanks for getting back to me.   I've connected pin 6 to turn right, 7 to turn left, 10 forward, and 9 reverse just like your picture.  It's hard to see in your picture if 9 and 10 are soldered together or jumpered somehow.  I'm controlling Arduino Uno R3 from Windows 10 and just to make sure the remote control buttons work correctly.
Here's a picture of my connections.  I had to jumper 9 and 10 to get it to go forward.  Otherwise even forward didn't work.  Unfortunately Reverse is also going forward now and Right is not working at all.  The remote control light comes on, but the car does not respond. Any help you give is greatly appreciated!  This is the final piece before I start training mode.

hamuchiwa, 2017-12-30:
They are not soldered together. I don't think the remote control buttons will work anymore, when you press direction keys on the keyboard, the light on remote control should come on.

ederghar, 2018-01-01:
What kind of Arduino board are you using?  Manufacturer and version?

hamuchiwa, 2018-01-02:
I believe the one shown in the picture is a aftermarket UNO (I had two back then), but it shouldn't matter, it works the same way as the official one.

ederghar, 2018-01-03:
I was able to finally resolve the issue by level shifting the connection to the remote control from 5V to 2.5V and connecting to Arduino ground.  The same problem I had was encountered by another person and was solved by level shifting:
https://arduino.stackexchange.com/questions/21205/controlling-an-rc-car-using-arduino

hamuchiwa, 2018-01-03:
Nice find, thanks a lot~

closed 2018-01-03 by ederghar

Gooose1909, 2018-03-12:
Was the battery connected to the remote ??

edPratt, 2018-11-28:
@hamuchiwa so you connect the arduino to the controller and not the rc car itself? I have seen other posts online for converting rc cars to use arduino but they connect directly to the board on the rc car.

Sanjangowda97, 2018-12-03:
what is the another hardware along with the arduino board 
can u please send the connection diagram

hamuchiwa, 2018-12-03:
@edPratt @Sanjangowda97 Arduino connects to the controller directly, see #114. The connection does the job but is far from perfect, please feel free to improve the way on controlling the rc car.

sundar19, 2019-01-14:
> Hi!
> I have purchased the same RC car and remote as illustrated in the examples. Can you provide a wiring diagram or instructions on soldering wires to Arduino?
> 
> Thank you,
> 
> Ed.

Mr.Ed where did you buy same RC car can i have the link? thank you

ederghar, 2019-01-14:
I purchased it through Amazon:
https://www.amazon.com/RASTAR-Land-Rover-Range-Sport/dp/B07CPNT9R4/ref=asc_df_B07CPNT9R4/?tag=hyprod-20&linkCode=df0&hvadid=241944100740&hvpos=1o2&hvnetw=g&hvrand=11335554994827564541&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9031140&hvtargid=pla-782556634792&psc=1

kxrf97, 2019-02-21:
@ederghar my car can only move forward or reverse as yours, when i press reverse and right, my led does not light up...... how you solve it with level shifter

ederghar, 2019-02-21:
Take a look at this article.  It has information on level shifting.

https://arduino.stackexchange.com/questions/21205/controlling-an-rc-car-using-arduino

kxrf97, 2019-02-21:
I need to put resistor between the controller and arduino? I have no knowledge about level shifting....... May you provide more specific instructions

kxrf97, 2019-02-24:
@ederghar level shifting is put resistor between the controller and arduino? may i see how you connect your controller with arduino?

kxrf97, 2019-02-24:
@ederghar  with level shifter? cause i have no engineering base... so i don't know how to construct with the hardware for level shifter...

ederghar, 2019-02-24:
I have attached a picture of my level shifting setup between the remote and arduino.  I used 330 Ohm resistors.
[image]