hamuchiwa/AutoRCCar #65: 你好，我有遇到一些问题，你能帮我解决吗？关于rc_driver.py

Jeff939, 2018-01-23:
[image]

Jeff939, 2018-01-23:
以下是coding，我改了但是出现了以上的问题，你可以帮我看看吗？还有那个ip是跟着电脑的还是raspberry的？

           def __init__(self):
        self.model = cv2.ml_ANN_MLP()

    distance_thread = threading.Thread(target=server_thread2, args=('192.168.0.7', 8002))
    distance_thread.start()
    video_thread = threading.Thread(target=server_thread('192.168.0.7', 8000))
    video_thread.start()

__author__ = 'zhengwang'

import threading
import SocketServer
import serial
import cv2
import numpy as np
import math

# distance data measured by ultrasonic sensor
sensor_data = " "

class NeuralNetwork(object):

    def __init__(self):
        self.model = cv2.ml_ANN_MLP()

    def create(self):
        layer_size = np.int32([38400, 32, 4])
        self.model.create(layer_size)
        self.model.load('mlp_xml/mlp.xml')

    def predict(self, samples):
        ret, resp = self.model.predict(samples)
        return resp.argmax(-1)

class RCControl(object):

    def __init__(self):
        self.serial_port = serial.Serial('/dev/tty.usbmodem1421', 115200, timeout=1)

    def steer(self, prediction):
        if prediction == 2:
            self.serial_port.write(chr(1))
            print("Forward")
        elif prediction == 0:
            self.serial_port.write(chr(7))
            print("Left")
        elif prediction == 1:
            self.serial_port.write(chr(6))
            print("Right")
        else:
            self.stop()

    def stop(self):
        self.serial_port.write(chr(0))

class DistanceToCamera(object):

    def __init__(self):
        # camera params
        self.alpha = 8.0 * math.pi / 180
        self.v0 = 119.865631204
        self.ay = 332.262498472

    def calculate(self, v, h, x_shift, image):
        # compute and return the distance from the target point to the camera
        d = h / math.tan(self.alpha + math.atan((v - self.v0) / self.ay))
        if d > 0:
            cv2.putText(image, "%.1fcm" % d,
[... 199 lines cut ...]
                        self.rc_car.stop()
                        break

            cv2.destroyAllWindows()

        finally:
            print ("Connection closed on thread 1")

class ThreadServer(object):

    def server_thread(host, port):
        server = SocketServer.TCPServer((host, port), VideoStreamHandler)
        server.serve_forever()

    def server_thread2(host, port):
        server = SocketServer.TCPServer((host, port), SensorDataHandler)
        server.serve_forever()

    distance_thread = threading.Thread(target=server_thread2, args=('192.168.0.7', 8002))
    distance_thread.start()
    video_thread = threading.Thread(target=server_thread('192.168.0.7', 8000))
    video_thread.start()

if __name__ == '__main__':
    ThreadServer()

Jeff939, 2018-01-23:
Hi, I'm having some problems when i run the rc_driver.py. By the way, the IP addresses are according to the computer ip or raspberry ip?
[image]

hamuchiwa, 2018-01-23:
你好呀，这个问题也有人遇到过，应该还是版本问题，你看看[这个链接](https://github.com/hamuchiwa/AutoRCCar/issues/60#issuecomment-356751492)是不是你需要的。
这个项目比较老了，当时用的是OpenCV2，换成3的话要改写一些地方。
pi的IP是跟着电脑，因为是客户端连到主机端。

Jeff939, 2018-01-23:
[image]
那个连接我之前看过了，我试过开一个新的file跑https://github.com/hamuchiwa/AutoRCCar/issues/60#issuecomment-356751492，他出来的result如以下，但是我不清楚这个需要改在哪里一个部分，
[image]
我能加你facebook or wechat方便交流吗？

hamuchiwa, 2018-01-23:
那个改动是训练数据用的（`mlp_training.py`），`dim` 要和你用的图片的分辨率一致。
预测的时候也要改，OpenCV3确实精简了一些：
```python
class NeuralNetwork(object):
    def __init__(self, file_path):
        self.model = cv2.ml.ANN_MLP_load(file_path)

    def predict(self, samples):
        ret, resp = self.model.predict(samples)
        return resp.argmax(-1)
```
```python
# create neural network
nn = NeuralNetwork('mlp_xml/mlp.xml')
prediction = nn.predict(X)
```
我微信绑的QQ，搜我QQ号就行：343162765
要方便的话你留个微信也行，晚上我加你~

marcusvn, 2018-01-29:
How to fix: **SystemError: error return without exception set**

closed 2018-06-29 by hamuchiwa