hamuchiwa/AutoRCCar #70: streaming error from client to server- ConnectionResetError: [Errno 104] Connection reset by peer

Rahul16697, 2018-02-24:
Traceback (most recent call last):
  File "/home/pi/stream/stream_client.py", line 23, in <module>
    connection.flush()
  File "/usr/lib/python3.5/socket.py", line 594, in write
    return self._sock.send(b)
ConnectionResetError: [Errno 104] Connection reset by peer

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/stream/stream_client.py", line 32, in <module>
    connection.close()
  File "/usr/lib/python3.5/socket.py", line 594, in write
    return self._sock.send(b)
BrokenPipeError: [Errno 32] Broken pipe

aakashm301, 2018-04-04:
What where you running ? Collect_trainning_data.py ? or another file ?

kaskagorry, 2018-04-10:
I have the same problem

I execute "collect_training_data.py" in PC, and in Raspberry i execute "stream_client.py" and i receive the same error.

Do you have the solution to this error?

Rahul16697, 2018-04-11:
I had and resolved it ....check the IP properly and port too ....thts the problem.......try the raspberry pi ip if it din work try with ua PC IP

closed 2018-04-16 by hamuchiwa