hamuchiwa/AutoRCCar #75: Collect_training_data

Gooose1909, 2018-03-08:
When i run this collect training data script . 
It seems to run fine 
Even prints forward , left , right in shell 
But still my training_data folder is empty . Why ???

hamuchiwa, 2018-03-08:
Press 'q' to break the loop, then data is saved in "training_data" directory.

Gooose1909, 2018-03-08:
I tried
It didnt work

hamuchiwa, 2018-03-09:
Are you able to output the image array during streaming?

Gooose1909, 2018-03-11:
No , 
Its like i get a window showing camera stream from my pi 
And frames are saved in training_images 
But my training_data is empty even though if i press 'q'

Gooose1909, 2018-03-11:
@hamuchiwa can you contact me here : 
sugan170@gmail.com
I really need your help . Planning to tweak some changes that's why

hamuchiwa, 2018-03-12:
Are you using windows, mac or linux? Have you tried setting an absolute folder path?

Gooose1909, 2018-03-14:
Windows

KhyatiMehta3, 2018-03-14:
Hi, I have the same problem. Also, I wanted to know why you're connecting to a serial port in this script? All the driving commands are also written to the same serial port ( Like self.ser.write(char()) ) I don't understand the reason for this.

hamuchiwa, 2018-03-15:
@Gooose1909 @KhyatiMehta3 If "train" numpy array is not empty, then I would say it's something to do with the path. Is absolute path working?

Gooose1909, 2018-03-15:
I found my problem was by absolute path
But @hamuchiwa is there a way to control car without remote
Like adding motor driver to pi and wirelessly controlling like server and
client!??

hamuchiwa, 2018-03-15:
@Gooose1909 I haven't done it myself, but there are people discussing and trying to do it with raspberry pi only.

Gooose1909, 2018-03-16:
@hamuchiwa can you please explain how u have interfaced remote to arduino briefly 
Please.

hamuchiwa, 2018-03-16:
@Gooose1909 Computer sends the prediction value to Arduino, and then Arduino sets high or low to its pins which are connected to the remote. 
In the case if you would like to use Raspberry Pi, like you mentioned earlier, the computer sends the prediction back to pi and then pi controls the motor accordingly.

Gooose1909, 2018-03-16:
@hamuchiwa
I meant like cicuit connection
Like did u connect batteries to the remote
Or u used the 3.3v from arduino
And also i have doubt like did u use optoisolator or not

hamuchiwa, 2018-03-16:
@Gooose1909 I used AA battery for the remote, although I think using optoisolator is a better way to do it.

closed 2018-04-16 by hamuchiwa

arunph, 2019-05-10:
OSError: [WinError 10048] Only one usage of each socket address (protocol/network address/port) is normally permitte.  
I m having above error when I run collect_training_data.py