hamuchiwa/AutoRCCar #76: Socket Programming help

Gooose1909, 2018-03-11:
the problem i faced was interfacing the remote of a RC car to arduino so i thought of sending w,a,s,d to pi for controlling my RC 
i have replaced the RF module in RC car to l298n Motor driver and interfaced it to pi and kept in on the car chasis 
so i send data from pc to pi for car . 
 only problem is i use pc as server and pi as client i get this error 
server code is collect_training_data.py modified for sending w,a,s,d
client code is stream_client.py 

i really dunno whats wrong or am in the complete wrong way

Traceback (most recent call last):
  File "C:\Users\Bindu\Desktop\AutoRCCar-master\computer\test.py", line 177, in <module>
    CollectTrainingData()
  File "C:\Users\Bindu\Desktop\AutoRCCar-master\computer\test.py", line 41, in __init__
    self.collect_image()
  File "C:\Users\Bindu\Desktop\AutoRCCar-master\computer\test.py", line 117, in collect_image
    self.client_address.send('w')
AttributeError: 'str' object has no attribute 'send'

# server code 
`
import numpy as np
import cv2
import serial
import pygame
from pygame.locals import *
import socket
import time
import os

class CollectTrainingData(object):
    
    def __init__(self):

        self.server_socket = socket.socket()
        self.server_socket.bind(('192.168.1.7', 4002))
        self.server_socket.listen(0)

        # accept a single connection
        (self.connection, (self.client_address,port)) = self.server_socket.accept()
        self.connection = self.connection.makefile('rb')
        print self.client_address
        print self.connection

        self.send_inst = True

        # create labels
        self.k = np.zeros((4, 4), 'float')
        for i in range(4):
            self.k[i, i] = 1
        self.temp_label = np.zeros((1, 4), 'float')

        pygame.init()
        size = [70, 50]
        screen = pygame.display.set_mode(size)
        self.collect_image()

    def collect_image(self):

        saved_frame = 0
        total_frame = 0

        # collect images for training
        print 'Start collecting images...'
        e1 = cv2.getTickCount()
        image_array = np.zeros((1, 38400))
        label_array = np.zeros((1, 4), 'float')

        # stream video frames one by one
        try:
            stream_bytes = ' '
            frame = 1
            while self.send_inst:
                stream_bytes += self.connection.read(1024)
                first = stream_bytes.find('\xff\xd8')
                last = stream_bytes.find('\xff\xd9')
                if first != -1 and last != -1:
                    jpg = stream_bytes[first:last + 2]
                    stream_bytes = stream_bytes[last + 2:]
                    image = cv2.imdecode(np.fromstring(jpg, dtype=np.uint8), cv2.CV_LOAD_IMAGE_GRAYSCALE)
                    
                    # select lower half of the image
                    roi = image[120:240, :]
                    
                    # save streamed images
                    cv2.imwrite('training_images/frame{:>05}.jpg'.format(frame), image)
                    
                    #cv2.imshow('roi_image', roi)
                    cv2.imshow('image', image)
                    
                    # reshape the roi image into one row array
                    temp_array = roi.reshape(1, 38400).astype(np.float32)
                    
                    frame += 1
                    total_frame += 1

                    # get input from human driver
                    for event in pygame.event.get():
                        if event.type == KEYDOWN:
                            key_input = pygame.key.get_pressed()

                            if key_input[pygame.K_w]:
                                print("Forward")
                                saved_frame += 1
                                image_array = np.vstack((image_array, temp_array))
                                print image_array
                                label_array = np.vstack((label_array, self.k[2]))
                                self.client_address.send('w')

                            elif key_input[pygame.K_s]:
                                print("Reverse")
                                saved_frame += 1
                                image_array = np.vstack((image_array, temp_array))
                                label_array = np.vstack((label_array, self.k[3]))
                                self.client_address.send('s')
                            
                            elif key_input[pygame.K_d]:
                                print("Right")
                                image_array = np.vstack((image_array, temp_array))
                                label_array = np.vstack((label_array, self.k[1]))
                                saved_frame += 1
                                self.client_address.send('d')

                            elif key_input[pygame.K_a]:
                                print("Left")
                                image_array = np.vstack((image_array, temp_array))
                                label_array = np.vstack((label_array, self.k[0]))
                                saved_frame += 1
                                self.client_address.send('a')

                            elif key_input[pygame.K_x] or key_input[pygame.K_q]:
                                print 'exit'
                                break
                                    
                            elif event.type == pygame.KEYUP:
                                pass

            # save training images and labels
            train = image_array[1:, :]
            train_labels = label_array[1:, :]

            # save training data as a numpy file
            file_name =str(int(time.time()))
            directory = "training_data"
            if not os.path.exists(directory):
                os.makedirs(directory)
            try:    
                np.savez(directory + '/' + file_name + '.npz', train=train, train_labels=train_labels)
            except IOError as e:
                print(e)

            e2 = cv2.getTickCount()
            # calculate streaming duration
            time0 = (e2 - e1) / cv2.getTickFrequency()
            print 'Streaming duration:', time0

            print(train.shape)
            print(train_labels.shape)
            print 'Total frame:', total_frame
            print 'Saved frame:', saved_frame
            print 'Dropped frame', total_frame - saved_frame

        finally:
            self.connection.close()
            self.server_socket.close()

if __name__ == '__main__':
    CollectTrainingData()
`

# Client Code

`
import car as c
import io
import socket
import struct
import time
import picamera

client_socket = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
client_socket.connect(('192.168.1.7', 4002))
connection = client_socket.makefile('wb')

try:
    with picamera.PiCamera() as camera:
        camera.resolution = (320, 240)      
        camera.framerate = 10               
        time.sleep(2)                       
        start = time.time()
        stream = io.BytesIO()
        
       
        for foo in camera.capture_continuous(stream, 'jpeg', use_video_port = True):
            connection.write(struct.pack('<L', stream.tell()))
            connection.flush()
            stream.seek(0)
            connection.write(stream.read())
            if time.time() - start > 600:
                break
            stream.seek(0)
            stream.truncate()
    connection.write(struct.pack('<L', 0))
    a = client_socket.recv(1024)
    if a == 'w':
        print 'foward'
        c.forward(0.050)
    elif a == 'a':
        print 'left'
        c.left(0.050)
    elif a == 's':
        print 'reverse'
        c.reverse(0.050)
    elif a == 'd':
        print 'right'
        c.right(0.050)
    else:
        c.init()
finally:
    connection.close()
    client_socket.close()

`

i know its long but please i really dunno whats wrong or am in the complete wrong way

KhyatiMehta3, 2018-03-14:
I'm new at this too, but I did do one change, that might help you as well. The problem is, you're trying to send a string to the socket which needs bytes. So, I'd suggest two changes : 
1.  In the stream videos part - 
try:
        stream_bytes = bytes()
        frame = 1
        while self.send_inst:
            stream_bytes += self.connection.read(1024)
            first = stream_bytes.find(b'\xff\xd8')
            last = stream_bytes.find(b'\xff\xd9')
2. To the key command sending side - Ex : for forward, 
w = ('w').encode()
self.clientaddress.send(w)
and that way replace all the strings with bytes.

Do try this and let me know if it works out!

KhyatiMehta3, 2018-03-14:
In the second part, what is even better is self.clientaddress.send(b 'w')

Gooose1909, 2018-03-14:
I tried both method of what you said. There has been no change. Same error pops up

KhyatiMehta3, 2018-03-14:
Try 
a = client_socket.recv(1024).decode()

Gooose1909, 2018-03-14:
Yes i tried along with decode.

KhyatiMehta3, 2018-03-14:
Is all your indentation according to the above code content? Sometimes a wrongly indented code might give such errors?

Gooose1909, 2018-03-14:
Everything is properly indented.
If u want i can even share you that .py file

KhyatiMehta3, 2018-03-14:
I think the function is "self.client_address.sendall('w')" and not "self.client_address.send('w')"

KhyatiMehta3, 2018-03-14:
I think the line is*

Gooose1909, 2018-03-15:
Didnt work 😪😪😪

KhyatiMehta3, 2018-03-15:
Can u share the client and server files?

Gooose1909, 2018-03-15:
Your mail id??

KhyatiMehta3, 2018-03-15:
khyasmailbox@rediffmail.com

KhyatiMehta3, 2018-03-15:
try this- self.client_address.sendall(str.encode("w"))

Gooose1909, 2018-03-15:
No improvement
I still grt the same error and i have sent you the mail od server and client 
Please help me to send a char or even an integer from server to client

KhyatiMehta3, 2018-03-15:
this will work - 
#self.con will be used to send info back to the pi.
(self.con, (self.client_address,port)) = self.server_socket.accept()
#Use this for reading picam data.
self.connection = self.con.makefile('rb')
#to send forward command - 
self.con.sendall(str.encode('w'))

Gooose1909, 2018-03-15:
Gives a new error
AttributeError : '_fileobject' object has no attribute sendall

rajaa1996, 2018-03-15:
When I start collect data py on i get a black box with a screen and when I try to open stream py file in raspberry pi i get some serial port error how to solve it

utsav57-2, 2018-03-17:
You will have to use two separate threads to send and receive at same time.

Gooose1909, 2018-03-18:
Can you elobrate a little please

KhyatiMehta3, 2018-03-18:
here's the server side code. :
import numpy as np
import cv2
import serial
import pygame
from pygame.locals import *
import socket
import time
import os

class CollectTrainingData(object):

    def __init__(self):

        self.server_socket = socket.socket()
        self.server_socket.bind(('192.168.1.3', 8000))
        self.server_socket.listen(0)

        # accept a single connection
        (self.con, (self.client_address,port)) = self.server_socket.accept()
        self.connection = self.con.makefile('rb')
        self.send_inst = True

        # create labels
        self.k = np.zeros((4, 4), 'float')
        for i in range(4):
            self.k[i, i] = 1
        self.temp_label = np.zeros((1, 4), 'float')

        pygame.init()
        size = [320, 240]
        screen = pygame.display.set_mode(size)
        self.collect_image()

    def collect_image(self):

        saved_frame = 0
        total_frame = 0

        # collect images for training
        print('Start collecting images...')
        e1 = cv2.getTickCount()
        image_array = np.zeros((1, 38400))
        label_array = np.zeros((1, 4), 'float')

        # stream video frames one by one
        try:
            stream_bytes = bytes()
            frame = 1
            while self.send_inst:
                stream_bytes += self.connection.read(1024)
                first = stream_bytes.find(b'\xff\xd8')
                last = stream_bytes.find(b'\xff\xd9')
                if first != -1 and last != -1:
                    jpg = stream_bytes[first:last + 2]
                    stream_bytes = stream_bytes[last + 2:]
                    image = cv2.imdecode(np.fromstring(jpg, dtype=np.uint8), cv2.IMREAD_GRAYSCALE)

                    # select lower half of the image
                    roi = image[120:240, :]

                    # save streamed images
                    cv2.imwrite('training_images/frame{:>05}.jpg'.format(frame), image)

                    #cv2.imshow('roi_image', roi)
                    cv2.imshow('image', image)

                    # reshape the roi image into one row array
                    temp_array = roi.reshape(1, 38400).astype(np.float32)

                    frame += 1
                    total_frame += 1

                    # get input from human driver
                    for event in pygame.event.get():
                        if event.type == KEYDOWN:
                            key_input = pygame.key.get_pressed()

                            if key_input[pygame.K_w]:
                                print("Forward")
                                saved_frame += 1
                                image_array = np.vstack((image_array, temp_array))
                                print(image_array)
                                label_array = np.vstack((label_array, self.k[2]))
                                self.con.sendall(str.encode('w'))

                            elif key_input[pygame.K_s]:
                                print("Reverse")
                                saved_frame += 1
                                image_array = np.vstack((image_array, temp_array))
                                label_array = np.vstack((label_array, self.k[3]))
                                self.con.sendall(str.encode('s'))

                            elif key_input[pygame.K_d]:
                                print("Right")
                                image_array = np.vstack((image_array, temp_array))
                                label_array = np.vstack((label_array, self.k[1]))
                                saved_frame += 1
                                self.con.sendall(str.encode('d'))

                            elif key_input[pygame.K_a]:
                                print("Left")
                                image_array = np.vstack((image_array, temp_array))
                                label_array = np.vstack((label_array, self.k[0]))
                                saved_frame += 1
                                self.con.sendall(str.encode('a'))

                            elif key_input[pygame.K_x] or key_input[pygame.K_q]:
                                self.con.sendall(str.encode('q'))
                                print('exit')
                                self.send_inst = False
                                break

                            elif event.type == pygame.KEYUP:
                                pass

            # save training images and labels
            train = image_array[1:, :]
            train_labels = label_array[1:, :]

            # save training data as a numpy file
            file_name =str(int(time.time()))
            directory = "training_data"
            if not os.path.exists(directory):
                os.makedirs(directory)
            try:
                np.savez(directory + '/' + file_name + '.npz', train=train, train_labels=train_labels)
            except IOError as e:
                print(e)

            e2 = cv2.getTickCount()
            # calculate streaming duration
            time0 = (e2 - e1) / cv2.getTickFrequency()
            #print('Streaming duration:', time0)

            #print(train.shape)
            #print(train_labels.shape)
            #print('Total frame:', total_frame)
            #print('Saved frame:', saved_frame)
            #print('Dropped frame', total_frame - saved_frame)
        finally:
            self.con.close()
            self.connection.close()
            self.server_socket.close()
if __name__ == '__main__':
    CollectTrainingData()

KhyatiMehta3, 2018-03-18:
At the sever side, there should be no problem.
Problem at the client side is that it will stream picamera data to the server but not receive commands from the server at the same time. If anyone has solution for this, then your problem would be solved.

Here's the client side code : 
import socket
import RPi.GPIO as GPIO
import sys
import picamera
import time
import io

cs = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
cs.connect(('192.168.1.3',8000))
connection = cs.makefile('wb')
con = cs.makefile('rb')

def init():
    GPIO.setmode(GPIO.BOARD)
    GPIO.setup(03, GPIO.OUT) #back motor pin
    GPIO.setup(05, GPIO.OUT) #back motor pin
    GPIO.setup(07, GPIO.OUT) #left pin
    GPIO.setup(11, GPIO.OUT) #right pin
def forwardGPIO():
    GPIO.output(03,True)
    GPIO.output(05,False)
    clean()
def reverseGPIO():
    GPIO.output(03,False)
    GPIO.output(05,True)
    clean()
def rightGPIO():
    GPIO.output(07,False)
    GPIO.output(11,True)
    clean()
def leftGPIO():
    GPIO.output(07,True)
    GPIO.output(11,False)
    clean()
def clean():
    GPIO.cleanup()

try:
    init()
    with picamera.PiCamera() as cam:
        cam.resolution = (320,240)
        cam.framerate = 10
        time.sleep(2)
        start = time.time()
        stream = io.BytesIO()

        for foo in camera.capture_continuous(stream,'jpeg',use_video_port=True):
            connection.write(struct.pack('<L',stream.tell()))
            connection.flush()
            stream.seek(0)
            connection.write(stream.read())
            if time.time() - start > 600:
                break
            stream.seek(0)
            stream.truncate()
    connection.write(struct.pack('<L',0))
    with con as c:
        direction = c.read(1024).decode()
        if direction == 'w':
            forwardGPIO()
        elif direction == 'a':
            leftGPIO()
        elif direction == 's':
            reverseGPIO()
        elif direction == 'd':
            rightGPIO()
        else:
            clean()
finally:
    con.close()
    connection.close()
    cs.close()

Gooose1909, 2018-03-18:
Using these server and client code
Solves the error
Now it wont any issuea right????

Gooose1909, 2018-03-18:
Am having issues with indentations
Can u mail those to sugan170@gmail.com
Thanks alot mate

KhyatiMehta3, 2018-03-18:
I'm not sure if you;ll have issues or not, because i'm having a slight issue at the client side. server side has no problems. Sure, I'll send the files.

utsav57-2, 2018-03-19:
If you are still having problems just make two client files, one for streaming and other for controlling the car.
And on server side accept two connections and send the commands(w,s,a,d or whatever) to 2nd client.

closed 2018-06-29 by hamuchiwa