hamuchiwa/AutoRCCar #77: Delay in steam.py

rajaa1996, 2018-03-15:
I get too much delay during streaming of video a lag of about 20sec i use vncserver chrome extension is there any other way to stream or changes to make the delay small

Gooose1909, 2018-03-15:
If u follow his method  for streaming u will have stream without any delays

rajaa1996, 2018-03-15:
I did the same as mentioned in this repository but I dono how to stream to my pc using which software

Gooose1909, 2018-03-15:
U dont have to use any software
Just python
Go to test folder from ur pc
And under stream_test edit the ip to your computer's ip, 4000 as port
And paste the raspberry pi folder in ur raspberry pi
And open stream python file and change the ip your computer 's Ip and same
port number (4000) for that to work

Note: both. py will have your computer' s ip and both your computer and
raspberry pi should be in same network (same wifi)
After this run the py file first in ur Computer then on your pi

rajaa1996, 2018-03-15:
Tysm ?!!great man!!!!

Gooose1909, 2018-03-15:
Works fine now?????

rajaa1996, 2018-03-15:
Where should I run rc_driver.py on my computer or on my raspberry pi

Gooose1909, 2018-03-15:
Computer
The file in 'computer'  should be run in computer
The files in' raspberry pi 'should be run on pi

rajaa1996, 2018-03-15:
I copied my the computer folder on my pc then how should I run can u tell me little in detail plz

Gooose1909, 2018-03-15:
Did u flash the arduino file to your arduino and connect it to the RC car's
remote??

rajaa1996, 2018-03-15:
I used l293d connected to arduibo nano then connect it to my raspberry pi

rajaa1996, 2018-03-15:
I control via my PCs keyboard

Gooose1909, 2018-03-15:
Can u send me your project files and details??

rajaa1996, 2018-03-15:
@Goose1909 
Check this post
https://github.com/sidroopdaska/SelfDrivingRCCar/blob/master/README.md

Gooose1909, 2018-03-15:
@rajaa1996
Thats same as of hamuchiwa's
The reason i tried to send string via socket was i couldn't able to
interface RC remote to arduino
Cos i couldn't find correct points
Thats why
Anyone wanna help me with that problem

rajaa1996, 2018-03-16:
I have some prob can u give ur mail id so that I can contact u

Gooose1909, 2018-03-16:
sugan170@gmail.com

marcusvn, 2018-06-11:
hello @rajaa1996  I would like to have some question, can you contact me plz? my email: nguyenphuc1206@gmail.com

kxrf97, 2019-03-06:
@Gooose1909 can you advise me on training the model? my car just keep going left .....

ShwetaLalwani, 2019-03-23:
@Gooose1909 How to select port ? Or any port number is fine? How you chose 4000?

Gooose1909, 2019-03-23:
You can select any number for port .
I prefer 4000 and 4001
If you want u cant select 5000 too

Gooose1909, 2019-03-23:
I faced the same issue i just made my own RC car fixed everything

kxrf97, 2019-03-23:
I mean how you train the car? My car just keep going randomly lol. 😂😂

kxrf97, 2019-03-23:
I manage to stream 15 to 20 frame per second

Gooose1909, 2019-03-23:
Run collect training data.py
To collect your images
Then run training
Then u can run rc driver

kxrf97, 2019-03-23:
Yaaa after i trained, and run rc driver, my car cant successfull navigate within the track

Gooose1909, 2019-03-23:
Accuracy ??

kxrf97, 2019-03-23:
Around 70, how many data u collect from a single short track

Gooose1909, 2019-03-23:
200+

kxrf97, 2019-03-23:
Accuracy?

Gooose1909, 2019-03-23:
89+

ShwetaLalwani, 2019-05-09:
Is your project completed? @Gooose1909

arpitpartani, 2019-06-24:
> U dont have to use any software Just python Go to test folder from ur pc And under stream_test edit the ip to your computer's ip, 4000 as port And paste the raspberry pi folder in ur raspberry pi [...]

still getting an error of cannot assingn requested address while running test_stream_server in computer linux

Sirajabudabe, 2019-09-05:
@Gooose1909  could you please contact me on my e-mail : serajabudaber@gmail.com