hamuchiwa/AutoRCCar #78: The Car drive wrong way !

marcusvn, 2018-03-15:
I do all work for the Car but when I run Rc_Driver.py, The Car drives wrong way. Anyone have any idea to help me?

hamuchiwa, 2018-03-15:
Number of images and data quality can both affect the result, what is the accuracy of your model?

marcusvn, 2018-03-15:
Just 50% after run mpl_tranning. So I should collecting a lot of data or what ?

hamuchiwa, 2018-03-15:
0.5 is too low, you need at least 0.8 to generate good result. For a simple short track, roughly 200 images should be enough.

marcusvn, 2018-03-16:
Should I use Collecting_Tranning_Data for a lot of times or just use it one two times?

marcusvn, 2018-03-16:
When I running first Collecting_Tranning Data again!
[image]

hamuchiwa, 2018-03-16:
I would suggest collecting multiple files, it's just easier to do in that way in case you need to discard any data. All individual files will be merged together in the training phase.

marcusvn, 2018-03-16:
Collecting tranning file when train accurcy reach 0,8 that right?

hamuchiwa, 2018-03-16:
Yes, you can try with a straight line, a left turn, then a right turn, and see what happens.

marcusvn, 2018-03-17:
Thanks For helping me! I will test again! If I encounter problems I will bother you! thanks!

closed 2018-03-19 by hamuchiwa

kxrf97, 2019-03-06:
@marcusvn did you manage tackle the training problem?